OpenCPN/OpenCPN #3445: Flatpak install grib plugin won’t create email

crayiii, 2023-10-01:
**Describe the bug**
Installed the Flatpak version of OpenCPN in Debian. When I create a grib request in the grib plugin it will not create the email in the default app. I’ve tried Thunderbird, Evolution, and Geary. 

**To Reproduce**
Steps to reproduce the behavior:
Create grib request and click send

**Expected behavior**
A email would be created ready to send to saildocs

**Desktop (please complete the following information if applicable):**
 Debian 12 with Gnome
.

leamas, 2023-10-02:
hm... sending email from Flatpak depends on /usr/bin/xdg-email. On Debian, this is part of the xdg-utils package.

This is a bug, no doubt. That said, have you installed the xdg-utils package? If not, what happens if you do? Still no email?

TwoCanPlugIn, 2023-10-03:
Could it be a flatpak permissions issue?

leamas, 2023-10-04:
@TwoCanPlugIn Probably not.. There is a. "portal' which is exposed through xdg-email. The net result is that x-e is visible in the sandbox. However, this depends on that x-e is available in the native environment i. e., that xdg-utils is installed

leamas, 2023-10-04:
The core problem is that on Debian the Flatpak Opencpn installation depends on xdg-utils, but there is no way to express this.

Perhaps the flatpak package should pull in x- e, at least as a weak Recommends:

leamas, 2023-10-04:
@TwoCanPlugIn Probably not.. There is a. "portal' which is exposed through xdg-email. The net result is that x-e is visible on the sandbox. However, this depends on that x-e is available in the native environment i.

TwoCanPlugIn, 2023-10-04:
Are the flatpak-xdg-utils required?.

leamas, 2023-10-04:
For xdg-email I suspect that.

leamas, 2023-10-04:
@crayiii Could you please:
  - First check if anything changes after installing the _xdg-utils_ package?
  - If still the same situation, make a new check after installing _flatpak-xdg-utils_?

crayiii, 2023-10-04:
I ended up installing OpenCPN from the Ubuntu repository because we needed the system up and running ASAP. 

I’ll try the solution as quickly as possible though because Flatpak would be convenient for updates.

leamas, 2023-10-04:
I'm  actually suspicious about the xdg-utils package, I have reproduced this locally and found something which looks like another bug. Please stay tuned

leamas, 2023-10-04:
@crayiii : OK, after some messing around we have a test build available using

    $ flatpak install --user https://dl.flathub.org/build-repo/53650/org.opencpn.OpenCPN.flatpakref` 
    $ flatpak run org.opencpn.OpenCPN//test

Note that this test version is installed in parallel with you current installation and won't affect it in any way. Could you test and verify that this fixes your problem?

FWIW, it does fix it for me.

crayiii, 2023-10-04:
Fixed!

leamas, 2023-10-04:
Applied to a new downstream flatpak release 5.8.4-2.  It might take some hours before everything is uploaded and available.

This bug also affects native builds. 

Fixed in b941bf8. Closing

PR #3450 (2023-10-04): grib_pi: Fix shell syntax invoking xdg-email (#3445)
Closes: #3445
commit b941bf8d (2023-10-04): grib_pi: Fix shell syntax invoking xdg-email (#3445)
Closes: #3445