OpenCPN/OpenCPN #3448: Add proper support for arm64 and universal binaries to plugin manger

nohal, 2023-10-03:
Both OpenCPN and plugins on macOS can be built for either `arm64`, `x86_64` or universal including both architectures in the same binary. We need to be able to support it in the plugin metadata.

leamas, 2023-10-03:
There are many parts here.. 
  - The first would be to add new ABI definitions to plugins/ocpn-plugin.xsd in the plugins project
  - We need to update core openpcn, plugin_handler.cpp, the function `isCompatible()` so it loads the correct plugins.
  - We need to update the templates, shipdriver and testplugin, in order to get build what should be built.

As for the last bullet: should we build all three variants of each plugin, or should we limit it to either universal or aarch64+x86_64?

nohal, 2023-10-03:
I would build only universal, for both OpenCPN and the plugins. I see no need for single platform packages so far.
I am currently working on proper detection of the runtime architecture. 
The plugin build system update is very simple, basically https://github.com/nohal/nsk_pi/commit/5cbff5eac1e0380262f7adcd57f5016785c14d1a
What's left is the metadata, and handling them.

nohal, 2024-02-13:
Implemented

closed 2024-02-13 by nohal