OpenCPN/OpenCPN #3451: Active track is no more one pixel wide when zooming + Zooming make all tracks turn yellow

Ric121, 2023-10-05:
I report only  one issue because I think this is linked to the same issue.

**Describe the bug**
In version 5.2.x, the active track was traced in red, while old tracks were yellow by default. When you zoomed and overzoomed, the active track width size was still one pixel.
In version 5.8.x, the active track is traced in red but when you overzoom on the chart it is replaced with a yellow track with a wide width.

This led to confusion with previous tracks that can be present at the same place which are now all yellow, whaterevr their actual coulor settin is

When you use the anchor watch feature you can't  see anymore if you're boat is moving in a reasonable circle as the track width is too large and just fill the circle in 5 minutes

See screenshot attached for a better understanding.

**To Reproduce**
Steps to reproduce the behavior:

Example 1
1- Record 2 tracks with default colour, in the same zone.
2- Change the colour of one track, for instance dark blue 
3- Check that you see one track yellow and one track dark blue (Screenshot 1)
4- Then start zooming. At some point a Yellow track appears below the dark blue track, thus kind of greenish (Screenshot 2)
5- Keep zooming, the dark blue track is now totally yellow (Screenshot 3)

Example 2:
1- Enable tracking and move the boat
2- Check that the track is drawn RED
3- overzoom on the boat: the track turns yellow

Example 3:
1- Enable tracking, boat stoped
2- Enable an Anchor watch with a Readius od 40m
3- Overzoom on the boat to visuallize the anchor circle
=> The track is yellow but because of the width of the tracks it fast overfills the circle, making the anchorwatch function not too usefull (Screenshot from 2023-10-05 15-50-58)

**Expected behavior**

- Like in previous version, The active track should be one pixel width. It shouldn't be oversized when you zoom.
- The passed tracks should keep their color even when you zoom.

**Desktop (please complete the following information if applicable):**
 - OS: Raspbian 11 (Bulleys)
 - Opencpn 5.8.4

**Additional context**
Attch screenshot
[image]
[image]
[image]
[image]

Hakansv, 2023-10-05:
Rac121..
The active, red, track is default sized unless "hacked" in the config file.
The same goes for the historical, yellow, track.
The color for neither of these tracks are able to change in Options.

But (maybe confusing?)
You can enable Options=>Ships=>Own ship: [  ] "Highlight Tracks" and the color for that highlight is able to change.
Looking at your zoomed anchor picture I think you have enable Highlight Track in yellow and that's huge. Looking closely you can also see the red "normal" track indicating a small south-west moving.
Try to in Options disable "Highlight Track" and study the effect for your other examples.

Ric121, 2023-10-05:
Yes that's it! What confused me is that I didn't change anything , just updated the new version and the behaviour was different. Maybe this setting was not applied to the active track in the previous version. I was not aware of it (I'm still not sure to understand what this setting is usefull for actually ;-). Thanks a lot.

Hakansv, 2023-10-06:
> (I'm still not sure to understand what this setting is usefull for actually

Since the option is implemented someone does like it. That's one of the benefits of a opensource project like OCPN. The community can contribute features but better make it optional since we are not all made from the same mold. 

If you're happy and this is not a bug would you please close the issue?
Thanks.

Ric121, 2023-10-06:
OK. Thanks

closed 2023-10-06 by Ric121