OpenCPN/OpenCPN #3452: Incorrect Mode Indicator for generated RMB, RMC and XTE sentences

TwoCanPlugIn, 2023-10-05:
When a route or waypoint is activated, OpenCPN generates APB, RMB, RMC and XTE sentences.

First problem is that the XTE sentence lacks the mode indicator field. This prevents the faithful conversion of a XTE sentence to its NMEA 2000 equivalent; PGN 129283. This in turn prevents the control of a NMEA 2000 autopilot.

Second problem is that the mode indicator field of generated RMB & RMC sentences does not accurately reflect the mode indicator field of a GGA sentence that is transmitted by a GPS transceiver (or a GPS simulator) that provides position information to OpenCPN.

Third problem is that the status field of the generated RMB & RMC sentences does not accurately reflect the mode indicator field. The status field must be set to V (invalid) for all values of the mode indicator field except A (Autonomous) or D (Differential.

The following GGA & GLL sentences are sent to OpenCPN.
$GPGGA,215054,3811.00,S,14430.00,E,8,10,0.98,1113.0,M,-21.3,M,25,10*71
$GPGLL,3811.00,S,14430.00,E,215054.00,A,S*68

Note the GPS Quality Indicator field (field 6) of the GGA sentence is set to 8, meaning Simulated Mode.
Note the Mode Indicator field (field 7) is set to S, meaning Simulated Mode.

When a route or waypoint is activated, the following sentences are generated:
$ECRMB,A,0.000,L,001,002,3812.170,S,14431.320,E,1.564,138.434,-0.000,V,A*4A
$ECRMC,203329,A,3811.000,S,14430.000,E,0.000,0.000,051023,11.803,E,A*29
$ECAPB,A,A,0.000,L,N,V,V,138.434,T,002,138.434,T,138.434,T*38
$ECXTE,A,A,0.000,L,N*4F

Note the Mode Indicator (last field) of the generated RMB & RMC sentences is set to A (Autonomous) instead of correctly reflecting the Mode Indicator field of the GGA & GLL sentences. 

Also note that the Status field (first and second field of RMB & RMC respectively) is set to A (Valid) and should be set to V (Invalid) as the Mode Indicator is not one of A (Autonomous) or D (Differential)

Finally note that the XTE sentence is missing the Mode Indicator field.
It should be:
$--XTE,A,A,x.x,a,N,a*hh<CR><LF>  

These bugs are preventing the release of the next version of the TwoCan plugin and its support for controlling NMEA 2000 based autopilots.

Hakansv, 2023-10-10:
Discussion and opinions ....

I think we should take care implementing strict relations to received N0183 sentences as conditions to use the Auto pilot functions in OCPN.
One advantage with OCPN is it's possible to use also on dated low budget ships using navigation tool of different brands and production dates. The very strict dependencies you suggest can destroy the ability to use the AP function for many users. Please see my opinions herein.

Regardless of any mode indicators of various N0183 sentences it must be the actual skippers responsibility to decide if the AP functions can be used or not. If for example a GNSS receiver is in a simulation mode and the skipper decide to test the AP function OCPN should not hinder that. We, OCPN, must trust any skipper/user does have control.

> First problem is that the XTE sentence lacks the mode indicator field.

The mode indicator field to XTE was introduced as optional after N0183 version 2.3 (1998)
I and probably others have a AP (like Navico WP30) manufactured before 1998 and it may not understand a XTE of version >=2.3. I think it's wise OCPN don't use the optional mode indicator since it may jeopardize connection to older auto pilots.
If that mode field is necessary for a N2k transmission I'd suggest you set it good if omitted.

> Second problem is that the mode indicator field of generated RMB & RMC sentences does not accurately reflect the mode indicator field of a GGA sentence that is transmitted by a GPS transceiver

I've similar doubts to introduce a strict dependency of received GGA and/or GLL sentences. 
Not all GNSS receivers are transmitting these sentences. For example many AIS transponders don't and users may have omitted them from a GNSS receiver to support other sentences using restricted bauderate e.g. 4800.

TwoCanPlugIn, 2023-10-13:
Thanks for your comments Hakan,

> I think we should take care implementing strict relations to received N0183 sentences as conditions to use the Auto pilot functions in OCPN.

This is not just for autopilot control,, per se it is for correctly parsing GGA, GLL & RMC NMEA 0183 sentences and the implication for generating XTE and APB sentences or for converting them to the respective NMEA 2000 PGN's, 129283 & 129284. 

> One advantage with OCPN is it's possible to use also on dated low budget ships using navigation tool of different brands and production dates.

In the same way that we have an option for ignoring NMEA 183 checksums, there should also be an option for ignoring NMEA 0183 Mode Indicator field (which was introduced way back in 1998 with NMEA 183 version 2.3, so should be supported on quite old equipment)

> Not all GNSS receivers are transmitting these sentences.

I'm not aware of any GPS that doesn't transmit GGA, GLL and RMC sentences. Similarly I am not aware of any AIS that doesn't, although most allow the user to optionally disable the transmission of these.

The simple fact is that XTE, RMB and APB sentences are not faithfully generated based on the information present in GGA, GLL or RMC sentences.

My example of using simulated mode in GGA, GLL and RMC sentences was just to demonstrate that OpenCPN does not faithfully parse or interpret them.

If instead, in a GGA sentence I change the GPS Quality Mode from 1 or 2 (GPS or Differential GPS fix) to 0 (fix invalid or unavailable)  whilst navigating to a waypoint of following a route, OpenCPN incorrectly generates APB, RMB and XTE sentences with the Status fields set to 'A' (Valid) which is totally wrong. And other than the GPS icon changing from green bars to a green circle, OpenCPN doesn't even raise an alert indicating the GPS fix has been lost. This is just plain wrong and dangerous.

The following GGA sentence
`$GPGGA,012905,3811.00,S,14430.00,E,0,10,0.98,1113.0,M,-21.3,M,25,10*71`

should **not** result in OpenCPN generating the following:

`$ECRMB,A,0.000,L,002,001,3812.721,S,14431.292,E,1.999,149.470,-0.000,V,A*48`
`$ECAPB,A,A,0.000,L,N,V,V,149.470,T,002,149.470,T,149.470,T*3E`
`$ECXTE,A,A,0.000,L,N*4F`

This is just plain wrong.

For features I am testing in future releases of TwoCan, I am loathe to introduce yet another workaround because of deficiencies in OpenCPN.

bdbcat, 2023-10-30:
All..
We need to address this bug, but gently.  Bug fix must not break existing user's installations unless unavoidable.
We set aside the point about XTE sentence, temporarily.

Consider the GGA Quality Indicator field. All GPS receivers that transmit GGA can be relied upon to support values of 0, 1, or 2 in this field.  More modern devices may support other flags here, such as Simulation mode.

I propose that we establish a new global representing GPS "validity" as follows:
1.  If incoming GGL sentence contains "0" in Quality Indicator field, "validity" is INVALID.  Otherwise, GPS is judged valid according to GGA.
2. In incoming RMC contains "A" in the last field, GPS "validity" is INVALID according to RMC.
3. If either RMC or GGA report INVALID, the "validity" is INVALID.
4. The GPS "validity" value is used to mark the Mode field of autopilot output sentences, "V" or "A".

Full stop here, inviting comments.

TwoCanPlugIn, 2023-10-30:
If the GLL sentence is also used by OpenCPN for a position fix, then if the Mode Indicator (last field) is anything other than 'A' or 'D', the Status (second last field) is 'V' (Invalid).

For NMEA 2000, in PGN 129029, you should exhibit the same behaviour for field 8 (GNSS Method(, 0 - No Fix)

That extends 3. to read; If RMC or GGA or GLL report INVALID or PGN 129029 Method == 0, the "validity" is INVALID.

TwoCanPlugIn, 2023-10-30:
I should also add, and I do not know if Timo's libraries report this, if the latitude or longitude for PGN 129025 and 129029 are equal to INT_MAX (Data Unavailable), INT_MAX - 1 (Error) and INT_MAX - 2 (which I think usually is Reserved), the fix is also invalid.

bdbcat, 2023-10-30:
OK, understood.
This will involve the OCPN priority logic, especially the "fail-over" logic.  That is, transmitted AP sentences must follow the status of the currently active (i.e. highest priority) nav source status, whether N0183 or N2k, and the sub-messages within.

We are not using Timo's library verbatim.  We have a partial local copy in our source tree. We can and should make mods as necessary for OCPN functionality.

bdbcat, 2023-11-14:
I have published a patch which corrects the most critical of your reported errors, namely the incorrect RMC, RMB, and APB sentences transmitted when the GPS receiver signal is lost.  The autopilot control sentences are now marked as invalid when OCPN position determination is deemed invalid.
Testing and verification will be appreciated.

I did **not** make patches to require the autopilot output control sentences to match the exact mode indications of corresponding GGA or GLL input sentences.  Here is the logic:
1.  OCPN can determined ownship's position from various sources, including live GPS receiver input as N0183 or N2k, signalK, or AIVDO sentences.
2.  OCPN synthesizes autopilot output sentences based upon its resolved view of ownship position and nav status.  Such a view may not derive from N0183 GGA or GLL sentences at all.
3. Consequently, it is not a good general case to attempt to be more specific on the mode in the output sentences.  "A" or "V" should be enough for the large majority of autopilot control logics.

Finally, nothing has changed for XTE sentence.  This requires additional UI work to allow de-selection of NMEA 2.3 mode for XTE output.  Perhaps this should be the subject of another bug report.

bdbcat, 2024-01-04:
TwoCan...
Done here?  Further thoughts?

Hakansv, 2024-04-07:
I think this issue is solved as far we want?
RMB/RMC "IsDataValid" mode is true(A) if we have valid position else false. A valid position is when the GPS Compass is green and boat symbol is red. (So not reflecting a GNSS simulate mode but who want to start the autopilot in that case unless testing?)
XTE is not including the FAA mode indicator (NMEA 2.3 and later, optional). Personally I think that's how we should use this. We have users out there using a AP from before 1998 when this FAA mode was not introduced. (Including me :))

TwoCanPlugIn, 2024-04-09:
Is this patch in the current 5.9,x builds ?

Not sure that this is nailed down yet.

This is what I feed O.

`$GPGGA,013032,3811.00,S,14430.00,E,0,10,0.98,1113.0,M,-21.3,M,25,10*7D
$GPGLL,3811.00,S,14430.00,E,013032.00,V,N*66`

Field 6 of GGA (GPS Quality Indicator) is 0, meaning no GPS fix.
Field 6 of GLL (Status) is V, meaning Data Not Valid
Field 7 (Mode Indicator) of GLL is N, meaning Data Not Valid.

(Note for GLL, Status is A (Valid), only if Mode Indicator is A (Autonomous) or D D=(Differential)

This is the output when a route is active.

`$ECRMB,V,0.000,L,001,002,3844.074,S,14429.101,E,33.141,181.219,-0.000,V,A*6A
$ECRMC,004002,V,3811.000,S,14430.000,E,0.000,0.000,090424,0.000,E,A*04
$ECAPB,V,A,0.000,L,N,V,V,181.219,T,002,181.219,T,181.219,T*24
$ECXTE,A,A,0.000,L,N*4F

For RMB Sentence, Field 1 (Status) correctly shows V (Invalid), however last field (Mode Indicator) incorrectly shows A (Autonomous) instead of N (Data not Valid)
Similarly for RMC, Field 2 (Status) is correct, but last field shows the same error as the RMB sentence.
For the APB sentence, field 1 is correct, field 2 is incorrect, should be V (Invalid) and I'll grudgingly agree to the omission of the mode indicator field. (BTW, I have the same problem as Hakan, an old Robertson autopilot !)
For XTE, the  first two fields (status) are both incorrect, they should be V (Invalid) 

BTW, the boat symbol is green, satellite symbol is red.

Clear as mud !

bdbcat, 2024-04-09:
"BTW, the boat symbol is green, satellite symbol is red."
???

bdbcat, 2024-04-09:
OK now?

rgleason, 2024-04-09:
Hakan, should we add anything about this to the manual or not?
Maybe here  https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:connections:autopilot

or here?
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:connections:data-sharing

Hakansv, 2024-04-09:
My views:
No the RMC, .... discussion here is nothing for a general user taking care of the autopilot connection.

The autopilot discussion in "Data sharing" would fit better in "Autopilot" before the more rare discussion about Raymarine.

TwoCanPlugIn, 2024-04-09:
Agree with Hakan.

This stuff doesn't need to be documented in a user manual. In fact I think one of the problems with the manual is that it has too many sections with this kind of technical discussion.

For end users, the sections need to reflect actual tasks. Eg. Connecting an autopilot. Sharing data with other devices (eg. Navionics on a tablet)

bdbcat, 2024-04-09:
Ready to close?

rgleason, 2024-04-09:
Yes, I'll make those changes. Thanks. Agreed about the manual. We all are working on it.

TwoCanPlugIn, 2024-04-09:
This seems OK.

Following sentences:
`$GPGGA,230559,3811.00,S,14430.00,E,0,10,0.98,1113.0,M,-21.3,M,25,10*76`
`$GPGLL,3811.00,S,14430.00,E,230559.00,V,N*6D`

result in these
`$ECRMB,V,0.000,L,001,002,3803.798,S,14444.997,E,13.847,58.596,0.000,V,N*7A`
`$ECRMC,220254,V,3811.000,S,14430.000,E,0.000,0.000,090424,0.000,E,N*0E`
`$ECAPB,V,V,0.000,L,N,V,V,58.596,T,002,58.596,T,58.596,T*06`
`$ECXTE,V,V,0.000,L,N*4F`

Are you guys on a death march ?

However bug in SignalK processing.

Feeding SignalK with the above GGA & GLL sentences results in the following JSON

{"updates":[{"source":{"sentence":"GGA","talker":"GP","type":"NMEA0183","label":"NMEA183-UDP"},"timestamp":"2024-04-09T22:45:34.000Z","values":[{"path":"navigation.position","value":{"longitude":144.5,"latitude":-38.18333333333333}},{"path":"navigation.gnss.methodQuality","value":"no GPS"},{"path":"navigation.gnss.satellites","value":10},{"path":"navigation.gnss.antennaAltitude","value":1113},{"path":"navigation.gnss.horizontalDilution","value":0.98},{"path":"navigation.gnss.differentialAge","value":25},{"path":"navigation.gnss.differentialReference","value":10}],"$source":"NMEA183-UDP.GP"}],"context":"vessels.urn:mrn:signalk:uuid:1cb1a66a-814c-4478-8b84-701eec9524bb"}

But OpenCPN appears to ignore GPS Method Quality value (in this case No GPS)

Do you want a separate bug ?

Which also leads to me think of alerts, if a route is active, and a sailor is using an autopilot, should O not raise an alert and/or audible alarm? (I personally feel that O needs a lot more alarms, shallow water, lost GPS, etc.)

rgleason, 2024-04-09:
Watchdog has lost GPS and many other alarms.

TwoCanPlugIn, 2024-04-10:
Oops, you're right, I'd forgotten about the watchdog plugin.

However IIRC for those alarms such as depth, lost gps it doesn't parse SignalK or NMEA 2000 ?

bdbcat, 2024-04-10:
1.  Are you guys on a death march ? YES
2.  Do you want a separate bug ? YES
3.  This thread has nothing to do with Watchdog.

bdbcat, 2024-04-10:
Please close now, if satisfied

closed 2024-04-10 by TwoCanPlugIn