OpenCPN/OpenCPN #3454: OpenCPN 5.8.4 crashes on start, jack server is not running or cannot be started on a fresh Debian Bookworm install.

as365n4, 2023-10-08:
**Describe the bug**
OpenCPN 5.8.4 crashes on start, jack server is not running or cannot be started on a fresh Debian Bookworm install.

Installed Debian 12.2 on my SBC (arm64), installed XFCE 4 Desktop Environment (xfce4/stable,now 4.18 all), installed OpenCPN (opencpn/stable-backports,now 5.8.4+dfsg-1~bpo12+1 arm64) from Bookworm-Backports.
Tried to start the app and it imediately crashes.
Examining the Bash output it seems to require JACK but Debian Bookworm ships with PulseAudio (pulseaudio/stable,now 16.1+dfsg1-2+b1 arm64) by default.

#3380 debian/rules: Use default portaudio sound refers.

**Desktop (please complete the following information if applicable):**
 - OS: Linux 6.1.0-13-arm64 SMP Debian 6.1.55-1 (2023-09-29) aarch64 GNU/Linux
 - Version [12.2]

**Additional context**

Bash output:

ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
ALSA lib pcm_a52.c:1001:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib conf.c:5670:(snd_config_expand) Unknown parameters {AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2 CARD 0}
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2 CARD 0}
ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
13:20:07: Debug: Device: 0: SPDIF: ff030000.spdif-dit-hifi dit-hifi-0 (hw:0,0)
13:20:07: Debug: Device: 1: HDMI: ff000000.i2s-i2s-hifi i2s-hifi-0 (hw:1,0)
13:20:07: Debug: Device: 2: sysdefault
13:20:07: Debug: Device: 3: lavrate
13:20:07: Debug: Device: 4: samplerate
13:20:07: Debug: Device: 5: speexrate
13:20:07: Debug: Device: 6: pulse
13:20:07: Debug: Device: 7: speex
13:20:07: Debug: Device: 8: upmix
13:20:07: Debug: Device: 9: vdownmix
13:20:07: Debug: Device: 10: dmix
13:20:07: Debug: Device: 11: default
13:20:07: Debug: Device: 0: SPDIF: ff030000.spdif-dit-hifi dit-hifi-0 (hw:0,0)
13:20:07: Debug: Device: 1: HDMI: ff000000.i2s-i2s-hifi i2s-hifi-0 (hw:1,0)
13:20:07: Debug: Device: 2: sysdefault
13:20:07: Debug: Device: 3: lavrate
13:20:07: Debug: Device: 4: samplerate
13:20:07: Debug: Device: 5: speexrate
13:20:07: Debug: Device: 6: pulse
13:20:07: Debug: Device: 7: speex
13:20:07: Debug: Device: 8: upmix
13:20:07: Debug: Device: 9: vdownmix
13:20:07: Debug: Device: 10: dmix
13:20:07: Debug: Device: 11: default
13:20:07: Debug: Device: 0: SPDIF: ff030000.spdif-dit-hifi dit-hifi-0 (hw:0,0)
13:20:07: Debug: Device: 1: HDMI: ff000000.i2s-i2s-hifi i2s-hifi-0 (hw:1,0)
13:20:07: Debug: Device: 2: sysdefault
13:20:07: Debug: Device: 3: lavrate
13:20:07: Debug: Device: 4: samplerate
13:20:07: Debug: Device: 5: speexrate
13:20:07: Debug: Device: 6: pulse
13:20:07: Debug: Device: 7: speex
13:20:07: Debug: Device: 8: upmix
13:20:07: Debug: Device: 9: vdownmix
13:20:07: Debug: Device: 10: dmix
13:20:07: Debug: Device: 11: default
13:20:07: Debug: Device: 0: SPDIF: ff030000.spdif-dit-hifi dit-hifi-0 (hw:0,0)
13:20:07: Debug: Device: 1: HDMI: ff000000.i2s-i2s-hifi i2s-hifi-0 (hw:1,0)
13:20:07: Debug: Device: 2: sysdefault
13:20:07: Debug: Device: 3: lavrate
13:20:07: Debug: Device: 4: samplerate
13:20:07: Debug: Device: 5: speexrate
13:20:07: Debug: Device: 6: pulse
13:20:07: Debug: Device: 7: speex
13:20:07: Debug: Device: 8: upmix
13:20:07: Debug: Device: 9: vdownmix
13:20:07: Debug: Device: 10: dmix
13:20:07: Debug: Device: 11: default
13:20:07: Segmentation fault.

Content of .opencpn/opencpn.log:

13:20:10.199    INFO logger.cpp:130 ------- OpenCPN version 5.8.4 restarted at 2023-10-08 -------

13:20:10.199    INFO logger.cpp:130 Using loglevel INFO
13:20:10.199 MESSAGE ocpn_app.cpp:1234 wxWidgets version: wxWidgets 3.2.2 Linux 64 bit wxGTK
13:20:10.215 MESSAGE ocpn_app.cpp:1239 OS Version reports as:  6.1
13:20:10.216 MESSAGE ocpn_app.cpp:1241 MemoryStatus:  mem_total: 3973 mb,  mem_initial: 617 mb
13:20:10.216 MESSAGE ocpn_app.cpp:1253 OCPN_OSDetail:  arm64 ; Debian GNU/Linux ; 12 ; debian ;
13:20:10.216 MESSAGE ocpn_app.cpp:1260 SData_Locn is /usr/share/opencpn/
13:20:10.216 MESSAGE ocpn_app.cpp:1275 PrivateDataDir is /home/user/.opencpn
13:20:10.217 MESSAGE ocpn_app.cpp:1364 Using existing Config_File: /home/user/.opencpn/opencpn.conf
13:20:10.232 MESSAGE navutil.cpp:1263 winPluginDir, read from ini file:
13:20:10.232 MESSAGE navutil.cpp:1347 Setting Ownship Lat/Lon 33.358, -79.282
13:20:10.233 MESSAGE navutil.cpp:3213 Switch InlandEcdis mode Off
13:20:10.249 MESSAGE styles.cpp:841 Styles loading from /usr/share/opencpn/uidata/styles.xml
13:20:10.261 MESSAGE styles.cpp:818 No styles found at: /home/user/
13:20:10.262 MESSAGE styles.cpp:818 No styles found at: /home/user/.opencpn/
13:20:10.310 MESSAGE styles.cpp:775 Using SVG Icons
13:20:10.310 MESSAGE ocpn_app.cpp:1425 Detected display size (horizontal): 508 mm
13:20:10.311 MESSAGE OCPNPlatform.cpp:925 Getting DefaultSystemLocale...
13:20:10.312 MESSAGE ocpn_app.cpp:1461 System default Language:  en_GB
13:20:10.312 MESSAGE ocpn_app.cpp:1464 Config file language:
13:20:10.312 MESSAGE ocpn_app.cpp:1469 Adjusted App language:
13:20:10.312 MESSAGE OCPNPlatform.cpp:1006 ChangeLocale: Language load for:
13:20:10.312 MESSAGE OCPNPlatform.cpp:1047 ChangeLocale: Fall back to en_US
13:20:10.313 MESSAGE ocpn_app.cpp:1476 Opencpn language set to:
13:20:10.313 MESSAGE ConfigMgr.cpp:624 Loading Configs catalog: /home/user/.opencpn/Configs/configs.xml
13:20:10.317 MESSAGE ocpn_app.cpp:1712 Creating MyFrame...size(1344, 737)  position(0, 27)
13:20:10.319   ERROR file.cpp:216 can't open file '/home/user/.opencpn/mmsitoname.csv' (error 2: No such file or directory)
13:20:10.320 MESSAGE ocpn_frame.cpp:839  **** Connect stuff
13:20:10.330 MESSAGE OCPNPlatform.cpp:823 Starting OpenGL test...
13:20:12.148 MESSAGE OCPNPlatform.cpp:829 OpenGL test complete.
13:20:12.148 MESSAGE OCPNPlatform.cpp:850 OpenGL determined CAPABLE.

leamas, 2023-10-08:
I'll take a look at this, late today or tomorrow.

nohal, 2023-10-08:
The crash very likely has nothing at all to do with sound though. Looking at the log it seems more likely to be the usual cause of all evil, the OpenGL driver. Please edit `~/.opencpn/opencpn.conf` and set `DisableOpenGL=1`

as365n4, 2023-10-08:
Thank you @nohal this soleved the issue with imediately crashing, but it when I look at my Bash output it still tries to open Jack which is not present on a standard Bookworm install.
Anyway, closed the issue as oOpenCPN is working now.
Thank you for your quick help :-)

closed 2023-10-08 by as365n4

leamas, 2023-10-08:
@as365n4: Running without OpenGL means quite a performance hit in some use cases. If this becomes a problem for you you might want to try the Flatpak version, it might perhaps work also with OpenGL enabled.

as365n4, 2023-10-08:
Yeah, disabeling OpenGL is an issue.
MBTiles requires OpenGL so I set `SoftwareGL=1` and set `DisableOpenGL=0` which works too.

I can't do anything about the OpenGL Hardware acceleration as this depends on the Mesa/Lima driver  in  Debian. 
My SBC has an Rockchip RK3328 CPU and supports in Hardware: OpenGL ES 1.1 or 2.0, OpenVG1.1 as it has a Mail-450 GPU.

leamas, 2023-10-08:
so, a try using flatpak should be your first stop.

as365n4, 2023-10-08:
SoftwareGL works for my use case just fine, it enables the use of MBTiles and I can spare the 10% higher CPU load.
As this machine sole purpose is to display a map with all the vessels my AIS receiver receives and feeds to MarineTraffic.
Also I want to keep the setup as simple as possible as I remotly manage this machine and for that I prefer an apt only managed environment. :-)

Hakansv, 2023-10-08:
For the RPi I've found the software OpenGL gives better performance than the hardware (when working). It seems the software GL is better distributed by the four processors avoiding overload on one.