OpenCPN/OpenCPN #3455: RFE: Please add extra driver attributes

antipole2, 2023-10-11:
**Is your feature request related to a problem? Please describe.**
It is sometimes difficult to identify a required driver.
Driver attributes vary by connection type but typically include the protocol, the netPort and netAddress.
Sometimes these may change - for example if the user reconfigures the Connections.

**Describe the solution you'd like**
Driver attributes to additionally include the Comment field.
This would allow a plugin to identify the required handle by the comment which could be used to identify what is on that connection (say "Actisense W2K-1").

**Describe alternatives you've considered**
None

**Additional context**
Any other attributes useful while we are in there?

antipole2, 2024-03-07:
Please can we include connection direction  - receive, transmit or both.
This is so a plugin can check it is not attempting to write to a receive-only connection, etc.

antipole2, 2024-07-02:
Not addressed in Beta 2

bdbcat, 2024-07-17:
Added new attributes.
"userComment"
"ioDirection"    ("IN", "OUT", or  "IN/OUT")

antipole2, 2024-07-22:
Looks good in v5.9B3 - thank you

closed 2024-07-22 by antipole2