OpenCPN/OpenCPN #3456: Bump wxWidgets dependencies to 3.2.3

nohal, 2023-10-13:
https://wxwidgets.org/news/2023/10/wxwidgets-3.2.3-released/

We need to rebuild the various dependency bundles and shift the dependencies used in the builds where it is appropriate.

Hakansv, 2023-10-13:
I've tested on my local repo using VS2022. No obstacles nor detected runtime differences.

leamas, 2023-10-13:
Filed https://github.com/Rasbats/shipdriver_pi/issues/564 for shipdriver update. The 3.2.3  release is officially ABI stable, so this should not be that problematic. 

@bdbcat Assuming we keep the build flags as they are...

leamas, 2023-10-13:
@nohal : Is this the point we should make sure there is a macos m2/aarch64  wxwidgets bundle as well?

transmitterdan, 2023-10-13:
I tested building/debugging in Windows using winConfig.bat and found no issues.  Submitted PR https://github.com/OpenCPN/OpenCPN/pull/3457 to use wx 3.2.3 version for local building.

transmitterdan, 2023-10-13:
Today I was able to crash O with new wxWidgets. It may not be exactly the new wx causing it. During an "onSize" event O thought the canvas width was zero (probably because of new Windows 11 auto sizing feature). Trying to create a zero width bitmap is now apparently a failure in wX3.2 rather than a warning so wx hits a wall (at least in debug mode).

Hakansv, 2023-10-13:
re: 

> probably because of new Windows 11 auto sizing feature

I can't reproduce anything similar on Win 10. Or do you have an example how to force it?

transmitterdan, 2023-10-13:
> re:
> 
> > probably because of new Windows 11 auto sizing feature
> 
> I can't reproduce anything similar on Win 10. Or do you have an example how to force it?

In Windows 11 Insider Build there are all these new auto-arrange possibilities for multiple apps.  If you move the drag bar of one app toward a corner or edge it gives you options for how to arrange multiple apps on the screen.  Somehow, during that process some OnSize event got generated for O that had zero width while the windowing system is figuring out how to size some of the other windows.  Maybe it's a Windows Insider bug.  Still testing...

But we have had zero size bitmap warnings for long time in other scenarios.

nohal, 2023-10-18:
I've bumped the Windows and macOS stuff used in the CI builds to 3.2.3, which leaves only Flatpak to be updated as far as I can tell.

leamas, 2023-10-18:
ok, I'll look into that tomorrow.

leamas, 2023-10-18:
Can't sleep. The flatpak fix is in the branch flatpak-323, ready to merge. I have done some basic smoke tests without problems.

closed 2023-10-19 by rgleason