OpenCPN/OpenCPN #3459: OpenCPN Freezes for a few minutes

Ray22113, 2023-10-13:
**Describe the bug**
I've been experiencing an issue where OpenCPN will freeze and not accept any input from keyboard or mouse.  It will eventual start working again after a few minutes.  Here's what terminal shows when I run it from the terminal:
Query PTR _companion-link._tcp.local.
Query PTR lb._dns-sd._udp.local.
Query PTR _airplay._tcp.local.
Query PTR _raop._tcp.local.
Query PTR _sleep-proxy._udp.local.
Query PTR _companion-link._tcp.local.
Query PTR lb._dns-sd._udp.local.
Query PTR _airplay._tcp.local.
Query PTR _raop._tcp.local.
Query PTR _sleep-proxy._udp.local.
Query PTR _victron-car-charger._tcp.local.
Query PTR _companion-link._tcp.local.
Query PTR lb._dns-sd._udp.local.
Query PTR _airplay._tcp.local.
Query PTR _raop._tcp.local.
Query PTR _sleep-proxy._udp.local.
Query PTR _companion-link._tcp.local.
Query PTR lb._dns-sd._udp.local.
Query PTR _airplay._tcp.local.
Query PTR _raop._tcp.local.
Query PTR _sleep-proxy._udp.local.
Query PTR _companion-link._tcp.local.
Query PTR _sleep-proxy._udp.local.
Query PTR _airplay._tcp.local.
Query PTR _raop._tcp.local.
Query PTR _companion-link._tcp.local.
Query PTR _sleep-proxy._udp.local.
Query PTR _airplay._tcp.local.
Query PTR _raop._tcp.local.
Query PTR _companion-link._tcp.local.
Query PTR lb._dns-sd._udp.local.
Query PTR _airplay._tcp.local.
Query PTR _raop._tcp.local.
Query PTR _sleep-proxy._udp.local.
Query ANY Holly’s iPad._rdlink._tcp.local.
Query ANY Hollys-iPad.local.
Query ANY Holly’s iPad._rdlink._tcp.local.
Query ANY Hollys-iPad.local.
Query ANY Holly’s iPad._rdlink._tcp.local.
Query ANY Hollys-iPad.local.
Query ANY Holly’s iPad._rdlink._tcp.local.
Query ANY Hollys-iPad.local.
Query ANY Holly’s iPad._rdlink._tcp.local.
Query ANY Hollys-iPad.local.
Query ANY Holly’s iPad._rdlink._tcp.local.
Query ANY Hollys-iPad.local.
Query PTR _airplay._tcp.local.
Query PTR _raop._tcp.local.
Query PTR _airplay._tcp.local.
Query PTR _raop._tcp.local.
Query PTR _airplay._tcp.local.
Query PTR _raop._tcp.local.
Query PTR _airplay._tcp.local.
Query PTR _raop._tcp.local.
Query PTR _companion-link._tcp.local.
Query PTR lb._dns-sd._udp.local.
Query PTR _sleep-proxy._udp.local.
Query ANY iPhone._rdlink._tcp.local.
Query ANY iPhone.local.
Query ANY iPhone._rdlink._tcp.local.
Query ANY iPhone.local.
Query ANY iPhone._rdlink._tcp.local.
Query ANY iPhone.local.
Query ANY iPhone._rdlink._tcp.local.
Query ANY iPhone.local.
Query ANY iPhone._rdlink._tcp.local.
Query ANY iPhone.local.
Query PTR _companion-link._tcp.local.
Query PTR lb._dns-sd._udp.local.
Query PTR _sleep-proxy._udp.local.
Query PTR _companion-link._tcp.local.
Query PTR lb._dns-sd._udp.local.
Query PTR _sleep-proxy._udp.local.
Query PTR _spotify-connect._tcp.local.
Query PTR _spotify-connect._tcp.local.
Query PTR _companion-link._tcp.local.
Query PTR lb._dns-sd._udp.local.
Query PTR _sleep-proxy._udp.local.
Query PTR _companion-link._tcp.local.
Query PTR lb._dns-sd._udp.local.
Query PTR _sleep-proxy._udp.local.
Query ANY iPhone._rdlink._tcp.local.
Query ANY iPhone.local.
Query ANY iPhone._rdlink._tcp.local.
Query ANY iPhone.local.
Query ANY iPhone._rdlink._tcp.local.
Query ANY iPhone.local.
Query ANY iPhone._rdlink._tcp.local.
Query ANY iPhone.local.
Query ANY iPhone._rdlink._tcp.local.
Query ANY iPhone.local.
Query ANY iPhone._rdlink._tcp.local.
Query ANY iPhone.local.
Query PTR _companion-link._tcp.local.
Query PTR lb._dns-sd._udp.local.
Query PTR _sleep-proxy._udp.local.
Query PTR _companion-link._tcp.local.
Query PTR lb._dns-sd._udp.local.
Query PTR _sleep-proxy._udp.local.
Query PTR _victron-car-charger._tcp.local.

**Desktop (please complete the following information if applicable):**
Debian 12.2

**Additional context**
Why is it scanning the network like that?

leamas, 2023-10-14:
The scanning is about discovering other opencpn instances on the same network. It is the underpinnings for functions like  "Send to Peer" which is available in the route context menu.

The logging here should be considered a bug, it should not be visible in a regular run.

I have also experienced this "opencpn does not respond" occasionally.  It would be great if we could identify a way standardized way to trigger  this error. However, For the time being I see no relation to the peer discovery and related logging.

leamas, 2023-10-14:
btw: which platform are you on? OS? Hardware?

Ray22113, 2023-10-14:
This is on a Dell XPS 13 9360. i5, 16gb

This is the only install I’m having this issue on. No problems on an AMD NUC.

leamas, 2023-10-14:
hm, interesting. I'm occasionally on a XPS 13 9370, for example right now. On Fedora, though.

Now, the interesting part: Can identify some steps which reliably triggers this error?

Ray22113, 2023-10-14:
I’ll try some things to see if I can reproduce it. One thing that I think I notice is when I have another app full screen and send OpenCPN to the background, it will do this when I switch back to OpenCPN. 

I’ll do some tests on battery and when plugged in. I’m thinking it could have something to do with power management.

leamas, 2024-04-26:
@ghost: Ping?

bdbcat, 2025-12-04:
Closing, OBE.

closed 2025-12-04 by bdbcat