OpenCPN/OpenCPN #3465: Flatpak build incompatible with SSE3 mipmat optimizations

nohal, 2023-10-18:
SSE3 fix from https://github.com/OpenCPN/OpenCPN/pull/3428 which is generally viable is incompatible with the flatpak build environment:

https://app.circleci.com/pipelines/github/OpenCPN/OpenCPN/1804/workflows/1a1a2668-8abf-44cb-bf9e-5a8b3a575207/jobs/6260

```
/usr/lib/gcc/x86_64-unknown-linux-gnu/12.2.0/../../../../x86_64-unknown-linux-gnu/bin/ld: CMakeFiles/opencpn.dir/src/plugin_handler.cpp.o: in function `PluginHandler::installPlugin(PluginMetadata)':
/run/build/opencpn/src/plugin_handler.cpp:1104: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
/usr/lib/gcc/x86_64-unknown-linux-gnu/12.2.0/../../../../x86_64-unknown-linux-gnu/bin/ld: libs/mipmap/libMIPMAP.a(mipmap.c.o): in function `MipMap_ResolveRoutines':
/run/build/opencpn/libs/mipmap/src/mipmap.c:148: undefined reference to `MipMap_24_ssse3'
/usr/lib/gcc/x86_64-unknown-linux-gnu/12.2.0/../../../../x86_64-unknown-linux-gnu/bin/ld: /run/build/opencpn/libs/mipmap/src/mipmap.c:157: undefined reference to `MipMap_32_avx2'
```

bdbcat, 2023-10-18:
Which is probably why it was removed and forgotten...

nohal, 2023-10-18:
Not only flatpak, the SSSE3 detection is actually (still) broken everywhere, it is just visible after merging the PR as the code actually is being compiled.