OpenCPN/OpenCPN #3475: Linker error due to commit 1a0f0c05c3e32d93b5b265b9e0d186c56ba1a8e1

RooieDirk, 2023-10-22:
**Describe the bug**
Linker error due to commit:
commit 1a0f0c05c3e32d93b5b265b9e0d186c56ba1a8e1
Merge: 90d38f884 fb1682337
Author: Pavel Kalian <nohal@users.noreply.github.com>
Date:   Tue Oct 17 22:33:43 2023 -0300

    Merge pull request #3428 from seandepagnier/fix_ssse3
    
    fix optimized mipmap support using ssse3

**To Reproduce**
Steps to reproduce the behavior:
revert git before this commit and compiling and linking goes as usual

**Expected behavior**
make shoul finnish without errors

**Screenshots**
linker output:
/home/dirk/Development/OpenCPN/src/plugin_handler.cpp:1104: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
/usr/bin/ld: libs/mipmap/libMIPMAP.a(mipmap.c.o): in function `MipMap_ResolveRoutines':
/home/dirk/Development/OpenCPN/libs/mipmap/src/mipmap.c:157: undefined reference to `MipMap_32_avx2'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/opencpn.dir/build.make:2751: opencpn] Error 1
make[2]: Leaving directory '/home/dirk/Development/OpenCPN/build'
make[1]: *** [CMakeFiles/Makefile2:756: CMakeFiles/opencpn.dir/all] Error 2
make[1]: Leaving directory '/home/dirk/Devel

**Desktop (please complete the following information if applicable):**
 - OS: linux
 - Version [ubuntu 22-4
 -Kernel Version: 5.15.0-84-generic (64-bit)
Graphics Platform: X11
Processors: 8 × AMD Ryzen 7 4700U with Radeon Graphics
Memory: 15.0 GiB of RAM
Graphics Processor: RENOIR

**Smartphone/Tablet (please complete the following information if applicable):**

**Additional context**

nohal, 2023-10-22:
Please clean your build directory and in case the error persists, show me the full output of `cmake ..`

RooieDirk, 2023-10-22:
Shit... Should have thought of it by myself. Sorry me bad.

closed 2023-10-22 by RooieDirk