OpenCPN/OpenCPN #3478: PIM: Fails to detect missing metadata.xml

transmitterdan, 2023-10-26:
**Describe the bug**
Plugin-Manager fails to detect missing metadata.xml file in tarball

**To Reproduce**
Steps to reproduce the behavior:
1. Create tarball without metadata.xml file
2. Install tarball
3. Crash

**Expected behavior**
PIM should report missing metadata.xml file

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Version 11

**Additional context**
The PluginHandler::extractTarball() function ignore the 'only_metadata' flag.  Probably the 'only_metadata' flag should be removed as an argument as it isn't used by any caller.  And the extractTarball() function should return false if the metadata.xml file was never able to be extracted.

leamas, 2023-10-26:
It's on my  desk, but not top priority.

rgleason, 2023-10-26:
I have found that when building TP locally and "Import Plugin" without having run the cloudsmith_download.sh script locally, that there is a clear message "missing metadata.xml in tarball" or something like that.

transmitterdan, 2023-10-26:
It should be enough to just delete metadata.xml from an existing tarball.  When I get real internet I can upload the errant tarball.

leamas, 2023-10-26:
Top of stack trace:
```
0  0x00007ffff5552276 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () at /lib64/libstdc++.so.6
#1  0x0000000000a6dc31 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (__str=<error reading variable: Cannot access memory at address 0x8>, this=0x7fffffffb560) at /usr/include/c++/13/bits/basic_string.h:1596
#2  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator=(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (__str=<error reading variable: Cannot access memory at address 0x8>, this=this@entry=0x7fffffffb560) at /usr/include/c++/13/bits/basic_string.h:802
#3  PluginMetadata::operator=(PluginMetadata const&) (this=this@entry=0x7fffffffb560) at /home/mk/src/OpenCPN/include/catalog_parser.h:44
#4  0x0000000000a7a486 in PluginHandler::ExtractMetadata(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, PluginMetadata&)
    (this=this@entry=0x4ffaa60, path="/home/mk/Calculator-3.0.0+2305141810.f79e9690_fedora-38-x86_64.tar.gz", metadata=...) at /usr/include/c++/13/bits
```

leamas, 2023-10-26:
Fedora testcase
[Calculator-3.0.0+2305141810.f79e9690_fedora-38-x86_64.tar.gz](https://github.com/OpenCPN/OpenCPN/files/13178400/Calculator-3.0.0%2B2305141810.f79e9690_fedora-38-x86_64.tar.gz)

bdbcat, 2024-01-04:
Assuming that an imported tarball without metadata is unusable, a simple GUI message ("Corrupt tarball, missing metadata") should suffice?

leamas, 2024-01-04:
Yes.

closed 2024-01-04 by bdbcat