OpenCPN/OpenCPN #3479: AIS on Android issue

rafalZad, 2023-10-27:
**Describe the bug**
When using AIS UART cable on Android device (Lenovo P11 tab), its got delay in data transmision after few minutes. It works on all data transfered by cable - position, hdg/cog, all AIS targets. Data is good quality, no interruptuon, yet delay is increesing and i can see "past situatuon" on screen. Delay is also on Android phone (Samsung Xcover Pro).

**To Reproduce**
Just plug cable and navigate.

**Expected behavior**
Real life view without delay

**Smartphone/Tablet (please complete the following information if applicable):**
 Lenovo p11 tablet (Android 12)
 Samsung Galaxy Xcover Pro (Android 13)

**Additional context**
Cable works fine with pc and windows. It feels like something is "blocking" data from beeing displayed on proper moment. refresh rate of the screen is 60hz. Is the OpenCPN working on 30hz?

leamas, 2023-10-27:
hm... What's the speed on that cable?

rafalZad, 2023-10-27:
> hm... What's the speed on that cable?
> 
> EDIT: That is, could it be related to #3391?

Cable is workig only on 38400. All the same setup on android and windows - on windows no problem, on android building up delay.

Doesnt look like similar bug like 3391. Here i have constant, building up delay of situation.

leamas, 2023-10-27:
Indeed. However, the tentative fix described there should work here as well, assuming that this is indeed  about opencpn not being able to process data in the same speed as it's entered. 

So, while the symptoms are different this might very well be the same root cause. **might**, that is.

rafalZad, 2023-10-27:
I will test by filtering NMEA sentences. 
Interresting observaton:
Ive selected priority between build-in gps and AIS cable (ais have external gps connected to it) and own position priority was from buold-in gps, yet HDG, COG, and all AIS target from cable. 
Position on the chart was displayed properly - without delay, yet all other info (including targets) was delayed. I did 180 degrees turn, and vectors showed it after minutes, while position was real-time.

rafalZad, 2023-11-19:
I tried to cut unnececary messages. Left only AIS related + GPS data. Problem still exists. Looks like no one have solution to this issue.

bdbcat, 2023-11-19:
I have an understanding of the root cause for this issue.  It is in the data buffer management, up on the Java side of the app.
On my long list of bugs to address, as resources (time) allows.
Thanks for your patience.

rafalZad, 2023-11-19:
That means problem is known and identified! that is great news! Awaiting for solution soon:) brgds:)

bdbcat, 2023-11-28:
rafalZad....
OpenCPN for Android V5.8.4{105} is now available from the Playstore.
Please test and confirm the correction of this bug.
Thanks

rafalZad, 2023-11-28:
> rafalZad.... OpenCPN for Android V5.8.4{105} is now available from the Playstore. Please test and confirm the correction of this bug. Thanks

I am testing it now. Will let you know within few days:) thank you:)

rafalZad, 2023-11-29:
I am happy to inform, that issue is fixed and all is working fine. Thank you for support!!

bdbcat, 2023-11-29:
That is good to hear.
If you are satisfied, please close this issue.
Thanks

closed 2023-11-29 by rafalZad