OpenCPN/OpenCPN #3480: RFE: Noaa MBtile Charts on high res displays

rgleason, 2023-10-28:
**Is your feature request related to a problem? Please describe.**
Noaa MBtiles appear to be a good way to use charts, however the text size of place names is far too small and does not seem to be affected by normal controls.  Additionally the text completely disappears as you zoom in, rendering the use of this format unacceptable for high res displays.

**Describe the solution you'd like**
Better control of text size.

**Describe alternatives you've considered**
Using ENC Charts instead.

**Additional context**
Add any other context or screenshots about the feature request here.

bdbcat, 2023-10-28:
Not practical, or really even possible.
MBtiles are raster images stored by geo region in a database.  The images are direct extractions from RNCs.
There is no other metadata available, such as Feature names, or even Unit of Measure for soundings.

closed 2023-10-28 by bdbcat

rgleason, 2023-10-28:
Thanks for explaining. I will remove these from the computer.