OpenCPN/OpenCPN #3481: RFE: Tidal Current Prediction display & controls

rgleason, 2023-10-28:
**Is your feature request related to a problem? Please describe.**
Please refer to David Burch's "ENC Essentials" booklet, page 43 showing qtVlm's current menu which is similar to OpenCPN but offers more information and selections:

- top dropdown offers data from other depths at that station.  (not in OpenCPN)
- yellow bar marks span of daylight with the dark parts being twilight (not in OpenCPN)
- values at present time are marked on the chart.  (yes)
- time selector offers UTC or local time, plus station time (which is valuable for planning a route in a different time zone). (missing Station Time)
- calendar button allows selection of any date in the past or future. (missing)
- Similar display for harmonic tide predictions.   (Similar comments for Tide display)

sebastien-rosset, 2024-07-08:
For the time zone configuration, it would be useful to have a global toggle that makes it possible to configure the display time, rather than each base component or plugin exposing a knob. Sometimes the date/time is displayed without timezone information.

1. Grib data table
2. Grib time
3. Weather routing (has option to select UTC or local time)
4. Deviation table
5. Celestial navigation
6. tide (has option to select UTC or station time)

sebastien-rosset, 2024-07-08:
How about if there was a global setting to control the date and time? Widgets that currently display time would have to be enhanced to read the date/time global settings, plus local override if useful (such as for tides based on local station).

[image]

leamas, 2024-07-08:
>  Widgets that currently display time would have to be enhanced to read the date/time global settings [...]

This is probably the wrong way to do it, there are problems if the user updates the central configuration. In order to detect this  plugins will need to poll this central configuration which is evil for obvious reasons.

A better approach is probably to think in terms of messages. Core opencpn sends a "config updated" message as part of startup and as soon as anything in the configuration is changed.  Plugins just listens to these messages and updates the internal configuration according to message contents.

sebastien-rosset, 2024-07-08:
Yes, I agree. I meant read the global settings when the widget is created and listen for changes, similar to what is done elsewhere in the code. I was wondering if conceptually having a global settings would even be considered.

rgleason, 2024-07-09:
Sebstein, this would be a big improvement if done properly I thnk.
It will help with planning and data display.