OpenCPN/OpenCPN #3483: Reviewed and clean up Surfpro3 ENV and organized it.

rgleason, 2023-10-29:
Reviewed and clean up Surfpro3 ENV and organized it.
Under source/repos/opencpn Decided to git pull origin master on both laptop and SurfacePro3. Then ran  from VS2022 command prompt  .\buildwin\winConfig.bat --clean and then  .\buildwin\winConfig.bat  

The laptop looks like it built debug, release and debinforelease and right now it is hung up and blinking on chartimg.cpp.

The surfpro3 still does not find wxWidgets.  Here is the set output. I have checked ENV and it seems fine. No wxWidgets refs, many microsoft that I don't get.
[ENV-set.txt](https://github.com/OpenCPN/OpenCPN/files/13196975/ENV-set.txt)

[ENV-set-separated-into-lines.txt](https://github.com/OpenCPN/OpenCPN/files/13196981/ENV-set-separated-into-lines.txt)

_Originally posted by @rgleason in https://github.com/OpenCPN/OpenCPN/discussions/2852#discussioncomment-7413664_

rgleason, 2023-10-29:
Two computers:

1. Laptop - Dell 9700 17" Windows 11 after git pull origin master, and executing the batch file .\buildwin\winConfig.bat it built and completed and runs VS2022 in C:\users\fcgle\source\repos\OpenCPN
2. Microsoft Surface Pro 3 Windows 10 tablet, after git pull origin master, and executing the batch file .\buildwin\winConfig.bat it fails and the reports are in the post above. This is located on D:\Frederick\source\Repos\Opencpn because the tablet has a small SSD and I have to use the microSD for development.

rgleason, 2023-10-29:
Just ran it again.

```
-- Found OpenGL....
--     GL Lib: opengl32glu32
--     GL Include:
--     Revised GL Lib: opengl32
-- Could NOT find wxWidgets (missing: gl net xml html adv aui core base webview)
-- Could not find wxWidgets webview add-on
-- Could NOT find wxWidgets (missing: gl net xml html adv aui core base)
CMake Error at CMakeLists.txt:762 (message):
  Cannot find wxWidgets, giving up.  On linux you might want to try using
  -DOCPN_FORCE_GTK3.
```

Yet I find:
D:\Frederick\source\Repos\OpenCPN\cache\wxWidgets-3.2.2
D:\Frederick\source\Repos\OpenCPN\cache\buildwxWidgets
D:\Frederick\source\Repos\OpenCPN\cache\buildwxWidgets\lib\vc_dll

Should the wxWidgets-3.2.2 be removed?

rgleason, 2023-10-29:
The laptop has these folders in the cache 

- buildwxWidgets
- wxWidgets-3.2.1
- wxWidgets-3.2.2
- buildwin
- buildwintemp
- and file OCPNWindowsCoreBuildSupport.zip  19mb
- file wx-config.bat

The SurfacePro3 tablet has much the same on the D:drive but with no wxWidgets-3.2.1

rgleason, 2023-10-29:
Also is there any reason to keep VS 2017 if I have VS 2022?

rgleason, 2023-10-29:
Have run "choco update all" on both machines.  A number of programs were updated.   choco --version shows v0.11.2 for the SurfacePro3 and 2.2.2 for the laptop Dell 9700 17"  

I can't seem to get the SurfacePro to update to 2.2.2.

rgleason, 2023-10-29:
configDev.bat now has this

```
set "wxDIR=D:\Frederick\source\Repos\OpenCPN\cache\buildwxWidgets" 
set "wxWidgets_ROOfT_DIR=D:\Frederick\source\Repos\OpenCPN\cache\buildwxWidgets" 
set "wxWidgets_LIB_DIR=D:\Frederick\source\Repos\OpenCPN\cache\buildwxWidgets\lib\vc_dll" 
set "VCver=17" 
set "VCstr=Visual Studio 17" 
```

transmitterdan, 2023-10-29:
> Just ran it again.
>
> ```
> -- Found OpenGL....
> --     GL Lib: opengl32glu32
> --     GL Include:
> --     Revised GL Lib: opengl32
> -- Could NOT find wxWidgets (missing: gl net xml html adv aui core base [...]

Rick,

I don't think you need to delete anything.  Try:

buildwin\winconfig.bat --clean

transmitterdan, 2023-10-29:
> configDev.bat now has this
> 
> ```
> set "wxDIR=D:\Frederick\source\Repos\OpenCPN\cache\buildwxWidgets" 
> set "wxWidgets_ROOfT_DIR=D:\Frederick\source\Repos\OpenCPN\cache\buildwxWidgets" 
> set "wxWidgets_LIB_DIR=D:\Frederick\source\Repos\OpenCPN\cache\buildwxWidgets\lib\vc_dll" 
> set "VCver=17" 
> set "VCstr=Visual Studio 17" 
> ```
> 
> The laptop has similar, basically identical!

Rick,

Don't worry about what is in confdev.bat.  That is not the problem.  If it winConfig.bat cannot find wxWidgets it is something else.  You can safely delete configdev.bat if you have any doubt.

rgleason, 2023-10-29:
OK Thanks, I am trying 
.\buildwin\winConfig.bat --clean and then "y" to removing programs, since I had tried just --clean before and it did not work.

rgleason, 2023-10-29:
It seems to be working now. Just downloaded wxWidgets-3.2.3 expanded it and copied to buildwxWidgets
Its slow, but it is working I think.
Thanks. 
The laptop same setup has only wxWidgets 3.2.2  so I think I have to do  .\buildwin\winConfig.bat --clean and then pick "Yes"

bdbcat, 2023-10-29:
This looks suspicious:

`set "wxWidgets_ROOfT_DIR=
`
What would "RoofT" reference?

rgleason, 2023-10-29:
Yes, something to keep in mind, It should be Root, of course, but Dan says I can delete configDev.bat so that probably is not used. The build seems to be progressing processing all the files...now doing *.cpp.  It will take several hours probably.

rgleason, 2023-10-29:
I guess I was a little too optimistic. For some reason I wonder if there is some write problem with this microSD chip.
... Yep, the chip isn't being read now.
[failed-in-processing.txt](https://github.com/OpenCPN/OpenCPN/files/13199054/failed-in-processing.txt)

rgleason, 2023-10-29:
Reseated Samsung EVO microSD card  128gb just 50gb used so far.
Then successfully ran winConfig.bat --clean and it looked like a the wxWidgets and NSIS were built, but it failed.  Then I tried just running winConfig.bat again to save time.

-- *** Package will include documentation ***
-- Found OpenGL....
--     GL Lib: opengl32glu32
--     GL Include:
--     Revised GL Lib: opengl32
-- Could NOT find wxWidgets (missing: gl net xml html adv aui core base webview)
-- Could not find wxWidgets webview add-on
-- Could NOT find wxWidgets (missing: gl net xml html adv aui core base)
CMake Error at CMakeLists.txt:762 (message):
  Cannot find wxWidgets, giving up.  On linux you might want to try using
  -DOCPN_FORCE_GTK3.

```
-- Configuring incomplete, errors occurred!
See also "D:/Frederick/source/Repos/OpenCPN/build/CMakeFiles/CMakeOutput.log".
CMake failed to configure OpenCPN build folder.
Review the error messages and read the OpenCPN
Developer Manual for help.
Press any key to continue . . .
```

rgleason, 2023-10-29:
Output from the "set" command shows:

```
wxDIR=D:\Frederick\source\Repos\OpenCPN\cache\buildwxWidgets
wxWidgets_LIB_DIR=D:\Frederick\source\Repos\OpenCPN\cache\buildwxWidgets\lib\vc_dll
wxWidgets_ROOT_DIR=D:\Frederick\source\Repos\OpenCPN\cache\buildwxWidgets
```

rgleason, 2023-10-29:
[image]

transmitterdan, 2023-10-31:
Rick,

The winConfig.bat file should be creating log files in the OpenCPN/build folder.  Look there and see if there are any .log files. If there are files that begin MSBUILD_blah_blah.log please attach them here so I can see what might be going on. (MSBUILD_blah_blah.log may be, for example, MSBUILD_Debug_WIN32_Debug.log or other things that look similar).

rgleason, 2023-10-31:
I don't seem to have MSBUILD_blah_blau.log files in the build directory.  The CMakeCasche.txt is the closest I can get to a report. Many of the directories are empty. Debug, RelwithDebInfo, Release and MinSizRel  folders are empty except for the empty plugins folder.

[CMakeCache.txt](https://github.com/OpenCPN/OpenCPN/files/13210969/CMakeCache.txt)

I am going to remove the file here and try it again. I think perhaps that is the issue!
I will try running the batch file again.

rgleason, 2023-10-31:
Same error wxWidgets not found, but CMakeCache.txt has wxWidgets info, and it just missing the path to the executable.

```
//Cleared.
WX_xrcd:FILEPATH=WX_xrcd-NOTFOUND

//Cleared.
WX_zlib:FILEPATH=D:/Frederick/source/Repos/OpenCPN/cache/buildwxWidgets/lib/vc_dll/wxzlib.lib

//Cleared.
WX_zlibd:FILEPATH=D:/Frederick/source/Repos/OpenCPN/cache/buildwxWidgets/lib/vc_dll/wxzlibd.lib

//Set wxWidgets configuration (mswud;mswu;mswud;mswu)
wxWidgets_CONFIGURATION:STRING=mswu

//Cleared.
wxWidgets_LIB_DIR:PATH=D:/Frederick/source/Repos/OpenCPN/cache/buildwxWidgets/lib/vc_dll

//wxWidgets base/installation directory
wxWidgets_ROOT_DIR:PATH=D:/Frederick/source/Repos/OpenCPN/cache/buildwxWidgets

//Use release and debug configurations?
wxWidgets_USE_REL_AND_DBG:BOOL=ON

//Location of wxWidgets resource file compiler binary (wxrc)
wxWidgets_wxrc_EXECUTABLE:FILEPATH=wxWidgets_wxrc_EXECUTABLE-NOTFOUND
```

rgleason, 2023-10-31:
The other CMakeCache.txt file on my Laptop which does compile has
```
//Value Computed by CMake
wmm_pi_SOURCE_DIR:STATIC=C:/Users/fcgle/source/repos/OpenCPN/plugins/wmm_pi

//Set wxWidgets configuration (mswud;mswu)
wxWidgets_CONFIGURATION:STRING=mswu

//Path to wxWidgets libraries
wxWidgets_LIB_DIR:PATH=C:/Users/fcgle/source/repos/OpenCPN/cache/buildwxWidgets/lib/vc_dll

//wxWidgets base/installation directory
wxWidgets_ROOT_DIR:PATH=C:/Users/fcgle/source/repos/OpenCPN/cache/buildwxWidgets

//Use release and debug configurations?
wxWidgets_USE_REL_AND_DBG:BOOL=ON

//Location of wxWidgets resource file compiler binary (wxrc)
wxWidgets_wxrc_EXECUTABLE:FILEPATH=C:/Users/fcgle/source/repos/opencpn/wxWidgets-3.2.2/lib/vc14x_dll/wxrc.exe
```

So for the laptop, I find that under the "cache" folders wxWidgets-3.2.2    and wxWidgets-3.2.1  do exist and were found somehow, and wxWidgets-3.2.2  does have "wxrc.exe".
I also found that C:/Users/fcgle/source/repos/OpenCPN/cache/buildwxWidgets
  does have a wxWidgets-3.2.3.zip file  (39mb)  that has NOT been unzipped.

I also find this same file in the SurfacePro3 computer in the same place, also not expanded. 
However C:/Users/fcgle/source/repos/opencpn/wxWidgets-3.2.2  does not exist!!!

rgleason, 2023-10-31:
I am going to expand the wxWidgets-3.2.3.zip and see what happens next time.

rgleason, 2023-10-31:
I searched Nortion Security History and Quarantine and did not find any "wxrc.exe" files being removed.

So the problem must be something else preventing the file from being created?

transmitterdan, 2023-10-31:
Rick,

You are looking in the wrong place.  Try this.  In the winConfig.bat file there is a line around 467 that looks like this:

```cmake -A Win32 -G "%VCstr%" ^```
Edit the file and add the --debug-find option to make it look like this:
```cmake -A Win32 -G "%VCstr%" --debug-find ^```

Then run:
```buildwin\winConfig.bat --clean```
again.  Post the output here.

transmitterdan, 2023-10-31:
You don't need wxrc.exe.  It isn't used for anything.

rgleason, 2023-10-31:
Agreed.  This morning I removed all files from my laptop  source/repos/  directory and git clone and ran the batch files. How simple!

It built properly and VS2022 build and ran Debug and RelwithDebinfo (checked)!
Looking at CMakeCache.txt  I found those lines down around 853

```
//Location of wxWidgets resource file compiler binary (wxrc)
wxWidgets_wxrc_EXECUTABLE:FILEPATH=wxWidgets_wxrc_EXECUTABLE-NOTFOUND
```
So not having wxrc.exe was a red herring.   I believe the entire problem was from old artifacts left over from previous installations and that I should always start with an empty repos directory rather than doing "git pull origin master"!!!

Is that true?  I am going to remove the opencpn installation on SurfacePro3 and reinstall to hopefully get it working.

rgleason, 2023-10-31:
SurfacePro3 Tablet with most current version of Windows 10
has Microsoft.NET Framework 4.5.1 and Microsoft Visual C++ 2008, 2010, 2013, and 2015-2022 Redistributables
Starting from scratch with an empty D: ...\source\repos\  directory
The installation failed again, the problem was "cannot find wxWidgets" again!

rgleason, 2023-10-31:
TransmitterDan:
> You are looking in the wrong place. Try this. In the winConfig.bat file there is a line around 467 that looks like this:
>
> cmake -A Win32 -G "%VCstr%" ^
> Edit the file and add the --debug-find option [...]

After following this above. These are the results.
The command output is in "debug-build-output.txt"  
[debug-build-output.txt](https://github.com/OpenCPN/OpenCPN/files/13221702/debug-build-output.txt)
CmakeOutput.log
[CMakeOutput.log.txt](https://github.com/OpenCPN/OpenCPN/files/13221705/CMakeOutput.log.txt)
CMakeCache.txt
[CMakeCache.txt](https://github.com/OpenCPN/OpenCPN/files/13221706/CMakeCache.txt)
Zipped files from build  (the debug, release, relwithdebinfo, min were all empty and with only empty plugins.
[debug-build.zip](https://github.com/OpenCPN/OpenCPN/files/13221704/debug-build.zip)

transmitterdan, 2023-11-01:
Rick,

I'm pretty sure the problem is your cmake version is way out of date.  Check the version with:
```cmake --version```
Mine is:
```cmake version 3.26.4```

rgleason, 2023-11-01:
I thought I had upgraded it recently....

D:\Frederick\source>cmake --version
cmake version 3.21.3

transmitterdan, 2023-11-01:
Older versions of cmake do not know about the latest version of wxWidgets.

rgleason, 2023-11-01:
Yes, I think that was it!   An old version of CMake. That's another thing needed when updating the configuration.

It has been compiling for hours, now, now doing relandebinfo, etc.

Suggestion for Manual when rebuilding 
1 .Update CMake and the other critical support programs.
2. Start fresh with an empty directory. Delete and clone again.

Thank you, TransmitterDan for your help.  I am going to close this now.

closed 2023-11-01 by rgleason

transmitterdan, 2023-11-01:
Rick,

The instructions don't even tell you to install CMake.  That's because CMake comes with Visual Studio. So, if your Visual Studio installation is up to date then you have the right CMake version.  Visual Studio constantly hounds you to update every time a new release is issued.

rgleason, 2023-11-01:
Ok. Perhaps I missed that. And I will check. What I did on the SurfacePro3 was completely remove VS2017 and then install vs2022. Then I thought i had updated all the other programs, poedit, get..., git, gihub desktop, etc.

Apparently not.

transmitterdan, 2023-11-02:
The instructions say that you only have to install Git and VS2022 (or VS2019).  Select the option to include CMake in the VS2022 installation.  One no longer has to install all the other tools like POEdit and so forth.  The winConfig.bat will do all of that automatically.