OpenCPN/OpenCPN #3485: OpenCPN Plugin Manager displays plugins for multiple OS's

jongough, 2023-10-30:
**Describe the bug**
Try to install Testplugin_pi through plugin manager on Ubuntu Jammy and get presented version for Jammy and darwin-wx32

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Plugin manager
2. Go to settings
3. select new flavour as 'ubuntu-wx32-x86_64:22.04'
4. Click Done
5. Update master catalog
6. Find and select Testplugin_pi for install
7. get presented with ubuntu and darwin versions of the plugin

**Expected behavior**
Only the relevant OS versions of the plugin should be displayed.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Linux mint 20.3 cinnamon - based on Ubuntu desktop LTS 22.04
 - Version [e.g. 22]

**Smartphone/Tablet (please complete the following information if applicable):**
 - N/A

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to illustrate problems, attach them if applicable.

This is using OpenCPN 5.9.0

leamas, 2023-10-30:
This is using the Beta branch, I presume?

bdbcat, 2023-10-30:
I have also seen this, using plugin's master branch, on O5.9 built from O's master branch, on Debian/12, x86.
Quick investigation.
1.  Erroneous matches are always "darwin-wx32" target.
2. Something fishy at plugin_handler.cpp:367

```
  if (plugin.abi() == "darwin-wx32") {
    OCPN_OSDetail *detail = g_BasePlatform->GetOSDetail();
    auto found = metadata.target_arch.find(detail->osd_arch);
    if(found != std::string::npos) {
      rv = true;
    }
  }

```
If the plugin target is "darwin-wx32", and the system plugin arch is the same as system arch, then the plugin is accepted.
What about matching the plugin abi to the host abi, also?

jongough, 2023-10-30:
https://github.com/OpenCPN/OpenCPN/issues/3485#issuecomment-1786131749
This my build of the master branch of OCPN. I use the master circleci buid of testplugin, currently 1.0.276.0, but it does show another version of testplugin which is correct as that one hasn't been removed from the plugins repository. I was only reporting the issue with being presented a macos build as well as a ubuntu build.

leamas, 2023-10-30:
For background this is about 02361fec4c, Pavel's patches for universal macos builds. 

@nohal : thoughts?

bdbcat, 2023-10-30:
Jon...
"the issue with being presented a macos build as well as a ubuntu build"
I acknowledge the bug.  That is the "fishy" point.  Pavel owns it, I think.

nohal, 2023-10-30:
Yes, will send a fix after dinner...

commit 68df2ce7 (2023-10-31): Properly limit the architecture checks for macOS universal binaries to the macOS platform. Fixes #3485
closed 2023-10-31 by nohal