OpenCPN/OpenCPN #3486: Plugin list not properly constrained in size with wxWidgets 3.2.3

nohal, 2023-10-31:
The list of plugins fills the whole parent window instead of just the scrolled window. Reproducible at least on macOS.

**Screenshots**
[image]

leamas, 2023-11-01:
Rebuilding on Debian sid against 3.2.3 does not show exhibit this problem. 

[image]

leamas, 2023-11-01:
hm... rebuilding on Ventura using wxw 3.2.3 as done by ci/generic-build-macos.sh does not exhibit this problem for me:

[image]

leamas, 2023-11-01:
Arg... This is not Ventura, it's Monterey 12.7.1. Sorry, really newbie trying to navigate the MacOS jungle.

Build from current master at 7e0c83621

nohal, 2023-11-19:
Gone with current deps bundle containing wx 3.2.4

closed 2023-11-19 by nohal