OpenCPN/OpenCPN #3487: Macos:ventura : wxWidgets download failure

leamas, 2023-11-01:
**Describe the bug**
On a clean Ventura the script ci/generic-build-macos.sh fails on not finding wxWidgets

**To Reproduce**
- Get hands a clean ventura installation
- Install command libe developmetn tools
- run git clone
- cd  OpenCPN; ci/generic-build-macos.sh

**Expected behavior**
Successful build

**Additional context**

The reason was that curl failed due to the bug described in https://stackoverflow.com/questions/69734654.  IO used the *export CURL_SSL_BACKEND="secure-transport"* work arund described there to handle it.

IMHO,, the script should handle this error one way or another.

leamas, 2023-11-01:
While on it: The script did not install the required brew libpng and pcre2 packages

nohal, 2023-11-01:
Hmmm.... The later comment in the linked SO question says "macOS 12.3 was officially released on march 13th, 2022 and the issue reported on this question has been fixed. Personally I have removed the workaround I had deployed and I suggest others do the same to keep up with the new versions."
Is your Ventura install up to date?

leamas, 2023-11-01:
Away for now. Please note the two missing packages above

nohal, 2023-11-01:
I'm not entirely convinced we need them, both are bundled with wxWidgets as submodules and linked statically. How exactly does the problem manifested itself on your system? They probably will be needed if you build against wxWidgets from Homebrew, but then they will be pulled in as dependencies.

nohal, 2023-11-01:
What we do need to do is to either drop `generic-build-macos.sh` completely or bring it back at the level of `universal-build-macos.sh`, including new wxWidgets bundle updated to 3.2.3. But do we want to?

leamas, 2023-11-01:
I'm not really sure what to do, and a bit unfocused; this is sort of a drive-by issue for me. Furthermore, I honestly don't  understand the alternatives.

That said, the only hard requirement for this script is that if builds from scratch in a CI build. As long as it does, it should also be a working setup for a local build.

The problem with the missing libraries occurred in the fixup-bundle stage which was looking for them and failed if they were  not found. Simply installing  them made the build work.

nohal, 2023-11-19:
I have changed the `generic-build-macos.sh` script to build with only dependencies from Homebrew, which minimizes the voodoo required to get going and makes sure we are not diverging too much from what is available in a "normal" environment on a developer machine. It is used in the GA build, so we will immediately see if it falls apart.
All the production builds on CCI against the prebuilt dependency bundles are done by the `universal-build-macos.sh` script which is now also able to do single-architecture stuff.

leamas, 2023-11-19:
All this looks extremely sound to me.  I understand that this means that we don't support really old versions of MacOS, and that we consider the cleaner build worth it. I certainly have no objections to that.

Could you perhaps take a look at the description in the manual of how to compile MacOS. Is it outdated after these changes?

nohal, 2023-11-19:
We actually support even older macOS versions now (10.13) than we did in 5.8.4 (11.0) while still being ABI compatible for the existing plugins.
I have updated the developer manual to reflect the current state already. Probably good enough for now, until I see someone complain it misses something.

nohal, 2023-12-01:
@leamas do you see some more work to be done here? If not, I don't plan any.

leamas, 2023-12-01:
No. Let's close it.

closed 2023-12-01 by nohal