OpenCPN/OpenCPN #3488: GetN2000Payload(nmea2kId, ev) returning wrong size of vector

antipole2, 2023-11-02:
**Describe the bug**
` GetN2000Payload(nmea2kId, ev)` returns a `std::vector<uint8_t>` payload
The payload size is one too big.
Processing it with 	`for (int i = 0; i < payload.size(); i++){` delivers an extra byte value 085 (0x55)  on the end.

**To Reproduce**
Using the JavaScript plugin v2.2 (presently available from the alpha repository) the following script delivers a sample payload.  It uses the above `for` loop.

```JavaScript
OCPNonNMEA2k(gotIt, 129540);	// listen out for system time

function gotIt(data){
        print(data.payload, "\n");
       }
```
Sample output:
````[147,19,3,16,240,1,255,1,255,255,255,255,8,96,48,196,76,64,208,186,37,85];````
**Expected behavior**
`payload.size()` should return the correct number of bytes

**Desktop (please complete the following information if applicable):**
 - OS: MacOS
 - Version 14.0

leamas, 2023-11-02:
This is most likely related to the driver,  the very function  `GetN2000Payload()` is basically  a oneliner. 

So, which driver is involved in here?

antipole2, 2023-11-02:
> So, which driver is involved in here?
What am I looking for to answer this?  The driver attributes?

leamas, 2023-11-02:
>  The driver attributes?

Yes, in  particular _commPort_

leamas, 2023-11-02:
... and of course _protocol_, sorry

antipole2, 2023-11-02:
These dumps were taken for me on a RaspberryPi.  I need to bring in duichan.
@duichan can you please provide details?
To display the driver attributes, you can run the following JavaScript on the plugin v2.2 that you have.
```javascript
handles = OCPNgetActiveDriverHandles();
for (h = 0; h < handles.length; h++){
	attributes = OCPNgetDriverAttributes(handles[h]);
	print(attributes, "\n");
	}
````

duichan, 2023-11-02:
Script returns: ReferenceError: identifier 'OCPNgetActiveDriverHandles' undefined

antipole2, 2023-11-02:
@duichan Oops... overlooked you need more recent build 9e256d0 for this to work as written.  Please update plugin from [alpha repo. here](https://cloudsmith.io/~opencpn/repos/javascript-alpha/packages/)

duichan, 2023-11-02:
Works now:
nmea2000!@!socketCAN-can0		{"canAddress":"72","canPort":"can0","protocol":"nmea2000"}

leamas, 2023-11-02:
OK, this should t hen be about comm_drv_n2k_socketcan.cpp. Looking intpo that I see very little logging. Are up and running rebuilding main OpenCPN with a patch?

antipole2, 2023-11-03:
>  Are up and running rebuilding main OpenCPN with a patch?
I think this should read "Are you up and running for rebuilding..." ?
I am not in a position to help here - I have no NMEA2000 input at home.  I am depending on @duichan for this.
The script in my original post shows how to test any patched OCPN.

leamas, 2023-11-03:
What we need is  a custom build of OpenCPN with a untested patch like below. Is it feasible?

That is, we need to see what data the driver sees and forwards to the plugin. 
```
diff --git a/src/comm_drv_n2k_socketcan.cpp b/src/comm_drv_n2k_socketcan.cpp
index 4551d63e2..541ec7420 100644
--- a/src/comm_drv_n2k_socketcan.cpp
+++ b/src/comm_drv_n2k_socketcan.cpp
@@ -717,6 +717,12 @@ void Worker::HandleInput(CanFrame frame) {
     auto src_addr = m_parent_driver->GetAddress(m_parent_driver->node_name);
     auto msg = std::make_shared<const Nmea2000Msg>(header.pgn, vec, src_addr);
     auto msg_all = std::make_shared<const Nmea2000Msg>(1, vec, src_addr);
+    if (wxLog::GetLogLevel() >= wxLOG_Debug) {
+      std::stringstream ss;
+      ss << "N2K Socketcan: receiving: ";
+      for (auto& i : vec) ss << std::hex << i << " ";
+      wxLogDebug(ss.str().c_str());
+    }
 
     ProcessRxMessages(msg);
     m_parent_driver->m_listener.Notify(std::move(msg));

```

TwoCanPlugIn, 2023-11-04:
Isn't the last byte the CRC because irrespective of the driver, Actisense, Yacht Devices or SocketCan, the payload is in "Actisense" format ?

The "real NMEA 2000 message" payload length is at byte 12 (8), the data starts at byte 13 and in your example is
(96,48,196,76,64,208,186,37) and the last byte the CRC checksum.

duichan, 2023-11-04:
But if you look at the listing of a variety of different PGNs here:
https://github.com/antipole2/JavaScript_pi/issues/74#issuecomment-1791367373
the last byte is ALWAYS 85 so it can't be the CRC checksum

leamas, 2023-11-04:
Getting back to my probably bad phrased question above. Would it be possible for someone to rebuild opencpn with the patch above so we can see what's actually coming from the network?

It **should** be the same, but we cannot know for sure until we have logged it. 

The lack of logging is actually a bug, if no one else files it I will.

TwoCanPlugIn, 2023-11-04:
The powers that be, in their infinite wisdom  adopted the encapsulation of all NMEA 2000 messages in the Actisense protocol.

All messages arriving from the SicketCAN interface are encapsulated in this ridiculousness,, however some of the fields in this are not completed. Eg. the 4 byte date field is set to 255, the CRC field is not calculated, rather  it is set to a dummy value
From comm_drrv_socketcan
data.push_back(0x55);  // CRC dummy, not checked

That's why the value is always the same.....

The way to obtain the NMEA 2000 payload using your method is
for (I=0; I < payload[12]; i++) {
    dó something with payload[13 +I]

TwoCanPlugIn, 2023-11-04:
The lack of logging is actually a bug, if no one else files it I will."
I thought we had this discussion before, under the discussions tab.
https://github.com/OpenCPN/OpenCPN/discussions/3109
I would posit that given there are a myriad of ways to log N2K data, there are other more important fish to fry. How many users have posted queries or bugs regarding NMEA 2000?.

antipole2, 2023-11-06:
So if the dummy CRC 0x55 is always going to be  there by design, it can be dropped/ignored.
In that case, I am happy to close this issue.  OK?
The  logging issue is best pursued elsewhere?

TwoCanPlugIn, 2023-11-06:
If you're going to document a dummy CRC you need to also document the madness of encapsulating NMEA 2000 messages within an Actisense payload, in particular the NMEA 2000 
The CRC is only checked ( or should be checked) on incoming Actisense messages before they are consumed by OpenCPN to ensure the integrity of received data. The assumption would be that if a plugin receives a N2K message it is by definition, valid and there is no benefit to a plugin verifying the message and it's CRC. We are only in this situation because of the decision to pass an entire Actisense payload format to plugins instead of just the NMEA message.

leamas, 2023-11-06:
> If you're going to document a dummy CRC you need to also document the madness of encapsulating NMEA 2000 messages within an Actisense payload, in particular the NMEA 2000

Indeed. 

Not saying I will do it, but someone really should...

leamas, 2023-11-06:
@bdbcat : ^ thoughs?

antipole2, 2023-11-06:
The Actisense header includes the pgn, which is not in the NMEA data itself.  I find that useful when processing multiple pgns in one handler.

I just need things to be stable.  So are we are agreed no change:  to keep the header and the redundant CRC byte? For the JavaScript_pi, I will drop the CRC byte in the API before returning the rest to the script.  I just need it not to be excluded in some future version.  I will document this in my API code.

leamas, 2023-11-06:
> I just need things to be stable. So are we are agreed no change: to keep the header and the redundant CRC byte?

I would say yes, we should not destabilize if not  necessary. 

That would then mean documentation in ocpn_plugin.h and perhaps also some convenience method to unpack the "real" n2k data + PGN from the raw data.   But then again, we need @bdbcat 's input before proceeding.

TwoCanPlugIn, 2023-11-06:
> The Actisense header includes the pgn, which is not in the NMEA data itself. I find that useful when processing multiple pgns in one handler.

I haven't investigated whether the ObservedEvt includes the pgn, but I'm wondering whether this is a chicken or egg question. Even though you could potentially have one handler for all PGN's, as far as I know you need to know the id (pgn) in order to retrieve the payload.

This is a sample of code from the Engine Dashboard
`// PGN 127489 Engine Parameters Dynamic`
`	wxDEFINE_EVENT(EVT_N2K_127489, ObservedEvt);`
`	NMEA2000Id id_127489 = NMEA2000Id(127489);`
`	listener_127489 = std::move(GetListener(id_127489, EVT_N2K_127489, this));`
`	Bind(EVT_N2K_127489, [&](ObservedEvt ev) {`
`		HandleN2K_127489(ev);`
`	});`
.`.......`
`// PGN 127489 Engine Dynamic `
`void dashboard_pi::HandleN2K_127489(ObservedEvt ev) {`
`	NMEA2000Id id_127489(127489);`
`	std::vector<uint8_t>payload = GetN2000Payload(id_127489, ev);`
`.......`
As you can see, you need to know the PGN in order to retrieve the payload, or perhaps there is another method that I am unaware of.

Re, plugin development, there is no way for plugins to "listen" to all received NMEA 2000 messages and unless the ObservedEvt contains info on the NMEA 2000 message details (pgn, destination, source, priority), then we are left with the one handler per pgn dilemma.

Perhaps the ObservedEvt should have a ClientData object filled with am instance of the CanHeader class.

leamas, 2023-11-06:
> As you can see, you need to know the PGN in order to retrieve the payload, or perhaps there is another method that I am unaware of.

Indeed. And yes, there is no other way. This is on purpose  for performance reasons

> Perhaps the ObservedEvt should have a ClientData object filled with am instance of the CanHeader class.

Perhaps. However, this brings us into what happens in the core. Here, the only thing handled is `Nmea200Msg`, defined in _comm_navmsg.h_. If we should follow this path that would mean adding the CAN header here (trivial), and update all drivers  to handle it (more work).

The last piece would  be to make it available in the plugin interface. We don't expose Nmea200Msg here, it's way too complicated to fit in a single header file. Using ClientData is probably a bad idea (who owns that data?). But it should be possible to add some new function like GetN2000Whatever() which returns data containing both the raw data and possible other things like the CAN header.

Doable, but is it worth it? In any case, we seem to come to a point where we should do some driver work.

TwoCanPlugIn, 2023-11-06:
If there is no way to "listen to all messages", then there is little point in adding the CanHeader to the ObservedEvt ClientData, nor even something more trivial such as including the PGN using the event's SetInt/GetInt functions.

bdbcat, 2023-11-06:
Re: the "madness":

As we have discovered by drilling down a bit, the GetN2000Payload() method returns everything that OCPN core knows about the message, albeit in a rather inconvenient verbatim format.  As Tony has discovered, the method delivers all that is required in order to implement a functional plugin, however mad it may be.

One might argue that a "prettier" struct might be passed to the plugin, instead of a vec that needs to be parsed using opaque pre-knowledge of its contents.  Maybe we should have done that in O584.  But we did not.
I would be open to adding such a method.  This would of course prompt a new plugin API version.
But keep in mind that if Tony, for instance, wants to use the "prettier" method, he will need to  code branch on the detected  core API version, and then implement both methods to support legacy O584 with his plugin  If we know anything about users, it is that they are very slow to upgrade OCPN in the wild.

So, I suggest the pragmatic approach that Tony can do today:
`For the JavaScript_pi, I will drop the CRC byte in the API before returning the rest to the script
`
Of course, it would be a good idea to document the true layout of the passed vector.  But this clearly not strictly necessary, and we have lots of fish to fry elsewhere....

General comment:
As we get "smarter", we will continue to discover places in the codewhere we will say  "I wish this had been done better/faster/smarter from the start.  Sure would make life easier now.."  That is the nature of RAD with a small distributed team.

The users do not care about mad and inconvenient internal code.  They care about functionality and stability.
Someone must persistently speak for the end-users.  That will be me.

antipole2, 2023-11-06:
>As you can see, you need to know the PGN in order to retrieve the payload,

Not so, actually.  If you look at the code implementing GetPayload, you will see it ignores the pgn argument.

While this is not something to do regularly, I wanted to find out what pgns were being received on duichan's system.  He ran the JavaScript below for me (requires JS plugin v2.2 under development).  What it does is:

1. Fetch all descriptors in its copy of the canboat descriptors library
2. Listens out for them all with a single JavaScript handler `gotOne`
3. The handler extracts the pgn from the payload header and pushes it onto an array of pgns received
4. After 20 seconds, function `timesUp` is called.  It cancels outstanding listeners and prints the array of received pgns.

It worked just fine.  It delivered a list of the 17 pgns received.

```javascript
descriptors = require("pgnDescriptors.js")();
timeLimit = 20;  // seconds

pgns= [];

for (d = 0; d < descriptors.length; d++){
	pgn = descriptors[d].PGN;
	OCPNonNMEA2000(gotOne, pgn);
	}
onSeconds(timesUp, timeLimit);

function gotOne(result){
	payload = result.payload;
	pgn = getBytes(payload, 3, 3);	// extract the pgn
	pgns.push(pgn);
	}

function getBytes(v, start, bytes){	// little endean!
	offset = start+bytes-1;
	result = v[offset--];
	for ( ; offset >= start; offset--){
		// result = (result << 8) | v[offset];  shift uses 32 bits, so to be 64 bit safe we use...
		result = (result * 256) + v[offset];
		}
	return result;
	};

function timesUp(){
	OCPNonNMEA2000();	// cancel outstanding ones
	print(pgns, "\n");
	}
```

leamas, 2023-11-06:
> Listens out for them all with a single JavaScript handler gotOne

... and this is where you loose it.  The indended usage pattern is to have specific handlers for each thing you listen to. Using a lambda, this is dead simple and does not require defining a separata function.

leamas, 2023-11-06:
> I would be open to adding such a method. This would of course prompt a new plugin API version.

Not necessarily. We could feed tha data obtained today to some parsing method whcih returns the "real" data + perhaps PGN.  Such a method  could live in opencpn-libs without the need for updating the API.

bdbcat, 2023-11-06:
How could a plugin dev use the new method for a plugin intended to be used on O584?  It would not exist in O584 core.
Indeed, the plugin would fail to load on O584.

leamas, 2023-11-06:
> Indeed, the plugin would fail to load on O584.

No. It would fail in the build phase if opencpn-libs is not updated. However, what it uses from o-l is statically linked, so this is not  a runtime problem. Some flesh, just a sketch:

In some library in o-l:
```
struct N2kData {
    uint64_t PGN;
    std::vector<uint8_t> data;   // The "real" data after removing the Actisense envelope.
};
N2kData ParseN2kData(const std::vector<uint8_t> raw_data);
```

Plugin usage:
```
   ...
   auto raw_data = GetN2000Payload(SomePgnId, evt);
   N2kData parsed_data =  ParseN2kData(raw_data);
```
  
This can be done without updating the API an any way.

antipole2, 2023-11-06:
> > Listens out for them all with a single JavaScript handler gotOne
> 
> ... and this is where you loose it. The indended usage pattern is to have specific handlers for each thing you listen to. Using a lambda, this is dead simple and does not require defining a separate function.

Not sure I understand you here, Alec.  Of course, in normal usage there would be a png-specific JavaScript handler.  But in the exploration test I used, I did not want to set up 373 png-specific handlers.  So using one handler and deducing the pgn from the payload was the way.  But I think we are going down a rabbit hole here.

bdbcat, 2023-11-06:
Ahhh.
"In some library in o-l:".... makes a difference.

leamas, 2023-11-06:
> But I think we are going down a rabbit hole here.

Yes. To be discussed in zulip if need be

TwoCanPlugIn, 2023-11-08:
> As you can see, you need to know the PGN in order to retrieve the payload, or perhaps there is another method that I am unaware of.

> Indeed. And yes, there is no other way. This is on purpose for performance reasons

Well Tony has proven this to be wrong.  That'll teach me to believe the API definition!

If I register a gazillion PGN's
`wxDEFINE_EVENT(EVT_N2K, ObservedEvt);`
`for (auto it : parameterGroupNumbers) {`
`	listeners.push_back(std::move(GetListener(NMEA2000Id(it), EVT_N2K, this)));`
`}`
and bind to a single handler
`Bind(EVT_N2K, [&](ObservedEvt ev) {`
`		HandleN2K(ev);`
`	});`

Essentially I can listen to all PGN's

And as Tony discovered the GetN2000Payload API doesn't actually check the id, so the following can be used to parse/log all received PGN's!

`void NetworkPlugin::HandleN2K(ObservedEvt ev) {`
`	std::vector<uint8_t>payload = GetN2000Payload(NMEA2000Id(123456), ev);`
`	unsigned int pgn = payload[3] | (payload[4] << 8) | (payload[5] << 16);`
`       .....`

While this doesn't "agree" with Alec's design goal of one handler per pgn, it solves the problem of listening to all PGN's., which can be useful for NMEA 2000 to NMEA 183 gateway, data logging/diagnostics etc.

Is there any intention to "close" this potentially useful bug ?

antipole2, 2023-11-08:
In my use, there could be more than one listener for a given pgn  and so the handler could be invoked more than once with the same payload.  For information, this is how I link the set up of the listener to the resulting event when using a single handler...
I set up the listener with an extra parameter `messageCntlId` in the lambda, which in my case is a random number.
````
Bind(EVT_JAVASCRIPT, [&, messageCntl](ObservedEvt ev) {
				HandleNMEA2k(ev, messageCntl.messageCntlId);
				});
````
So when the event happens and the lamda  fires,  the handler is called thus:
````
void Console::HandleNMEA2k(ObservedEvt& ev, int messageCntlId) {
````
and from the `messageCntlId` I work out for which listener this particular event is a response.  From the id I can work out the pgn and I collect the payload with the pgn parameter, although, as I have discovered,  it is actually ignored at present.

leamas, 2023-11-08:
There are ore than one way to skin a cat, for sure. However, to me this looks more complicated than necessary. 

 Binding to a single handler without adding the at "bind-time" known PGN creates the need to reconstruct it later. This is of course possible as demonstrated here, but certainly not the only way to handle this.

 A probably better alternative would be to  to create a class similar to ObsListener which handles listening and also directly binds to a handler like handleEvt(PGN, payload). Using such a class would be as simple as current bind calls.

Yes, it's possible to listen to a lot of PGNs. However, there is no "catch-all" like the one which actually exists for 0183, you need to know what PGNs you listen to in advance. This is still a basic limitation which keeps the critical code paths simple. and fast.

I suppose some bright mind could set up a loop creating listeners for all theoretically possible PGNs. Not really sure this is a good idea...

leamas, 2023-11-08:
Perhaps I should keep my mouth shut, dunno, but something like this:
```
class PluginListener : public wxEvthandler {
public:
   PluginListener(uint64_t PGN, SomeClass& parent) {
     const wxEventTypeTag<ObservedEvt> EvtJavaplugin(wxNewEventType());
     m_listener = GetListener(Nmea2000Id(PGN), EvtJavaplugin, this);
     Bind(EvtJavaplugin, [&] (ObservedEvt ev) { parent.handle(PGN,  GetN2000Payload(ev)); });
  }
}
```

And that's it . Such listeners coule be created in SomeClass. Creating an instance using for example `PluginListener listener(SomePgn, this)` would be enough to set u p listening. such instances could also be kept around as long as the listening should continue, the destructor does the right thing. 

And voilá, no need to reconstruct the PGN...

bdbcat, 2024-01-04:
Are we done here?  Any actionable bug left after dissection?

antipole2, 2024-01-04:
The original topic is that the payload length is one too large for the N2K message because it includes the dummy CRC '85' on the end.
We now have an explanation and, I think, the resolution is to just know that and ignore the extra byte. In my code, I just drop it.
I guess it would be cleaner for it to be dropped before returning the payload but that has not been proposed.
I am OK with leaving it as is, even if it is a bit untidy.

The topic then drifted into a discussion on multiple listeners.  That is more defuse.  I do not see any firm conclusion or action needed.
So I think this can be closed.  As the OP I will wait a week and then close if no reason against.

bdbcat, 2024-01-04:
Indeed, it would be more tidy to remove the extra byte.
But if we did that, then your plugin would need to branch on OCPN core version, to support existing O584 users.  Also untidy.
Lets leave it.

antipole2, 2024-01-05:
My plugin presently copies the data out of payload with
`for (int i = 0; i < payload.size()-1; i++){	// drop the dummy CRC end byte`

I have changed this to
```
int count = payload.size();
if (((count - payload.at(12) - 12) == 1) && (payload.back() == 85)) count--;	// drop any crc if present
for (int i = 0; i < count; i++){
```
so that it will work with or without the extra crc byte.  So payload can be tidied up to avoid confusion by the unwary.
Saves having to document the oddity somewhere.

bdbcat, 2024-01-05:
OK, good solution.
I'll make the change in O59 github master.

TwoCanPlugIn, 2024-01-08:
The NMEA 2000 payload should just be the payload without the extraneous Actisense preamble & CRC.

But too late now.....

bdbcat, 2024-01-08:
After some study of OCPN core code, and some plugins, I have decided to leave the API as is.  That is, leave the final 0x55 byte in place as part of the payload.
Reasons/discussion:
1.  For those plugins that use the N2K library functions available in opencpn-libs submodule (e.g. Dashboard), the correction could be made transparent. to the plugin.
2. However, there are at least two plugins (javascript and AutoTrackRaymarine) that do not use the n2K library.  These two plugins parse the N2K payload manually, picking out the required fields by simply knowing their offsets and format within the payload vector.
3. As you have described, javascript plugin knows how to detect and discard the extraneous byte.  But the effect on AutoTrackRaymarine is unknown without further investigation.

So I think we can declare this issue as "stet", however untidy it may be.

Comments?

leamas, 2024-01-08:
At a minimum this should IMHO be fully documented in ocpn_plugin.h, describing what user actually gets when running  GetN2000Payload().

antipole2, 2024-01-08:
> At a minimum this should IMHO be fully documented in ocpn_plugin.h, describing what user actually gets when running GetN2000Payload().

An example is often clearest.  How about including as comment the following example lifted from my test data.  It is JavaScript but very clear what is meant.
```
payload: [147,19,      // fixed part of Actisense header
	3,		// priority
	16,240,1, // pgn 126992 encoded little endian
	255,        // destination
	1,		// origin
	255,255,255,255, // timestamp always undefined when received
	8,		// count of following NMEA2000 data
	13,240,207,76,208,3,94,40,	 // NMEA2000 data
        85           // dummy CRC not included in count
	];
```

bdbcat, 2024-01-09:
Ready to close this now.

closed 2024-01-09 by antipole2

leamas, 2024-01-10:
Updated the docs and also the doxygen setup in d7992745c. The new docs is available [here](http://opencpn.github.io/OpenCPN/api-docs/ocpn__plugin_8h.html#a30ab197f18c0e36c659fab5c9fcbf84e)