OpenCPN/OpenCPN #3489: BUG? COM Data not read? opened?

Corsair-63, 2023-11-02:
**Describe the bug**
in pdf attached together with log file
[COM port NMEA data.docx](https://github.com/OpenCPN/OpenCPN/files/13238791/COM.port.NMEA.data.docx)
[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/13238792/opencpn.log)

Corsair-63, 2023-11-02:
Found previous version of CH340/341 COM drivers certified for windows 11 (version 3.7.2022.1).
uninstalled driver for com port in device manager.
removed (unplugged) device.
installed this version (3.7.2022.1)
plug-in device to USB.
double check that version 2002 is present instead of 2023.
connected via Termite and shown that correct data is being received.
started OCPN and now is showing the requested data.

it seems to be a problem of the drivers themselves.

leamas, 2023-11-02:
So this is probably not a bug in OpenCPN. Shall we close it, or perhaps convert it to a discussion instead?

Corsair-63, 2023-11-02:
I guess I'll close, this is a driver/windows issue, no need to go further, only if anybody will find same issue that could see it in order to solve it.

closed 2023-11-02 by Corsair-63