OpenCPN/OpenCPN #3490: OCPN master 5.9 will not build on MacOS X86

bdbcat, 2023-11-06:
Macbook, X86 native
MacOS 11.7
XCode 13.4

Build result:
Undefined symbol at link step:
_MipMap_32_avx2

I did some debugging.
It appears the the macro __AVX2__ gets defined and is active when compiling file mipmap.c.
However, __AVX2__ seems not defined when compiling mipmap_avx2.c.
As a result, the method _MipMap_32_av2 is not compiled.

Bisecting now....

nohal, 2023-11-06:
It will be 76f15fe3c540cd2010515847c468eaa5bb84d0c4 and the previous commit from Sean. But I'm a bit puzzled as it does build on both the CI Intel machines and on my Intel Mac without any problems.
Does the machine where it fails to build actually have AVX2 support?
I'm at sea now, so will have pretty limited chances to do about it in the next ~10 days.

transmitterdan, 2023-11-06:
Also, I believe I saw this on at least on Linux build.

LeaseOnLife, 2023-11-06:
possibly related:

LeaseOnLife, 2023-11-06:
possibly related: latest fresh master on a PI4 armhf Debian Bookworm:
libs/mipmap/src/mipmap.c:165: undefined reference to `MipMap_24_neon'

bdbcat, 2023-11-06:
On Mac.
The question revolves around the use of "-march=native".
In my understanding, this compile option tries to automatically include some optimizations for known CPU architecture.  "__AVX2__" is one such macro automatically defined when "-march=native" on X86 hardware.
And maybe that is not such a good thing.  Or at least we should be consistent.
Here is where we touch this:
libs/mipmap/CMakeLists.txt

```
if( NOT QT_ANDROID)
  # always use optimizations for the texture compression which is much faster
  if (NOT MSVC)
      set_property(TARGET MIPMAP PROPERTY COMPILE_FLAGS "-fvisibility=hidden -O3")
      if (NOT APPLE OR NOT OCPN_RELEASE STREQUAL "universal")
          set_source_files_properties(
            src/mipmap.c PROPERTIES COMPILE_FLAGS "-march=native")
      endif()

```

On my Mac, this enables "__AVX2__" for compilation of "mipmap.c"
But the file implementing AVX2 support for O, "mipmap_avx2.c" is not compiled with "-march=native".
So, build fails.

There is some google juice on march=native, plus and minus.  See here:
https://stackoverflow.com/questions/52653025/why-is-march-native-used-so-rarely

tl;dr;
If I disable "-march=native", build succeeds.

Thoughts?

nohal, 2023-11-06:
> [Tracking image][https://github.com/notifications/beacon/AAB5AAEMWIBYWAJ3WVMPTM3YDFEKLA5CNFSM6AAAAAA67UMD3OWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTLCMFQU.gif]

The -march=native already turned to be problematic in macOS universal builds so I disabled it there. The whole idea of the mipmap optimization is to build as many optimized versions as possible and select the correct one at runtime, so if we can get there without native, and I think we can, we should probably only enable the individual optimizations we detect as buildable

LeaseOnLife, 2023-11-06:
> If I disable "-march=native", build succeeds.

same on  PI4 armhf Debian Bookworm

leamas, 2023-11-06:
armhf on a Pi 4 must be considered a corner case...

bdbcat, 2023-11-07:
In the interest of broad field compatibility, I'll push a fix to avoid enabling "-march-native".  We will still get the various SSE options, if available at runtime.  I suspect the run-time performance difference will be practically unmeasurable.

commit b55afb96 (2023-11-07): Disable "-march=native" on Mac builds.  Closes #3490
closed 2023-11-07 by bdbcat