OpenCPN/OpenCPN #3496: SEND TO GPS doesn't work in 5.8.4 -- GARMIN to GARMIN

svMigration, 2023-11-14:
Connecting a Garmin GPSmap 76cx via USB serial to Windows 11 laptop with OpenCPN 5.8.4... routes cannot be sent Garmin to Garmin. This worked fin in 5.8.2.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect USB serial cable to computer and Garmin GPSmap 76cx
2. Change Garmin GPSmap settings to GARMIN protocol (not NMEA)
3. Make sure "Use Garmine GRMN (Host) more for uploads" is checked in settings
4. Right click on route and Select Send to GPS (<correct port>)
5. Error: "Error on Route Upload. Garmin GPS not connect. "

**Expected behavior**
Route should be sent to GPS

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Version 11

**Additional context**
I installed 5.8.2 to check this problem. It worked perfectly sending a route to the Garmin GPSmap 76cx. I then started up 5.8.4 and tried it. Received the message:Error on Route Upload. Garmin GPS not connect. Log says:
14:28:58.370 MESSAGE RolloverWin.cpp:185 Render texture 46
14:29:00.683 MESSAGE comm_drv_factory.cpp:54 MakeCommDriver: Serial:COM3
14:29:00.793 MESSAGE comm_n0183_output.cpp:439 Garmin GPS could not be initialized on port: COM3 Error Code is -4
LastGarminError is: CreateFile failed
14:29:06.466 MESSAGE comm_bridge.cpp:186 ***GPS Watchdog timeout at Lat:-16.6286 Lon: -143.572

===========

Tried it again on 5.8.2 and it worked. Here's the log:

Link_type 1 Device_command 0
14:34:58.598 MESSAGE gps_wx_logging.h:30 Waypoint: Transfer 100 Type 110
14:34:58.598 MESSAGE gps_wx_logging.h:30 Route: Transfer 201 Header 202 Type 110
14:34:58.598 MESSAGE gps_wx_logging.h:30 Track: Transfer 301 Type 302
14:34:59.345 MESSAGE comm_n0183_output.cpp:427 Sent Route to Garmin GPS on port: COM3
Unit identifies as: GPSMap76CX Software Version 4.20
14:35:07.193 MESSAGE comm_bridge.cpp:186 ***GPS Watchdog timeout at Lat:-16.6286 Lon: -143.572
14:35:36.224 MESSAGE ocpn_frame.cpp:5640 LOGBOOK: 2023-11-06 00:35:36 UTC DR Lat -16.62860 Lon -143.57230

Same computer, same cable, same USB port, same GPS.  So I think this confirms something that something got broken in 5.8.4. The problem started as soon as I upgraded from 5.8.2 to 5.8.4.

Thanks for looking into this.
Cheers,
Bruce

leamas, 2023-11-14:
Perhaps related to  #3205?

bdbcat, 2023-11-15:
Bug confirmed.
A correcting patch is ready for master, not pushed yet.
Could be backported to O58x series, but I'm thinking that correcting this issue is not worth the cost and complexity of a release cycle. This is the first report of this problem in the 5 months life of O584.  A workaround (O582) is available now.

Bruce, any trouble for you to revert to O5.8.2 until the next full release?
Thoughts?

svMigration, 2023-11-15:
Hi Dave,
No worries at all. I noticed it right away but didn't get around to reporting it until now. Just wait until future release. I have 5.8.2 installed now and can use that when needed.
Thanks for finding the problem!
Cheers,
B

bdbcat, 2023-11-16:
B...
Correcting change now pushed to github, so will appear in next version.
We can leave this bug open until Beta testing starts, whenever that may be.
Thanks

pa2wlt, 2024-02-11:
I ran into this bug as well. Costed me quite a few hours to check everything, before finding this issue, and seeing it's a bug. I reckon there is no fix available yet that can be installed yet, right?

nohal, 2024-02-11:
> I reckon there is no fix available yet that can be installed yet, right?

It depends on your platform, CI build binaries automatically published at cloudsmith.io/~david-register/repos/opencpn-unstable/packages/ already contain the fix, if you are not covered by those, you need to build from source.

thonglorais, 2024-02-11:
I do experience a similar effect with Garmin GPSMAP 66st albeit on 5.8.2, described [here](https://www.cruisersforum.com/forums/f134/garmin-gpsmap-66st-gps-receive-works-upload-does-not-work-for-routes-or-waypoints-283126.html)

Does this justify an own github issue or is it related to the bug described above?

nohal, 2024-02-11:
As your problem seems to be something completely different, it probably does justify having it's own issue with all the relevant information. It would also be good to test whether it is stil present on the current code as I doubt anybody lookig into it has access to the GPSMAP device.

thonglorais, 2024-02-11:
Yes, the same error occurs in OpenCPN 5.8.4-3+1637c28.
In fact, that's where I tested in first and after having read this issue #3496 I downgraded to 5.8.2 to check whether it also exiosts there.
I have created a new issue report: [#3651](https://github.com/OpenCPN/OpenCPN/issues/3651)

thonglorais, 2024-06-15:
@svMigration a new software version opencpn_5.9.1 is out with chagens to the Garmin upload code. I tested my issue https://github.com/OpenCPN/OpenCPN/issues/3651 and it still exists, but maybe your issue is fixed?

svMigration, 2024-06-16:
Will try it as soon as I can.

Cheers,
B

 

svMigration.com <https://svmigration.com/> 

 <https://chartlocker.brucebalan.com/> The Chart Locker

 <https://www.lifehack.org/articles/productivity/7-reasons-why-quitting-facebook-now-good-for-your-future.html> Quit Facebook. Disconnect from the net;  <https://www.humanetech.com/> reconnect to the world.

svMigration, 2024-06-17:
>  I tested my issue #3651 and it still exists, but maybe your issue is fixed?

@thonglorais: tested and my issue is fixed. Works with Garmin GPSmap 76cx via USB serial to Windows 11 laptop.

leamas, 2024-06-17:
@svMigration : Since it is fixed, could you please close it?

closed 2024-06-17 by svMigration