OpenCPN/OpenCPN #3497: The CLI interface is not consisten

leamas, 2023-11-14:
**Describe the bug**
The CLI interface is not consistent with long options random using either a single hyphen '-' or double `--`

**To Reproduce**
Steps to reproduce the behavior:
1. On the comman line, invoke `opencpn --help`. Results:
```
Usage: opencpn [-h] [-p] [-fullscreen] [-no_opengl] [-rebuild_gl_raster_cache] [-parse_all_enc] [-l <str>] [-unit_test_1 <num>] [-unit_test_2] [--unit_test_2] [-safe_mode] [-r] [-R] [-q] [-e] [-o <str>] [import GPX files...]
  -h, --help              	Show usage syntax.
  -p                      	Run in portable mode.
  -fullscreen             	Switch to full screen mode on start.
  -no_opengl              	Disable OpenGL video acceleration. This setting will be remembered.
  -rebuild_gl_raster_cache	Rebuild OpenGL raster cache on start.
  -parse_all_enc          	Convert all S-57 charts to OpenCPN's internal format on start.
  -l, --loglevel=<str>    	Amount of logging: error, warning, message, info, debug or trace
  -unit_test_1:<num>      	Display a slideshow of <num> charts and then exit. Zero or negative <num> specifies no limit.
  -unit_test_2            	
  --unit_test_2           	
  -safe_mode              	

```

**Expected behavior**
Long options should consistently use `--`.  

**Additional context**
This is  basically a simple bugfix. However, it will break the current cli interface.  I don't think this is that bad, current mess means that those who actually use the cli will get confused anyway trying to get the '-' vs '--' right.

leamas, 2023-11-14:
A simple update would give the following interface:
```
$ ./opencpn --help
Usage: opencpn [-h] [-p] [-f] [-G] [-g] [-P] [-l <str>] [-u <num>] [-U] [-s] [import GPX files...]
  -h, --help                   	Show usage syntax.
  -p, --portable               	Run in portable mode.
  -f, --fullscreen             	Switch to full screen mode on start.
  -G, --no_opengl              	Disable OpenGL video acceleration. This setting will be remembered.
  -g, --rebuild_gl_raster_cache	Rebuild OpenGL raster cache on start.
  -P, --parse_all_enc          	Convert all S-57 charts to OpenCPN's internal format on start.
  -l, --loglevel=<str>         	Amount of logging: error, warning, message, info, debug or trace
  -u, --unit_test_1=<num>      	Display a slideshow of <num> charts and then exit. Zero or negative <num> specifies no limit.
  -U, --unit_test_2            	
  -s, --safe_mode              	Run without plugins, opengl and other "dangerous" stuff
```
This would not affect the most common use cases like without any options at all, some GPX file(s) to load or a single `-p` option. However, for some other long options will be changed. Still, I think it's worth it.

leamas, 2023-11-14:
Note to me: patch: 
[0001-ocpn_app-clean-up-CLI-interface-3497.patch.gz](https://github.com/OpenCPN/OpenCPN/files/13347747/0001-ocpn_app-clean-up-CLI-interface-3497.patch.gz)

transmitterdan, 2023-11-14:
I agree, we should do the "normally expected thing" unless a really good reason not.  Best to fix this while it's still early days.

closed 2023-11-14 by bdbcat