OpenCPN/OpenCPN #3500: Sound test may cause crash

transmitterdan, 2023-11-16:
**Describe the bug**
In options.cpp at https://github.com/OpenCPN/OpenCPN/blob/70ecca98a8036b707e49b66ba00ddf9ec1338884/src/options.cpp#L4840 we set a callback which simply destroys 'sound'.  I'm pretty sure this is a crash waiting to happen.  The code that calls the callback functor is in the sound object itself and therefore this callback is forcing the object to call its own destructor which I think is not ok.

I'm not sure what was intended here but a simple fix is to rewrite the sound test function like this:
```
void OCPNSoundPanel::OnButtonTestSound(wxCommandEvent& event) {
  auto sound = SoundFactory();
  auto cmd_sound = dynamic_cast<SystemCmdSound*>(sound);
  if (cmd_sound) cmd_sound->SetCmd(g_CmdSoundString.mb_str());
  sound->Load(m_sound_file, g_iSoundDeviceIndex);
  sound->Play();
  delete sound;
}
```
This would not create a worker thread and just block until the sound playout finishes.  Why do we need to create a worker thread for a sound test?

leamas, 2023-11-16:
I don't think this is an actual problem. That code is executed in the  lambda context, that is basically in the same environment as `SetFinishedCallback()`is executed i. e., not "inside" the sound object.

Generally, handling thinks like playing sounds is IMHO better done in separate threads.

leamas, 2023-11-16:
IIRC, at least gdb is clever enough to debug this. That is, if you re-format the code to something like

```
sound->SetFinishedCallback([sound](void*) { 
  delete sound;
 }); 
```
you can set a breakpoint at `delete sound` and look around.

leamas, 2023-11-16:
This is actually not what is happening here if I get this correctly. Have you read up on lambdas?

transmitterdan, 2023-11-16:
When I run it in the VS debugger and it crashes after the return from:
https://github.com/OpenCPN/OpenCPN/blob/70ecca98a8036b707e49b66ba00ddf9ec1338884/libs/sound/src/OcpnWxSound.cpp#L59

The attempt to zero m_onFinished causes an exception apparently because invoking the lambda destroyed the OcpnWxSound object and thus member m_onFinished is no longer a valid write address.  If m_onFinshed would be zeroed by ~OcpnWxSound() then I think all would be ok.  But not after the destructor returns.

leamas, 2023-11-16:
Away for a while... but this indeed looks intriguing. However, making this blocking it not the solution.

> . If m_onFinshed would be zeroed by ~OcpnWxSound()...

Something you could try?

leamas, 2023-11-16:
hm... after some thinking:
  - This code has been in place for a long time. If the delete logic is wrong we should have had reports, but no such are filed.
  - I have encountered other devs having problems when trying to debug lambda expressions with VS. Perhaps part of this, perhaps just a misunderstanding.
  - It is possible to rewrite this using an EventVar. This would basically decouple the callback which would do Notify() from the actual deleting done in the listener on next main loop iteration. This would be a clean way to make things more defined. I'll try to make a patch during the weekend.

leamas, 2023-11-17:
Here is a patch which more definitely decouples deleting from the callback.  I see no actual difference when I use this on Linux, but in some ways it "looks" better. Any chance you could test?

[0001-options-Better-sound-handling-3500.patch.gz](https://github.com/OpenCPN/OpenCPN/files/13388149/0001-options-Better-sound-handling-3500.patch.gz)

BTW: How is your sound configured?  Are you using the MswSound backend or something else?

transmitterdan, 2023-11-17:
@leamas this patch works for Windows.  Also, my previous suggestion to delete the function pointer in ~OCPNSound() (https://github.com/OpenCPN/OpenCPN/issues/3500#issuecomment-1814330864) works also.

transmitterdan, 2023-11-17:
As I think about this, it seems presumptuous for the worker thread to zero out the callback function pointer.  It seems more intuitive that it should be up to the user of this object to change/clear the callback pointer.  This works for me in options.cpp:

  ```sound->SetFinishedCallback([sound](void*) {
    sound->SetFinishedCallback(0);
    delete sound;
  });
```
So, when the asynchronous notification happens, clear the callback and delete the object.

One could imagine scenarios where each time the Play() function is called it wants to be notified every time it finishes.  The way it is now, only the first Play() call will run as a thread. Any subsequent call to Play() for that object instance will be single threaded which is probably not what was intended.

Thoughts?

leamas, 2023-11-17:
Before we proceed here: how is your sound backend configured? Are you using MswSound or something else?

transmitterdan, 2023-11-18:
I am studying all Win32 compatible backends.  More report later today/tomorrow.  It's not urgent.

transmitterdan, 2023-11-23:
My proposed solution can be found here: https://github.com/transmitterdan/OpenCPN/commit/80653f5d107580b49b5814dad68060477ef28f49

Tested on Windows and Linux. Seems to be ok.  No more issues with the lambda and the patch seems pretty simple.

Background:
1. Deleting the callback function pointer in the worker() thread after it finishes a playout seems wrong. If a user of OcpnSound object does not continuously recreate the callback pointer after every call to Play() then the object will only play the first call to Play() in a thread.
2. If a caller really wants to clear the callback pointer after Play() it can easily be done in the callback function itself.  But no instance I found needs to do that.
3. The base class should manage its members as much as possible, so I made the base class destructor clear the function pointer even though it may be unnecessary even to do that.
4. I could not see why SystemCmdSound::worker() was organized differently than the other derived class implementations, so I harmonized it.

leamas, 2023-11-24:
I need some time to work through this. But I will. Please stay tuned...

leamas, 2023-11-25:
hm... I wonder if this is the right direction.  

There are two use cases, one of which being deleting the sound object after playing the sound. This is  the focus here,  and I assume it might work.

However, the other use case is that the sound is an  automatic variable, typically on the class level. The sound should here be used by playing consecutive sounds, one after another. 

Think of a bell sound repeated n times. The callback is here used to synchronize start of  new sounds when the previous us completed

From that perspective the patch does things like:
```

    wxLogDebug("mswSound::worker()");
    m_isPlaying = true;
    PlaySound(m_path.c_str(), NULL, SND_FILENAME);
-  if  (m_onFinished) {
+  m_isPlaying = false;
+  if (m_onFinished) {
        m_onFinished(m_callbackData);
-        m_onFinished = 0;
    }
-    m_isPlaying = false;
}
```

This is in the asynchronous worker thread. If I get this correctly, the patch breaks things if Play() is called after an earlier Play().  The contract rigth now is the Play() can commence a new sound if m_isPlaying and m_on_finished are cleared by completion of the any sound played earlier, else not.

Or?

leamas, 2023-11-25:
I also note that you have not touched PortAudioSound which is the default Linux backend. That is, it is left with the original logic...

transmitterdan, 2023-11-25:
> I also note that you have not touched PortAudioSound which is the default Linux backend. That is, it is left with the original logic...

I see. Will take a look at it.

leamas, 2023-11-26:
That is, I actually think the current basic logic is sound (pun intended).  However, this problem could be attacked from yet another angles:
1. Using `new` is basically always a bug. Would it be possible to use a sound variable in some class this fixes the problem since there is no need to use delete at all.
2. We could implement code in the destructor which waits until the sound is finished before proceeding. With this is place, we could just delete the sound after starting playing.

Both of these actually looks attractive to me.1.  should always be applicable, 2. seems basically makes things more consistent. It should probably limit the wait period so the destruction isn't  stopped by long running sounds. OTOH, the use cases we have are actually about reasonably short signals rather than playing a song.

No immediate plans to implement this.

transmitterdan, 2023-11-26:
I think it is simpler than that.  Just don't zero the callback function pointer except in the destructor.  With that simple change it seems to work for me in Windows and Linux.  I haven't looked at Android but don't feel we should touch that.  I made this simple change in PortAudio version with no ill effects.

https://github.com/transmitterdan/OpenCPN/commit/7fd9eca28d7156f383756084cd7f81e2d3056912

I can't think of any good reason to zero the callback pointer after the sound plays.  What is the point in that?

More pedantery:

I looked through PortAudio version and find that zeroing the callback breaks the blocking/non-blocking logic.  The class determines whether or not to use blocking or non-blocking sound based on the callback pointer value when the callback is created.  Creating a null callback means to use blocking audio playout.  Zeroing the callback pointer breaks the contract because now the class thinks non-blocking audio should be used but the callback pointer is zero and thus any notifications after the first one won't be made.

leamas, 2023-11-26:
> I can't think of any good reason to zero the callback pointer after the sound plays. What is the point in that?

The point is that after playing the sound file the sound object should be reset to the pristine state, the same as when it is created. In other words, the second sound should be executed in the same "environment" as the first.

leamas, 2023-11-26:
hm... This could really be seen  from two perspectives:
1. Current code does the wrong thing  when trying to delete the sound object from within the callback.
2. Current code is buggy since it is not possible to delete the sound object from within the cllback.

The more I think about this, I tend to lean to 1. That is, this isn't really about a bug in he sound object code as such but in how it used in this particular case.

transmitterdan, 2023-11-26:
> The point is that after playing the sound file the sound object should be reset to the pristine state, the same as when it is created. In other words, the second sound should be executed in the same [...]

Maybe it's a problem to communicate my idea. There is nothing wrong with the state of the object after setting the callback and issuing Play() which tells the object we wish to play sounds without blocking. One could easily keep calling the Play() function multiple times getting each time the notification that the sound completed. There is no need to keep setting the callback function before every Play(). At least I cannot see a reason. And if a user wants to do that nothing prohibits it.

Why must a user set the callback function pointer before every call to Play()?

leamas, 2023-11-26:
> Why must a user set the callback function pointer before every call to Play()?

Because the presence of the callback pointer is what makes the call asynchronous.

leamas, 2023-11-26:
Taking a step back here: What about my 1. vs 2. ^^^?

transmitterdan, 2023-11-28:
After more review this is what I think.

1) IMO, putting the sound object into the asynchronous mode (by setting a callback pointer) should be "sticky".  This is logical to me.
2) If you agree with that, then that is the entire solution to the problem of calling the destructor in the callback function. Doing so causes no problems.

I can see there are other points of view, but in my simple way of thinking, if I tell an object to be in a certain mode it should stay in that mode until I change it.  The object itself should not switch modes 'behind the scenes'.  I think it's no more complicated than that.

If you disagree, and feel it is important that the object revert back to synchronous mode after every asynchronous Play() call, then don't call the destructor in a callback function. Then one has to find another way to delete the object. Or declare it on the stack and let the compiler destroy it.

If the latter is preferred, I think there is a bug in the PortAudioSound implementation because it has a separate flag (m_isAsynch) indicating asynchronous mode. So, it can become confused by just zeroing the callback pointer.  But I don't totally understand that implementation, so I could be wrong.  I just note that the Windows implementations don't use 2 flags to indicate asynchronous mode.  They just rely on the callback pointer being NULL or not.

leamas, 2023-11-28:
We are setting sort of a good example here, discussing the change thoroughly. ;)

>  if you disagree, and feel it is important that the object revert back to synchronous mode after every asynchronous Play() call, then don't call the destructor in a callback function.

Yup. This just needs fixed comment + my patch [above](https://github.com/OpenCPN/OpenCPN/issues/3500#issuecomment-1815817733). This is basically about sending a message to self and handle it. This used to be a little painful, but using an EventVar and a ObsListener it's actually not much to write.

> I can see there are other points of view, but in my simple way of thinking [...}

If we were starting with a blank sheet of paper I would have easier to follow this thought. There are arguments against it, on is that the common "ring a bell n times" will need to  change the callback anyway. Another is that stateless things tends to be easier to understand -- and this is known to be hard to understand anyway.

But in the end, my main concern is about changing an existing interface for in my eyes little gain.  Shouldn't we be a little conservative with such changes if the benefit isn't that obvious? 

I think we both agree that the current interface isn't fundamentally wrong. And, lessons learned for me, the idea of deleting the object in it's own destructor isn't a good one.

leamas, 2023-11-28:
> , I think there is a bug in the PortAudioSound implementation

I'll try to take a look later. However, this is used in the official packages since God knows when, and we have no bug reports. So  it's probably not that bad. 

That is not to say there is no corner case lurking around.

leamas, 2023-12-28:
@transmitterdan: trying to breathe some new life into this by filing PR above.

One way to look at this is that here are two issues:
1. Deleting the sound object in the callback is a bad idea potentially leading to crashes
2. The interface to  OcpnSound is maybe not intuitive and could (should?) be improved.

If there was no way to address 1. without also modifying the interface these two issues would be tightly connected. However, as patch above shows i's possible to fix 1. with a minimal patch which does not change the interface.

This would then boil down to that we merge #3561 and close this issue. And then, perhaps, files a new issue about the interface being non-intuitive starting a fresh, separate discussion.

Thoughts?

leamas, 2023-12-28:
@transmitterdan :

>  I think there is a bug in the PortAudioSound implementation

Could you please be a little more specific here? I don' see it.

What I do see is a clumsy, error-prone non RAII spinlick handling. No to me obvious bugs, but should probably be fixed anyway.

transmitterdan, 2023-12-29:
> Could you please be a little more specific here? I don' see it.

I explained it as best I can.  The PortAudio implemenation uses a completly different scheme to indicate synch/asynch mode has been requested.  It uses 2 flags which seems wrong or at least redundant and prone to confusion.

commit 3011edb5 (2023-12-30): options: Fix possible sound crash (#3500)

Fixes problems described #3500 while keeping the current interface,
thus representing a minmal patch for the problem

Closes: #3500
closed 2023-12-30 by leamas