OpenCPN/OpenCPN #3501: OpenCPN crashes on macOS 10.13

tumic0, 2023-11-17:
OpenCPN 5.8.4 crashes on start on macOS 10.13 with the following stdout:

```
iMac:MacOS tumic$ ./OpenCPN 
 SENC: nCPU: 2    m_max_jobs :1
Generating RSA key...
dyld: lazy symbol binding failed: Symbol not found: ____chkstk_darwin
  Referenced from: /Applications/OpenCPN.app/Contents/MacOS/./../Frameworks/libcrypto.3.dylib (which was built for Mac OS X 11.0)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: ____chkstk_darwin
  Referenced from: /Applications/OpenCPN.app/Contents/MacOS/./../Frameworks/libcrypto.3.dylib (which was built for Mac OS X 11.0)
  Expected in: /usr/lib/libSystem.B.dylib

Abort trap: 6

```

According to the download page OpenCPN should work on macOS >= 10.10 but at least the included OpenSSL libs are compiled for macOS >= 11.0.

nohal, 2023-11-18:
Please try https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.9.0-legacyintel+6397.edb2c45f2-12.pkg
Built completely for 10.13 deployment target and works fine for me on 10.14, don't currently have anything more ancient at hand.

tumic0, 2023-11-19:
Seems to work on 10.13

leamas, 2023-11-28:
@tumic0  The habits here are that the reporter closes the issue when it's fixed.  

Since this seems to be fixed: could you please close this one?