OpenCPN/OpenCPN #3506: River mile/km display needs improvement

nohal, 2023-11-20:
**Describe the bug**
The rendering of rifer distance is not optimal.
The symbol always contains `km`, which is wrong on the US and other waterways where miles are used and overlaps with the value when it is available and rendered

[image]

When the numerical value is available, we should probably change the rendering not to use the raster symbol and render only a circle and the numerical value label instead.

bdbcat, 2024-04-23:
Need to install app to update symbology.
OK to close?

bdbcat, 2024-04-23:
[image]

nohal, 2024-04-23:
I think so

closed 2024-04-23 by nohal

bc38, 2024-04-23:
"km" removed?

bc38, 2024-04-23:
"km" should be left and the value should be 10mm from "km"
Almost everywhere there is "km" along S-52
[image]
[image]

nohal, 2024-04-23:
As far as I am concerned, I've heard enough of "because of some European fellows" :)

nohal, 2024-04-23:
ELWIS (DE IENC):
[image]
US Army Corps of Engineers:
[image]

I think we are pretty in line with the latest opinion of the official bodies with the new symbology

bc38, 2024-04-23:
then maybe it makes sense to color the value purple?
still need to stick to the S-52 display library
[image]

nohal, 2024-04-23:
I don't think so, even the Esri viewer on your picture used by NOAA has the value in black. The "km" symbol is part of the standard, but is simply unfortunate...