OpenCPN/OpenCPN #3518: Two finger zoom gesture is not working in OpenCPN on Bookworm, wxwidgets 3.2, .deb packaged OpenCPN 5.8.4

mgrouch, 2023-11-29:
Two finger zoom gesture is not working in OpenCPN on
Bookworm, wxwidgets 3.2, .deb packaged OpenCPN 5.8.4.
Running x11 instead of Wayland on pi4 or pi5. 

Thanks

bdbcat, 2023-12-01:
For reference, does it work on Bullseye?

aviatorhh, 2023-12-01:
> For reference, does it work on Bullseye?

FYI, for me it was working on bullseye with 5.9 (own build) and wx322. Now on Bookworm ist is not working anymore - OC version per today and wx324

bdbcat, 2023-12-01:
Not a surprise, actually.
The X11 emulation on top of Wayland in RPi Bookworm port is known to have some problems.

mgrouch, 2023-12-01:
> Not a surprise, actually. The X11 emulation on top of Wayland in RPi Bookworm port is known to have some problems.

It’s not working in bookworm on X11 without Wayland. (See original report at the beginning). 
There is no Wayland even installed in my case. 
Interestedly long touch for double click works.

mgrouch, 2023-12-01:
> IIUC, there is some interaction between X11 and Wayland on Bookworm. From https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/
> 
> `Our Wayland implementation includes a piece of software called XWayland — this is an X11 display server which sits on top of Wayland. It handles all the non-graphical parts of X itself, and passes any graphical parts to the underlying Wayland implementation. XWayland is designed to launch automatically as soon as an application requests anything from X, so this should all work seamlessly.`
> 
> Key words are "should all work seamlessly". Evidently not.

Again, this bug report is about X11 with no Wayland installed.

bdbcat, 2023-12-01:
It is not clear to me that X11 mode does not include any part of Wayland.  
In any case, if it woks on Bullseye/wx324/O5.9 (see private build above), then it must be that the changes to display manager in Bookworm are somehow involved.
Definitely a bug, possibly in OCPN/wx32

mgrouch, 2023-12-01:
> It is not clear to me that X11 mode does not include any part of Wayland. In any case, if it woks on Bullseye/wx324/O5.9 (see private build above), then it must be that the changes to display manager in Bookworm are somehow involved. Either that, or default X11 input handler configuration. Definitely a bug, possibly in OCPN/wx32

At least in the case for which I reported this bug there is no any parts of Wayland even installed. 
It’s installation of X11 (without Wayland) over raspberry OS lite.

bdbcat, 2023-12-01:
OK, that simplifies things, I think.  Thanks for that info.

nohal, 2024-02-19:
@mgrouch, I have seen a multitude of your posts lately stating that the gestures work on your BBN OS, does it mean you have a solution for this problem on Bookworm already? And if you do, did you have to fix anything on OpenCPN side?

mgrouch, 2024-02-19:
> @mgrouch, I have seen a multitude of your posts lately stating that the gestures work on your BBN OS, does it mean you have a solution for this problem on Bookworm already? And if you do, did you have to fix anything on OpenCPN side?

It’s done in BBN on OS level and it’s working that way in buster, bullseye, bookworm. I didn’t recompile OpenCPN.

nohal, 2024-02-19:
So this issue may be closed?

mgrouch, 2024-02-19:
> So this issue may be closed?

It wouldn’t then work in any Linux distro other than BBN.

nohal, 2024-02-19:
OK, what is the problem in OpenCPN then?

nohal, 2024-02-19:
FWIW, two finger zoom works just fine on my Fedora system with both current master and 5.8.4-3 flatpak.

mgrouch, 2024-02-19:
> OK, what is the problem in OpenCPN then?

It’s missing event handlers for two finger zoom on canvas. It’s missing event handlers for long touch for right click. They exist in WxWidgets 3.2. 

Try installing it on Linux with touchscreen and use it without keyboard and without mouse. Purely on touchscreen. Try selecting all charts in chart downloader on touchscreen.

mgrouch, 2024-02-19:
> FWIW, two finger zoom works just fine on my Fedora system with both current master and 5.8.4-3 flatpak.

Exactly. This report is about same missing in .deb. 
Thus report is not about flatpak version and you can see from the title

nohal, 2024-02-19:
ok, so is this report about a problem about the downstream Debian package or is it about something missing in OpenCPN code? And how did you fix that missing code in BBN if you did not change any code?
Sorry, but I still do not understand what you ask us to do. And no, I do not have a touchscreen available to try to deduce it from the bits you throw at me.

nohal, 2024-02-19:
Cool, so this is at the end not a bug report, not anything to do with the Debian package, but a feature request to implement support for `wxGestureEvent` when built against wx3.1 or newer. Thanks for the clarification.

nohal, 2024-02-19:
It also looks very much like a duplicate of https://github.com/OpenCPN/OpenCPN/issues/2057