OpenCPN/OpenCPN #3527: s52plib: Strange dependencies on main opencpn

leamas, 2023-12-09:
**Is your feature request related to a problem? Please describe.**
s52plib is situated under libs/. Libraries under libs/ are supposed to be self sufficient or possible depending on other libs, but should not have dependencies on main OpenCPN.  Such dependencies defeats the core benefits of a lib w r t testing, reusability  and maintenance.

**Describe the solution you'd like**
Not clear to me at the time of writing.

bdbcat, 2023-12-10:
The inclusion of ocpn_plugin.h is purely for convenience.  Seems reasonable to drop the header depndency, and implement the required function code as a small utility function in s52plib.  Will do that.

Any other objectionable dependencies?

leamas, 2023-12-10:
- ocpn_plugin.h is not problematic. it could be bundled with the library.
  - color_types.h: Could be moved from main opencpn to the library instead without problems, though
  - config.h:  could perhaps be bundled
  

And that's about that. After handling these the inclusion of ../../include can be removed and the library flies. I'm on it.

leamas, 2023-12-10:
hm... actually, depending on ocpn_plugin.h is not trivial at all. It means that the library depends on main opencpn, or the dummy API definitions available in api_shim. Basically, this means that the library cannot be compiled with satisfied library dependencies on Linux and MacOS where we don't have an opencpn library available.

This is a can of worms. However, a reusable S52 library would perhaps be really nice to have, so maybe we should put some efforts in it.

leamas, 2023-12-10:
sorry, missed your comment. Yes, re-implementing stubs from ocpn-plugin.h is the right way.

However, this is not the end of this story. Basically it must compile and link without main opencpn. I have some fixes for the linkage under way, but no idea about the cutil and georef stuff.

leamas, 2023-12-11:
This is not only about s52plib. Also libs/iso8211 has a similar `target_include_directories(ISO8211 PRIVATE ${CMAKE_CURRENT_SOURCE_DIR}/../../include)`from which it uses at leas s57.h

leamas, 2023-12-11:
... and sounds depends on androidUTIL.h  for AndroidSound.

leamas, 2023-12-12:
This is actually just the tip if the iceberg. The work we did with shipdriver and eventually opencpn-libs needs to be done also for the main libraries. Reviewing the CMakeLists.txt I see
```
if (NOT CMAKE_MODULE_PATH)
  set (CMAKE_MODULE_PATH ${CMAKE_CURRENT_SOURCE_DIR}/../../cmake)
endif ()
```

Does not work,  should be removed. It the library actually needs something like FindWhatever.cmake it should be bundled with the library

```´
if (OBJ_VISIBILITY) 
...
```

Is a dependency on main CMakeLists, easily replaced with compiler type conditionals.

```
target_include_directories(GEOPRIM PRIVATE ../../include)
```

Does not work, creates a dependency to main opencpn

```
target_include_directories(GEOPRIM PRIVATE ${wxWidgets_INCLUDE_DIRS})
```
we can assume that wxWidgets is available. However, if using `target_include_directories(` like here the must also be a corresponding `target_link_libraries` for the library to be reusable.

This is basically drive-by issues for while working with  #3513. Added remarks in various libs/*/CMakeLists.txt in e8df1f12e4934cc22fa3f78ab8ca9bd8427539de and moving on

bdbcat, 2023-12-14:
I think we need to break this down into manageable chunks.
I'm working with S52PLIB now. This is important to me, since the essentially identical code is present in o-charts_pi.  I have been trying to keep them in sync.
I have resolved the dependency on ocpn_plugin.h, cutil.h, and georef.h.  Mostly.  Still working.
How (if?) are you testing the standalone buildability of libs/S52plib?

leamas, 2023-12-14:
I havn't been systematically been testing, just stumbled over this when working on #3513. 

 I see no point in testing as long as we can spot problems anyway by  a simple review.

leamas, 2023-12-14:
The test would be to create a dummy cmake project and let the executable link to the library. If that works we should be happy.

leamas, 2024-01-15:
Here is also a mDNS mess, with a library + some files copied to src, referencing library files.

leamas, 2024-01-23:
#3685 should have fixed the simple parts.  Remaining stuff (at least)

  - sound
  - s52plib
  - mDNS

None of these are simple fixes

PR #3619 (2024-02-07): Libraries: Remove dependencies on main code (#3527)
Remove various library dependencies on main code with the goal the libraries should be self-contained.

Notable changes:
  - androidUTIL is split into two parts. The new library _android_jvm_ contains the singleton reference to the JNI java_jvm  + some stateless accessors which does not depend on OpenCPN. The rest is left in android_UTIL
  - The platform dependent GL libraries are refactored (again!) in to a new library _gl_headers_.

... and lots of minor stuff.

Closes: #3527
closed 2024-02-07 by leamas