OpenCPN/OpenCPN #3532: UI Problem with tracking

rgleason, 2023-12-12:
https://www.facebook.com/groups/2405863179652854/permalink/3518567378382423/?mibextid=RtaFA8

nohal, 2023-12-12:
We are not all members of this private FB group, please provide the information here.

rgleason, 2023-12-12:
Sorry, on cell. Will fix this later as standalone issue.

rgleason, 2023-12-13:
This report found on facebook Opencpn users group.

Does anyone know why the Record Track button on my AIS Target Query dialog box is always greyed out? There are times where I'd love to use this feature, but can't figure out why the feature isn't enabled on my system. 

OpenCPN 5.8.4-0+1637c28
MacOS Sonoma 14.1.1 
Raymarine AIS350 Class B Transceiver 

Does this work for others?

nohal, 2023-12-13:
As a workaround the track can be enabled from the AIS target list.

rgleason, 2023-12-13:
Yes but this is nonsensical as an interface, users find it very very strange and point to it as a weakness in the UI. and it is IMHO.

nohal, 2023-12-13:
Nobody argues about this being a bug.

nohal, 2023-12-13:
BTW the response given there is wrong, it does not do what the guy asks for but enables temporary short track for all the AIS targets.

rgleason, 2023-12-13:
Ok, sorry, but I have not tested it myself.

rgleason, 2024-01-24:
Pavel, I have not taken time to delve into this. Should I close this now? Thanks.

nohal, 2024-01-24:
Has anybody fixed it?

commit 5114136b (2024-01-25): Always enable "Record Track" button on AIS Target Info dialog.  Closes #3532