OpenCPN/OpenCPN #3543: Not calling Destroy on wxFrame derived

Kitiara, 2023-12-16:
Hello,

I have experienced a crash related with the call **MyFrame::RequestNewMasterToolbar**. I read your code and noticed that you are deleting a wxFrame derived class instead calling the **wxFrame::Destroy(**).

https://github.com/OpenCPN/OpenCPN/blob/089c8b60570f867154a2522936be5a24bcd1e2e2/src/ocpn_frame.cpp#L7005

It looks like a mistake.

leamas, 2024-02-07:
Hi, 

I'm not aware of any restrictions on using `delete` as long as it's not about handling the  `Close` event i. e. it's about shutting down `*this`.

This is certainly not about handling `wxCloseEvt`. Given this, can you describe what's wrong with using `delete`  here?

And, not least: can describe a reproducable way to show that the use of `delete` causes any actual problem?

bdbcat, 2024-02-08:
OP makes a good point, though it might be difficult to reproduce an error.
The exposure is that if a wxEventHdlr derived class (as all wxWindows are) is simply "deleted", there may be unprocessed events in the wx event queue that may later be sent to the just deleted handler.  Guaranteed crash.
As I say, difficult to reproduce.  And non-trivial to fix properly, due to those pesky global definitions for the window pointer.
Anyway, a valid point of exposure to consider.

Kitiara, 2024-02-08:
> Hi,
>
> I'm not aware of any restrictions on using `delete` as long as it's not about handling the `Close` event i. e., it's about shutting down `*this`.
>
> This is certainly not about handling [...]

The documentation of wxWidgets is pretty clear about the deletion, [here it is](https://docs.wxwidgets.org/3.0/classwx_window.html#a6bf0c5be864544d9ce0560087667b7fc). To reproduce the issue, you need to open up the settings window, click apply and then close. Doing that several times in a row is how i managed to reproduce the issue before.

leamas, 2024-02-08:
OK. got it, thanks.

bdbcat, 2024-04-22:
Global toolbar has been refactored, no longer a wxDialog.
AIS TargetQuery dialog deletion method corrected.
Ready to close.

closed 2024-04-23 by bdbcat