OpenCPN/OpenCPN #3549: Plugins disappear from plugin manager when disabled

nohal, 2023-12-18:
**Describe the bug**

When a plugin is disabled, it is removed from the plugin manager list and does not appear again until `opencpn.conf` is manually edited and `bEnabled` is set to `1` for the respective library.

I expect that usable locally available plugins should be visible regardless of being enabled or not.

leamas, 2023-12-20:
hm... this mess seems to be limited to the built plugins. Non-built in ones seems to be OK, both with respect to this issue and #3550.

But the handling of the built-in ones is definitely broken in all sorts of ways.

leamas, 2023-12-20:
Have you got an example plugin so I can reproduce?

nohal, 2023-12-20:
One more detail, this now seems to apply only to plugins that were not installed using the plugin manager = were placed to `~/.local/lib/opencpn/` some other way, for example manually by myself (normal during development of plugins).
When I now first import the plugin from a tarball, it seems to stay in the list even after being replaced by a different binary. Do not exactly understand why, because metadata for an older version of the plugin with the same name did exist before already when the problem was present and the new ones seem to look the same except the version number.
But the version number is not the issue as changing `~/.opencpn/plugins/install_data/<plugin>.version` to a complete nonsense does not break it again.

leamas, 2023-12-20:
>  were placed to ~/.local/lib/opencpn/ some other way, for example manually by myself

Ah... this is just not a supported usecase. Sorry.

This explains a lot of things, I think. When we import plugin the bundled metadata in the plugin is added to the catalog and stays  there until  plugin is  uninstalled.

Another thing is that a plugin which is disabled just isn't loaded, and  no info from the plugin interface is available in this case.

Final piece: the version displayed depends on the plugin API level. From 1.18 (1.17?)  the full version is available from the plugin and it is used. For older API levels, the full version is taken from the catalog metadata.

As I read this, all the symptoms you describe  seems to be caused be just dropping a library into _~/.local/lib/opencpn_. That this doesn't work is expected.

The core issue here is of course that it is too complicated to create an importable tarball. Shipdriver has a reasonable support in `make tarball`  but otherwise it's  a bit convoluted, agreed.

nohal, 2023-12-20:
I am using API 1.18

leamas, 2023-12-20:
OTOH, I have new bug: The built in plugin seems to have disappeared. Bisected down to 01ec5466a0c9aabdb3a

leamas, 2023-12-20:
> I am using API 1.18

Yes. But the core issue here: is this only about what happens when you manually drop something into  _~/.local/lib/opencpn_, or is it related to supported use case where you install a plugin from the catalog or imports a full tarball?

nohal, 2023-12-20:
So this is how it works for me:
- if a plugin is disabled AND `/home/nohal/.opencpn/plugins/install_data/imports/<plugin>.xml` does exist, it is in the list (and a possibly wrong version from `~/.opencpn/plugins/install_data/<plugin>.version` is shown)
- if a plugin is disabled AND `/home/nohal/.opencpn/plugins/install_data/imports/<plugin>.xml` does NOT exist, it is not the list
- if a plugin is enabled, it is in the list with the correct version it reports in the API, works and nothing cares about the (non)existence of `/home/nohal/.opencpn/plugins/install_data/imports/<plugin>.xml` at all.

leamas, 2023-12-20:
... and all this is as expected.

nohal, 2023-12-20:
Well, for me the last point seems completely inconsistent with the first two, but I can sure live with it, now that I know how this actually works and that is is what we want...

leamas, 2023-12-20:
OK. Close?

nohal, 2023-12-20:
One more related issue, imported tarballs don't seem to be put into `~/.opencpn/plugins/cache/tarballs/` so any attempt to reinstall them leads to a download request, which in this scenario usually means a non-existent URL

leamas, 2023-12-20:
Shall we make a discussion of this? I don't see the well-defined bug here...

nohal, 2023-12-20:
We sure can make this anything, preferably working in a way that does not generates this issue at all...