OpenCPN/OpenCPN #3550: Plugin manager showing version of disabled plugins as 0.0

nohal, 2023-12-18:
The actual version should be shown.

[image]

nohal, 2023-12-20:
The problem is slightly different for plugins installed by the manager, there the version does not change to `0.0` when they are disabled, but to whatever there is in `~/.opencpn/plugins/install_data/<plugin>.version`

nohal, 2023-12-20:
[image]

[image]

leamas, 2023-12-20:
See discussion in #3549.  This is actually as expected .

bdbcat, 2024-01-04:
Resolved?

nohal, 2024-01-04:
I think we are done here, at least until proven wrong...

closed 2024-01-04 by nohal