OpenCPN/OpenCPN #3557: Mark icons scaled to big in 5.9

Hakansv, 2023-12-25:
**Describe the bug**
Some mark icons are drawn to big using master 5.9
**To Reproduce**
Steps to reproduce the behavior:
1. Create a new WP near a WP created by O 5.8
2. Select e.g. 1st-Diamond 
3. Compare the new icon size with the old.

**Expected behavior**
Some icons in the select list are two big sized. Scale them "as usual".

**Screenshots**
See below

**Desktop (please complete the following information if applicable):**
 - OS: Win10 and Rpi-Debian

[image]

[image]

bdbcat, 2023-12-25:
Not apparent on linux.  Windows only?
This looks like it needs a (long)  "git bisect" to isolate.

Hakansv, 2023-12-25:
No I see the same on Rpi -Debian
Bisect - - Hmm. This I've seen long ago. Summer? I've forgot to mention all times.

bdbcat, 2023-12-25:
It will take me a while to build on rPI.  I'll come back.

Hakansv, 2023-12-25:
I haven't checked to edit but found this from 30 June.  A trace?

> @@ -485,10 +485,13 @@ MarkIcon *WayPointmanGui::ProcessLegacyIcon(wxString fileName, const wxString &k
>                                                  floor(displayDPmm * 12.0));
>    // legacy icon size
>    float pix_factor = nominal_legacy_icon_size_pixels / 68.0;
>  
> -  wxBitmap img = LoadSVG(fileName, -1, -1);
> -  bm_size = img.GetWidth() * pix_factor * g_MarkScaleFactorExp;
> +  unsigned int w, h;
> +  SVGDocumentPixelSize(fileName, w, h);
> +
> +  bm_size = w * pix_factor * g_MarkScaleFactorExp;
> +
>  #endif

bdbcat, 2023-12-25:
Definitely in the correct region.  Experiment?

Hakansv, 2023-12-25:
Of course. Suggestion what to test?

bdbcat, 2023-12-25:
First, compare the size of default icon to something like a standard buoy.  For me, they are about the same in both 584 and 590
[image]

Hakansv, 2023-12-25:
Win10:

[image]

bdbcat, 2023-12-25:
Must be something related to display scaling.  I cannot reproduce on w10 at 125% or 150%.

Hakansv, 2023-12-25:
Win10:
My normal scale is 175. I tested 150 and 200 - the same result. 
Are there some wxWidgets dependent issue now when on >=3.2.2? The extensive scaling work for 5.8 from last year or more was on wx <= 3.2.1 I think.
Ooh-- And I'm sorry my bad: All my RPi are sized as normal.  I wasn't comparing RPi but another Win10. To many AD windows open I suppose.

bdbcat, 2023-12-26:
I'm sure you checked, but:
Settings->UI scale factors all set to zero?

Hakansv, 2023-12-26:
My UI Scale factor is -2 but any change has no effect on WP size.
Chart object scale factor does has impact on WP size but, yes, that is here set to zero.

Hakansv, 2023-12-26:
I've tried to Debug and see where the icon size is set during Create WP - MyFrame::DropMarker() but lose the trail all time? What I can see it's not around the code snippet mentioned above. 
Hints?

bdbcat, 2023-12-26:
The icons are preloaded at startup.
Place a break on the snippet, and restart O.
Anydesk?

commit 15956ef8 (2023-12-30): Correct extended SVG icon sizing logic, Closes #3557
closed 2023-12-30 by bdbcat