OpenCPN/OpenCPN #3559: Settings window interface on raspberry

inhalelol, 2023-12-27:
**Describe the bug**
The interface breaks in the settings window in the display-units and display-advanced tabs if Russian is selected

**To Reproduce**
Steps to reproduce the behavior:
1. Select Russian
2. Relaunch OpenCPN
3. Click on Settings

**Expected behavior**
Display interface like in English

**Screenshots**
[image]
[image]

**Desktop (please complete the following information if applicable):**
 - Device: raspberry pi4 model b
 - OS: Raspbian (openplotter)
 - Version 5.9.0

rgleason, 2023-12-27:
Perhaps the text is too long?

nohal, 2024-02-09:
I'm not able to reproduce the problem so far. How exactly did you install OpenCPN 5.9.0 on your system?

inhalelol, 2024-02-19:
I installed the OpenPlotter image on the Raspberry Pi, which already had OpenCPN pre-installed. Unfortunately, I don't have access to the Raspberry Pi at the moment, and I can't recall if there was a similar issue in the OpenPlotter image initially. Afterward, I removed OpenCPN and compiled it from the source code, following the [instructions](https://opencpn-manuals.github.io/development/ocpn-dev-manual/5.3.1/od-compile-linux.html). However, I did notice the same problem on Ubuntu 22.04. In this case, I compiled OpenCPN from sources as per the [instructions](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_linux). Version 5.8.4, installed from the ppa repository, does not exhibit this problem.
[image]

nohal, 2024-02-19:
So this last screenshot is stock up to date Ubuntu 22.04 and a build of current master branch, right? From the two "build instructions" you linked, only https://opencpn-manuals.github.io/main/opencpn-dev/linux.html is the correct one, the other mentioned later is outdated (and I just removed it)

inhalelol, 2024-02-19:
Yes, that's right. I followed the correct instructions to build it for Raspberry earlier, and now I have rebuilt it for Ubuntu, however, the problem persists.

nohal, 2024-02-19:
OK, I am able to reproduce the Unit tab problem on Ubuntu 22.04 (Not so on the advance tab, that is correct), goes away immediately after any even like resize or scroll, so it seems the event causing the layout does not arrive when the dialog is created. Will have a look what can be done. Thanks for the info.

nohal, 2024-02-19:
After changing screen resolution I can see it even on the Advanced tab. Definitely a wxWidgets bug, but we should be able to work it around.