OpenCPN/OpenCPN #3562: OpenCPN 5.8.4 UI on Ubuntu 22.04 Dark mode is broken

mpuz, 2023-12-30:
When ubuntu standard dark mode is on - the menu fonts (at settings menu for example) are drawn dark on the dark background, also toolbar button's popover tooltips are drawn as light font on white background

**To Reproduce**
1. Ubuntu - go to settings, appearance, choose dark theme
2. Start OpenCPN, open Settings - dark font on dark background totally unreadable
3. Mouse over any toolbox buttons - popover is unreadable when light OpenCPN mode choosen - font is the same light color as background
[image]
[image]
[image]
[image]

 - OS: Ubuntu 22.04
 - Lenovo Yoga 720 ikb

nohal, 2024-02-11:
### Note to self
Big part of this is a wxWidgets 3.0 problem fixed in 3.2.
The custom widgets in the options dialog need to to be updated to reflect the system theme back/foreground colors instead of assuming dark foreground/light background.

nohal, 2024-02-13:
As far as I'm concerned the core application is fixed now.
The embedded Dashboard plugin is fully configurable, but perhaps should be improved to change the defaults according to the system theme.
3rd party plugins, like the Odometer, need to be fixed by their respective authors and need their own issues in the respective projects.

bc38, 2024-02-13:
There is no solution with Dark mode for Windows?

nohal, 2024-02-13:
No, there is nothing we can currently do on Windows. In the current master of wxWidgets initial support for dark mode on Windows is present, but there is not even an alpha release of wx 3.3 available yet so we can't switch to it for the next OpenCPN release, it would be an absolute disaster for the plugin ecosystem.

Rodemfr, 2024-02-17:
I'm also looking at fixing some colors here or there for dark theme but I'm facing an issue/dilemma for which some advice would be welcome.

I'm trying to fix some colors in tide/current window for dark theme :

[image]

The background color of the graph is debatable in case of a dark theme. Also some legends are drawn black on dark gray and the high/low tide time table on upper right has a light background. No problem, I can propose changes for that, but what strategy to use ?

Looking to the current code, I see that s52plib is used to get the colors for some of the drawings. The advantage of using this lib is that it automatically takes into account the color scheme (day, dusk, night). But the drawback is that not every color is taken from this library, especially the default window colors which are using wxWidgets palette. wxWidgets adapt its colors to the color scheme of the system (light/dark). With this mix of colors, it is somewhat complex to find a suitable set of colors for light/dark/day/dusk/night configurations. So comes my question : is there a preferred way to choose colors ? s52plib, wxWidgets palette ?

My intuition is that we should let s52plib handle colors for chart rendering and use wxWidgets palette for window rendering. It is mostly motivated by the fact that s52plib can sometime do color inversion in night mode, which make almost impossible to find a set of colors that fits wxWidgets window colors. The problem of this approach is that wxWidgets knows nothing about color scheme so it might open a super bright white window in night mode. We could handle that by creating a small set of color functions getting colors from wxWidgets and applying a luma factor for dusk and night, as it is done for mbtiles rendering.

Any opinion on this ?

nohal, 2024-02-17:
I also use the diming method for the gauges in DashboardSK plugin and am pretty happy with the outcome.

Rodemfr, 2024-02-18:
Thanks for the explanation. Very clear.
I modified tide window colors in #3670 following your advises. It greatly improves rendering in dark mode.
I have seen that `navutil` is not modifying window colors in case of dark mode. I think it should because the text color is too bright when in night scheme. I didn't include this modification in the pull request because it is impacting the entire software and I'm unsure of the consequences. I will have a look to it now.

nohal, 2024-02-18:
Yes, it proably should. But if the system theme already is dark maybe just bright colors like the text in Adwaita GTK theme.

mpuz, 2024-02-22:
i have built from source - it's perfectly fine now! thanks a lot
[image]

closed 2024-02-25 by nohal