OpenCPN/OpenCPN #3566: Reestablish TCP N0183 connections immedeately after a Windows suspend.

Hakansv, 2023-12-30:
**Describe the bug**
After a Windows suspend or hibernation any TCP connection is interrupted.
Present code is then timed out and wait a long time before any attempt to restart the connection. 
Also compare the now closed PR #3356. (I think we need a more general view on this)

**To Reproduce**
Steps to reproduce the behavior:
1. Establish a TCP N0183 connection. 
2. Let Windows suspend/hibernate
3. Wake the PC up.
4. Any TCP connection is timed out and need a restart. 

**Expected behavior**
A Window suspend is detectable by wxWidgets "wxHAS_POWER_EVENTS"
When this apply we could restart the connection immediately or after first timeout. No need to wait.
Maybe elaborate thoughts indicated [here](https://github.com/OpenCPN/OpenCPN/blob/a621fe901915fd2ac9b98223ef8a5eec6aebd86d/gui/src/ocpn_frame.cpp#L6986)

**Desktop (please complete the following information if applicable):**
 - OS: Windows. 
 - Not sure about comparable events on MacOS. Not seen on RPi-Debian.

**Smartphone/Tablet (please complete the following information if applicable):**
 - Not seen.

**Additional context**
The connection watchdog is now 5 seconds. We may consider a more dynamic timer. It's not unlikely a connection can have far longer pauses between messages. I've seen e.g. AIS receivers message very few targets with minutes in between.

leamas, 2024-01-01:
This is related to he more general #3407.

leamas, 2024-01-26:
hm... The core issue is the handling of the resume signal. If something should happen "immediately" this must be event driven, polling is just nit the way to achieve this.

I have been working on #3407 and now have the infrastructure in place for that + linux implementation.  That is, once merged  Linux will have an EventVar which is notified when  there are new USB devices attached. Drivers could listen to this and make the proper decisions.

Looking at resume, there is some wxWidgets handling of that, but only for wxWindows. Currently, this is OnResume()  in ocpn_frame. It would be just a few lines to create an EventVar which is notified about this. And if would also be easy to expand the linux listener to also handle suspend signals, it's just anotherl DBus signal and not much work.

All in all: the correct solution would be to make the resume signal more generally available using an EventVar and then do the right  thing on that signal instead of messing with polling and timeouts which certainly is a mixed blessing.  This solution is actually not far away once i have something working for #3407, perhaps next week.

Hakansv, 2024-01-26:
I'll of course test if your change will take of also this. 
And BTW: I mentioned TCP here but the same happens for Signalk. And that's also a socket connection I suppose.  There's a PR in queue about that socket. Let's see...

leamas, 2024-01-26:
Do you have a reference for that SignalK PR?

Hakansv, 2024-01-26:
PR #3207

leamas, 2024-01-27:
that PR does not seems to be related to this. Resolving this shoudl probably by doen at the driver level; #3207 is not touching  drivers at all.

leamas, 2024-01-28:
@Hakansv 

Well, since you offered some testing  cycles: the time has come to #3610. It's actually two parts: the network driver and the serial one.

As for the network driver I feel pretty confident. Thinking of two tests:
1. Overall testing of the driver, verifying that it works as before.
2. Verify that it reconnects after a suspend-resume cycle.

If you feel mystified, the flow after a resume is:
  - `MyFrame::OnResume(...)` in _ocpn_frame.cpp_ is invoked by wxWidgets code.
  - OnResume() calls `SystemEvents::GetInstance().evt_resume.Notify();`
  - The network driver has this snippet: 
```
+  auto resume_action = [&](ObservedEvt&) {
+    wxTimerEvent evt;
+    OnTimerSocket(evt); };
+  resume_listener.Init(SystemEvents::GetInstance().evt_resume, resume_action);

```
which basically means that when evt_resume is notified by OnResume() it will run the code in `resume_action` which, if I get this right,  will restart the driver (knock, knock...)

Let's start with this, leaving the serial driver for now.

Hakansv, 2024-01-30:
Fixed by PR 3610

PR #3610 (2024-02-07): Hardware events handling
Add handling of hardware events such as resume after suspend and new USB devices being attached.

The central piece is the new SystemEvents, a new evil singleton. It is a container for EventVar, one for each event. 

On Windows, already basically existing code notifies `evt_resume`  when resuming from suspend. New code is added which listens to native events for new/removed USB devices and notifies `evt_dev_change`.

On Linux, a new daemon listens for DBus events and notifies  `evt_resume`  and  also `evt_dev_change`. The latter is emitted when new devices are detected [... cut]
closed 2024-01-30 by Hakansv