OpenCPN/OpenCPN #3570: ENC update fails; File couldn't be removed

marcobergman, 2023-12-31:
On a clean windows install, when this new ENC chart is added, one chart is repeatedly 'prepared'. The log file shows each time:
```
00:33:06.184 MESSAGE Osenc.cpp:1166 WARNING---ENC Update failed.  Last valid update file is:C:\ProgramData\opencpn\SENC\1R7WAD03.001
00:33:06.184 MESSAGE Osenc.cpp:1167    This ENC exchange set should be updated and SENCs rebuilt.
00:33:06.184   ERROR filefn.cpp:1045 File 'C:\ProgramData\opencpn\SENC\1R7WAD03.000' couldn't be removed (error 32: The process cannot access the file because it is being used by another process.)
```
To reproduce:

- Install OpenCPN 5.8.4 on Windows
- Add chart source name=RWS, URL=https://www.fredhen.nl/rwsenc.xml, Chart Installation Directory = C:\rwsenc\ENC_ROOT
- Update Chart Source, select Inland_Waddenzee_week 52_NL, Download Selected Charts, OK.
- Hamburger menu: Show Chart Outlines.
- Pan over all available charts. They will be Prepared, one by one. One chart is likely to be repeatedly prepared.

[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/13802363/opencpn.log)

[image]

transmitterdan, 2024-01-01:
I can confirm there is some kind if issue here.  I haven't found the root cause yet. But initial indications are it is a heap corruption problem and probably nothing wrong with the chart itself.

FYI, the version info is: 5.8.4-0+1637c28

marcobergman, 2024-01-02:
It turns out this also occurs on Ubuntu 22.04, and here I can compile OpenCPN from source and debug. Any hints as to what piece of source code I'd need to look into?

leamas, 2024-01-02:
No hint, but there is always the brute-force `git bisect` approach,  assuming that this is some kind of regression.  The first check would then be to checkout previous version `Release_5.8.2`, compile it and check if this bug exists also there.  If so, bisecting should reveal the offending commit.

If the bug  does not exist in `Release_5.8.2` one could perhaps make a try also with  `Release_5.8.0`  for a possible working start point.

bdbcat, 2024-01-04:
Not a systematic error.  Occurs on one chart cell only.
There is an error applying one update to the referenced chart cell "1R7WAD03.000".
To workaround temporarily, you may delete/rename the update file 

`"rwsenc/ENC_ROOT/1R/7/1R7WAD03/1R7WAD03.002'`

Just change ".000" to ".000sav"

I will investigate further.

bdbcat, 2024-01-04:
Debugged this.
It is an error in the cell update file, .002.  The update file instructs the deletion of a record in the cell database that does not exist.  Not a common error in ENC update chain processing, but not unknown, either.

So, the root cause lies with the HO providing the cells.
But the OCPN question presents:  How should we handle this sort of case?  As Marco reports, we log the error as a "warning", but attempt to use the latest available cleanly updated cell.
Should we do more/other action?
Thoughts?

transmitterdan, 2024-01-04:
I don't think this is a regression. I could be wrong but it's so rare to have a bad HO created file that this behavior could have been in the code for years.

marcobergman, 2024-01-04:
Would it make any sense for me to try to report this back to my HO?

marcobergman, 2024-01-04:
From a user point of view I'd say make some message, but prevent the behavior I saw - repeatedly preparing the chart. That makes people not use the chart and complain about it in online forums, which I'm trying to defuse because it looks bad on OpenCPN.

bdbcat, 2024-01-05:
@tDan
No recent changes in that particular code module. So I agree not a regression.  But HO errors are not so rare as all that.  They happen, and several times we have helped the HO identify and correct the cell errors.
I'll dig into the specific details, leading to...

@Marco...
If you have a good technical contact at the HO, I'll provide some technical info on the error as observed.  And we shall see how they respond.
Please stand by.

On the user side, nothing to be done until O5.9+ ships.  I agree that we could do a better job of messaging here.  Fortunately, as Dan says, HO errors are rare.

bdbcat, 2024-01-05:
Marco...
Here is my analysis of the update file (.002) failure:

Error diagnosis on ENC update
-----------------------------

Update file name:  1R7WAD03.002

Record 1.  "DSID"
                Decodes OK

Record 2.   "VRID"      (Vector record)
                RCNM = 120
                RCID = 385703105 (0x16fd5cc1)
                RVER = 2
                RUIN = 2 (delete record, target is RCID)
                
                    Error, cannot find record 385703105.
                    
                    
Record dump follows:
----------------------------------------------
DDFRecord:
    nReuseHeader = 0
    nDataSize = 25
    _sizeFieldLength=1, _sizeFieldPos=1, _sizeFieldTag=4
  DDFField:
      Tag = `0001'
      DataSize = 3
      Data = 
\02\00\1E'
  DDFField:
      Tag = `VRID'
      DataSize = 9
      Data = 
\78\C1\5C\FD\16\02\00\02\1E'
      Subfield `RCNM' = 120
      Subfield `RCID' = 385703105
      Subfield `RVER' = 2
      Subfield `RUIN' = 2
----------------------------------------------------

marcobergman, 2024-01-05:
Thx. From the code of osenc.cpp I read that OpenCPN keeps on reprocessing the cell _by design_:

  // The logic will attempt to build a SENC on each instance of OCPN, so
  // eventually the updates may be corrected, and the chart SENC is built correctly.

I read '_on each instance of OCPN_' as '_each time OpenCPN is started_'. But it looks like it keeps processing it more often than that: it seems to process it after panning away and panning back to the cell concerned. I cannot replicate this consistently, though.

But rather than to keep on reprocessing the cell by design, I would suggest to call it quits until the next chart update. I don't expect end users to manually replace an offending update file, so reprocessing is not opportune in  most cases, and the result is confusing to end users. They have received a warning anyway, and there is always the button 'prepare all ENC charts'.

Also the ERROR `file couldn't be removed' does not seem to be appropriate when the exception was already handled.

bdbcat, 2024-01-05:
Marco...
Originally, OCPN tried to rebuild the SENC file only once per session.  Somewhere along the way, this behavior changed.  I'll see what happened there.  I think this (once per session) is a suitable solution, since the "Building SENC" yellow indicator gives at least some idea that there is an anomaly.

Also, the ERROR `file couldn't be removed' is certainly misleading.

bdbcat, 2024-01-08:
Marco...
Do you expect any information regarding this update error from the HO?

marcobergman, 2024-01-08:
I'm afraid not, can't find any direct contacts so I had to submit the issue into some response page, that can take months. The weekly update is 4 days late now, so I can't tell whether it was a one-off glitch either.

marcobergman, 2024-01-14:
A new release of HO's chart set has been released and it still contains the offending update file [1R7WAD03.002](https://github.com/OpenCPN/OpenCPN/issues/3570#issuecomment-1877715545). I guess such bad files must be seen as a fact of life.

leamas, 2026-09-13:
@bdbcat wrote [above](https://github.com/OpenCPN/OpenCPN/issues/3570#issuecomment-1877749366)

> Or, looked at another way: What UI action should we present so that the user understands that this is a chart bug, and not an OCPN bug? Thoughts?

Isn't this a good use case for the new notification system?

Hakansv, 2026-09-13:
> Isn't this a good use case for the new notification system?

I'm not sure about that. The notification function is mainly for informing about any problems related to navigation and where the user can reasonably be expected to act, such as a dropped connection to a GNSS receiver. An error in a map file is not something the user can normally do anything about while under way. 
Better with a relevant text for the log file?