OpenCPN/OpenCPN #3573: Build with sanitizers broken

nohal, 2024-01-03:
**Describe the bug**

Multiple components are not linked with sanitizers

```
cmake -DENABLE_SANITIZER=address ..
make
```

```
/usr/include/c++/13/bits/stl_vector.h:367: undefined reference to `__asan_report_load8'
/usr/bin/ld: /usr/include/c++/13/bits/stl_vector.h:367: undefined reference to `__asan_report_load8'
/usr/bin/ld: ../libs/observable/lib_OBSERVABLE.a(observable.cpp.o): in function `_sub_D_00099_0':
/home/nohal/source/OpenCPN/libs/observable/src/observable.cpp:120: undefined reference to `__asan_unregister_globals'
/usr/bin/ld: ../libs/observable/lib_OBSERVABLE.a(observable.cpp.o): in function `_sub_I_00099_1':
/home/nohal/source/OpenCPN/libs/observable/src/observable.cpp:120: undefined reference to `__asan_init'
/usr/bin/ld: /home/nohal/source/OpenCPN/libs/observable/src/observable.cpp:120: undefined reference to `__asan_version_mismatch_check_v8'
/usr/bin/ld: /home/nohal/source/OpenCPN/libs/observable/src/observable.cpp:120: undefined reference to `__asan_register_globals'
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/dbus_tests.dir/build.make:220: test/dbus_tests] Error 1
make[2]: Leaving directory '/home/nohal/source/OpenCPN/build'
make[1]: *** [CMakeFiles/Makefile2:2361: test/CMakeFiles/dbus_tests.dir/all] Error 2
```

```
/home/nohal/source/OpenCPN/model/src/logger.cpp:43: undefined reference to `__asan_report_store8'
/usr/bin/ld: /home/nohal/source/OpenCPN/model/src/logger.cpp:43: undefined reference to `__asan_report_store8'
/usr/bin/ld: /home/nohal/source/OpenCPN/model/src/logger.cpp:43: undefined reference to `__asan_report_store8'
/usr/bin/ld: CMakeFiles/wx-instance.dir/__/model/src/logger.cpp.o: in function `std::_Rb_tree_header::_M_reset()':
/usr/include/c++/13/bits/stl_tree.h:210: undefined reference to `__asan_report_store8'
/usr/bin/ld: CMakeFiles/wx-instance.dir/__/model/src/logger.cpp.o:/usr/include/c++/13/bits/stl_tree.h:209: more undefined references to `__asan_report_store8' follow
/usr/bin/ld: CMakeFiles/wx-instance.dir/__/model/src/logger.cpp.o: in function `_GLOBAL__sub_I__ZN7OcpnLog12LOG_BADLEVELE':
/home/nohal/source/OpenCPN/model/src/logger.cpp:140: undefined reference to `__asan_stack_malloc_4'
/usr/bin/ld: CMakeFiles/wx-instance.dir/__/model/src/logger.cpp.o: in function `std::_Rb_tree_header::_M_reset()':
/usr/include/c++/13/bits/stl_tree.h:211: undefined reference to `__asan_report_store8'
/usr/bin/ld: CMakeFiles/wx-instance.dir/__/model/src/logger.cpp.o: in function `__static_initialization_and_destruction_0':
/home/nohal/source/OpenCPN/model/src/logger.cpp:43: undefined reference to `__asan_report_store8'
/usr/bin/ld: CMakeFiles/wx-instance.dir/__/model/src/logger.cpp.o: in function `std::_Rb_tree_header::_M_reset()':
/usr/include/c++/13/bits/stl_tree.h:208: undefined reference to `__asan_report_store8'
/usr/bin/ld: /usr/include/c++/13/bits/stl_tree.h:211: undefined reference to `__asan_report_store8'
/usr/bin/ld: CMakeFiles/wx-instance.dir/__/model/src/logger.cpp.o: in function `std::less<unsigned long>::operator()(unsigned long const&, unsigned long const&) const':
/usr/include/c++/13/bits/stl_function.h:408: undefined reference to `__asan_report_load8'
/usr/bin/ld: /usr/include/c++/13/bits/stl_function.h:408: undefined reference to `__asan_report_load8'
/usr/bin/ld: CMakeFiles/wx-instance.dir/__/model/src/logger.cpp.o: in function `_sub_I_00099_1':
/home/nohal/source/OpenCPN/model/src/logger.cpp:140: undefined reference to `__asan_init'
/usr/bin/ld: /home/nohal/source/OpenCPN/model/src/logger.cpp:140: undefined reference to `__asan_version_mismatch_check_v8'
/usr/bin/ld: /home/nohal/source/OpenCPN/model/src/logger.cpp:140: undefined reference to `__asan_register_globals'
/usr/bin/ld: CMakeFiles/wx-instance.dir/__/model/src/logger.cpp.o: in function `_sub_D_00099_0':
/home/nohal/source/OpenCPN/model/src/logger.cpp:140: undefined reference to `__asan_unregister_globals'
```

leamas, 2024-01-03:
My bad, I havn't had it in my mind. I usually use gcc, so no test case. However, there is a new interface library ocpn::opencpn which is linked to many of the new targets, adding `target_link_libraries(_opencpn INTERFACE -fsanitize=${ENABLE_SANITIZER})` should fix quite  a few The model is a notable exception.

nohal, 2024-01-03:
GCC supports most of the sanitizers, probably the only exception is `-fsanitize=memory` which is only available with LLVM

leamas, 2024-01-03:
OK, I will spend some time on this tomorrow. Claiming issue, grab it if you want to do it yourself,

commit 556d8108 (2024-01-04): Build: link to suitable sanitizers libs (#3573)

Add "linkage" to possible  -fsanitize=whatever defined in
ENABLE_SANITIZER.

No checks for actual asan libraries availability added, this is left
to developer's discretion.

A bit clumsy, some test executables should not really be using this,
but kept simple. The original test was using MATCHES but it makes no
sense to test against what "none" matches to, using an equality test
instead.

Closes: #3573
closed 2024-01-04 by leamas