OpenCPN/OpenCPN #3577: Enhance Range Rings Functionality: Default size and on/off switch

svMigration, 2024-01-05:
**Is your feature request related to a problem? Please describe.**
I use range rings all the time -- especially in tight areas when using satellite MBTiles. They are used for OWN SHIP and Waypoints. It would be nice to make them easier to use.

**Describe the solution you'd like**
I would like to set the default size for waypoint range rings in the INI file. That way I don't have to enter it every time. It currently takes 7 clicks/data entries to set the range rings for a waypoint. Or perhaps the range rings distance could be retained from the last setting as it does for the OWN SHIP range rings.

Also, perhaps a checkbox for the waypoint range rings would be great so you could turn them on and off quickly.

Additionally, a keystroke to turn OWN SHIP range rings on and off would be super useful. It's currently 5 mouse clicks to turn them on or off.

**Describe alternatives you've considered**
I just use a lot of keystrokes/mouse clicks to set the range rings on and off.

**Additional context**
Thank you!

Hakansv, 2024-01-06:
Did you tried to change the default settings for WP range rings with the WP properties tool?

[image]

svMigration, 2024-01-06:
Excellent suggestion, Hakan. Thanks. It does work if you turn RR on, change the size. Then turn them off (in the default). Thank you.

svMigration, 2024-01-06:
Hi Hakan,

Brilliant suggestion. I changed defaults to what I wanted. Then turned range rings off and it retained the default RR size.

Thank you.

Bruce 

 

svMigration.com <https://svmigration.com/> 

 <https://chartlocker.brucebalan.com/> The Chart Locker

 <https://www.lifehack.org/articles/productivity/7-reasons-why-quitting-facebook-now-good-for-your-future.html> Quit Facebook. Disconnect from the net;  <https://www.humanetech.com/> reconnect to the world.

 

bild.png (view on web) <https://github.com/OpenCPN/OpenCPN/assets/7202854/271a0144-b217-4b18-9cff-7af519230858>

nohal, 2024-02-10:
Do we have some work to do here or is the current behavior OK?

svMigration, 2024-02-10:
Hakan's selection solves the default ring size for waypoints. But it would still be nice if the SHIP issue could be addressed:

A keystroke to turn OWN SHIP range rings on and off would be super useful. It's currently 5 mouse clicks to turn them on or off.

Thanks for following up.
Cheers,
Bruce

nohal, 2024-02-17:
Keyboard shortcut `R` implemented in 6fe4d8998cb6b7c34ee707831a4e19b259ae189f

closed 2024-02-17 by nohal

svMigration, 2024-02-17:
Thank you, Pavel!

Hakansv, 2024-02-17:
@nohal   Tested on Windows.
It's not working as expected. 
The key "R" is working fine but after it has been used rings are always shown at O start. Even if left with no rings.

This option is special. In Options->Ships->"Show range rings" there is no enable/disable control. There is only zero or some rings to select. "0" is the same as disable. 
So if I now use "R" to show rings and the actual selection is "0" then the logic set number of rings to "1" and that "1" is obviously saved to config and used on next O start up.
To get this new function, "R", to work and saved to config it may be we have to add a enable/disable control. Then the "R" would toggle that option instead. (If there are "0" rings at first time "R" set it to "1" as now done).
Or: 
"R" can show the number of rings and distance unit previously used. Next "R" would set "0" rings. But then we have to add a new function to save what's previously used. A new on/off may be easier?

nohal, 2024-02-17:
The reason of course is I did not want to persist the last state in the config file. But see I have to :)

rgleason, 2024-02-17:
Added to https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:user_interface:keyboard_shortcuts