OpenCPN/OpenCPN #3578: Plugin load error, Catalog error, Plugins double in catalog

BerndCirotzki, 2024-01-06:
After compiling this new OpenCPN Code, Plugins can not be loaded from Catalog.
(Example test with Autotrackraymarine_pi)
some Plugins are double in the Caltalog. ( EngineDashboard, Raymarine Autopilot.)
testet with Windows, wx3.2.4

[image]

transmitterdan, 2024-01-06:
Hello @BerndCirotzki 

This is to be expected.  The latest master branch is using a different version of wxWidgets than the currently supported release.  This is so we can test and prepare for the next supported release.  This is not a bug I think but just what happens when advancing the version of wxWidgets.

BerndCirotzki, 2024-01-06:
OK,
but it also happens, when compiling a Plugin with wx3.2.4 and installing the package to
c:\user\...\Appdata\local\opencpn\   ....

Ok, we will see.

nohal, 2024-01-06:
That old plugins laying around the system are found and not being compatible is to be expected. That the same plugin (unless it claimed to have a different name over time) appears twice is a bug though.

nohal, 2024-01-06:
If you find a plugin that should be compatible and is reported as not, we would like to see it  of course.

BerndCirotzki, 2024-01-06:
I deleted all old stuff. In:
c:\progamdata\opencpn\plugins\cache ...
C:\progamme(x86)\opencpn\plugins ...
c:\user...\Appdata\local\opencpn\
So I think, there is no old stuff any more.
then copliled OPENCPN with wx3.2.4 and (my) plugin with wx3.2.4 ... installing it, and it was not working. That's all what I can tell
you for help.

transmitterdan, 2024-01-06:
Can you point me to your github so I can build the plugin and test?  Does your build process create the tarball so the plugin can be imported?

nohal, 2024-01-06:
Are you sure not to mix Release and Debug builds? What plugin is it exactly?

BerndCirotzki, 2024-01-06:
made this :
cmake --build . --config release --target package
.. but I do now everthing again with the buildwin\win_deps.bat and  ci\appveyor.bat ! Ok ... wait this could take some time.
Plugin :
https://github.com/BerndCirotzki/raymarine_autopilot_pi

BerndCirotzki, 2024-01-06:
... sorry, same problem.
everthing is a release.

BerndCirotzki, 2024-01-06:
Logfile :

[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/13850480/opencpn.log)

bdbcat, 2024-01-06:
Pretty clear from the log:

```
17:42:15.171 MESSAGE plugin_loader.cpp:449 Checking plugin candidate: C:\Users\Home PC 64Bit\AppData\Local\opencpn\plugins\raymarine_autopilot_pi.dll
17:42:15.171 MESSAGE plugin_loader.cpp:516 Checking plugin compatibility: C:\Users\Home PC 64Bit\AppData\Local\opencpn\plugins\raymarine_autopilot_pi.dll
17:42:15.180 MESSAGE plugin_loader.cpp:1369 Plugin is compatible: false
17:42:15.180 MESSAGE plugin_loader.cpp:524 Incompatible plugin detected: C:\Users\Home PC 64Bit\AppData\Local\opencpn\plugins\raymarine_autopilot_pi.dll

```
Can you post the tarball somewhere, and we can step through the compatibility test?

transmitterdan, 2024-01-06:
@bdbcat 

That's not the only problem.  The first problem you have noted (causing the "Plugin is compatible: false") is the target tuple in the xml file is "msvc-wx" but it should be "msvc-wx32".  I fixed that in my local build but there is another problem.  I have to leave for an eye appointment but later today I will find the other problem.  It's probably something to do with wx dll version or naming.

BerndCirotzki, 2024-01-06:
> Can you post the tarball somewhere, and we can step through the compatibility test?
posted it to my repo ... to download
https://github.com/BerndCirotzki/raymarine_autopilot_pi

nohal, 2024-01-06:
Yes, there are more problems with this particular plugin.

I just did a quick test here and get to this:

```
13:56:21.013 MESSAGE plugin_loader.cpp:449 Checking plugin candidate: C:\Users\pavel\AppData\Local\opencpn\plugins\raymarine_autopilot_pi.dll
13:56:21.013 MESSAGE plugin_loader.cpp:516 Checking plugin compatibility: C:\Users\pavel\AppData\Local\opencpn\plugins\raymarine_autopilot_pi.dll
13:56:21.013 MESSAGE plugin_loader.cpp:1226 Compatible wxWidgets plugin library found for C:\Users\pavel\AppData\Local\opencpn\plugins\raymarine_autopilot_pi.dll: wxmsw32u_core_vc14x.dll
13:56:21.013 MESSAGE plugin_loader.cpp:1369 Plugin is compatible: true
13:56:21.013 MESSAGE plugin_loader.cpp:1385 PluginLoader: Loading PlugIn: C:\Users\pavel\AppData\Local\opencpn\plugins\raymarine_autopilot_pi.dll
13:56:21.013   ERROR dlmsw.cpp:167 Failed to load shared library 'C:\Users\pavel\AppData\Local\opencpn\plugins\raymarine_autopilot_pi.dll' (error 0: The operation completed successfully.)
13:56:21.013 MESSAGE plugin_loader.cpp:1430 Ignoring blacklisted plugin raymarine_autopilot_pi
13:56:21.013 MESSAGE plugin_loader.cpp:1440    PluginLoader: Cannot load library: C:\Users\pavel\AppData\Local\opencpn\plugins\raymarine_autopilot_pi.dll
```

BerndCirotzki, 2024-01-06:
for example ... AutoTrackRaymarine_pi and EngineDashboard_pi have same problem.

BerndCirotzki, 2024-01-06:
Notice : ... all of them use N2K !
the old version from ramarineautopilot_pi (v 1.0.40.0) works. There is no N2k !

transmitterdan, 2024-01-06:
> Notice : ... all of them use N2K ! the old version from ramarineautopilot_pi (v 1.0.40.0) works. There is no N2k !

Bernd, I don't know what any of this has to do with N2k.  The issue I see is that the .dll will not load at least in Windows.  The wxWidgets dll loader rejects it probably because Windows cannot find a dependency .dll. I have attached a tarball built for relwithdebinfo build.
[raymarine_autopilot_pi-2.0.2.0-msvc-x86_64-wx-10.0.26020-MSVC.tar.gz](https://github.com/OpenCPN/OpenCPN/files/13850941/raymarine_autopilot_pi-2.0.2.0-msvc-x86_64-wx-10.0.26020-MSVC.tar.gz)

transmitterdan, 2024-01-06:
What I had to do to get that tarball to pass the compatibility test:
1) After building I had to edit the raymarine_autopilot_pi-2.0.2.0-msvc-x86_64-wx-10.0.26020-MSVC.xml metadata file and change '<target>msvc-wx</target>' to '<target>msvc-wx32</target>'
2) Run the ./cloudsmith-upload.sh bash script to create the tarball with the edited metadata.xml file.

BerndCirotzki, 2024-01-06:
... Sorry not working changed 'msvc-wx' to 'msvc-wx32' and ./cloudsmith-upload.sh. I can Install the new tarball, but same error.
All what you wrote is the part "around" the plugin code. Comes as PR from Rick's Repo with I merge. So in this case you wrote, there is somewhere a problem with that. I am only the one who found it.
But however, it's also in other Plugins.
Understand. "msvc-wx32" ... but than there must be a correction as a PR for all other plugins.
And why dones it work 2 Weeks ago?

nohal, 2024-01-06:
The reality is there are several problems with this plugin (and probably others). It may well be possible that by some accident they were accepted two weeks ago, but that does not matter, these problems need to be fixed, not the core application broken back again.

nohal, 2024-01-06:
It of course is also possible there are some new problems in the current plugin manager code, we are looking into it now.

nohal, 2024-01-06:
OK, the problem on OpenCPN core side is related to the code reorganization which seem to have made [GetN2000Payload](https://github.com/OpenCPN/OpenCPN/blob/master/include/ocpn_plugin.h#L1700-L1701) from `model/src/comm_plugin_api.cpp` unavailable to plugins, which I suppose wasn't intentional:

```
19:44:58.469 MESSAGE plugin_loader.cpp:449 Checking plugin candidate: /home/nohal/.local/lib/opencpn/libraymarine_autopilot_pi.so
19:44:58.469 MESSAGE plugin_loader.cpp:516 Checking plugin compatibility: /home/nohal/.local/lib/opencpn/libraymarine_autopilot_pi.so
19:44:58.469 MESSAGE plugin_loader.cpp:1369 Plugin is compatible: true
19:44:58.469 MESSAGE plugin_loader.cpp:1385 PluginLoader: Loading PlugIn: /home/nohal/.local/lib/opencpn/libraymarine_autopilot_pi.so
19:44:58.470   ERROR dlunix.cpp:120 Failed to load shared library '/home/nohal/.local/lib/opencpn/libraymarine_autopilot_pi.so': /home/nohal/.local/lib/opencpn/libraymarine_autopilot_pi.so: undefined symbol: _Z15GetN2000Payload10NMEA2000Id11ObservedEvt
19:44:58.470 MESSAGE plugin_loader.cpp:1430 Ignoring blacklisted plugin libraymarine_autopilot_pi
19:44:58.470 MESSAGE plugin_loader.cpp:1440    PluginLoader: Cannot load library: /home/nohal/.local/lib/opencpn/libraymarine_autopilot_pi.so
```

Seems like something for @leamas to have a look at.

Then there are 2 problems with the packaging of the plugin, especially on Windows:

1. The `target` metadata value like `msvc-wx` is wrong for OpenCPN 5.8 and newer. The problem is caused by performing the build without setting `WX_VER` environment variable as can be seen in https://github.com/BerndCirotzki/raymarine_autopilot_pi/blob/master/cmake/PluginSetup.cmake#L81-L97.
The solution is to either change that logic, not perform local builds and use CI products (your CI build scripts do set that variable correctly) or not to forget to set that variable when you do the builds. (Also please do not commit build products to the repository, it is a bad habit and inflates the repository unnecessarily)

2. The windows package uses architecture `x86_64` taken from the builder machine and everything derived from it like filename of the tarball and `target-arch` metadata is wrong - the build product is 32bit `x86` at the moment. These can perhaps be considered "cosmetic" for now as we do not provide 64bit builds on Windows, but will become a problems once we do.

leamas, 2024-01-07:
> which seem to have made [GetN2000Payload](https://github.com/OpenCPN/OpenCPN/blob/master/include/ocpn_plugin.h#L1700-L1701) from model/src/comm_plugin_api.cpp unavailable to plugins

Indeed. The file is compiled, but the symbols is missing in the opencpn binary. Most likely a linker issue. I'm on it, but somewhat occupied by  Real Life over the weekend.

rgleason, 2024-01-07:
I really do not understand what the problem is. 
I have installed v2.0.2 using PIM and with "Import Plugins" and it loads fine. This plugin is basically derived from Bdbcat testplugin with some few updates from Jon's testplugin. 

[image]

I really think the context of the problem should be stated clearly upfront, otherwise it will lead everyone down the wrong path to reach the wrong conclusions.
Is Bernd trying to build it himself and install it?  Is that where this issue is coming from?

Furthermore Bdbcat's version hard codes the msvc-wx32  and this is perhaps an example why it is better than using an environment variable (Jon's version) which someone might not know about!  But this does not answer why Bernd's version is having issues.

rgleason, 2024-01-07:
Bernd, for local builds, have you taken a look at bld.bat?
I simply use the MSVS command prompt to run bld.bat
Then to get the metadata into the tarball, cd back to the main plugin directory and
use a bash prompt with  "./bld.bat" so that the bash ./cloudsmith-upload.sh script executes.

Operational part of the batch file.
```
REM Set local environment using wxWidgets-3.2.2
set "wxDIR=%WXWIN%"
set "wxWidgets_ROOT_DIR=%WXWIN%"
set "wxWidgets_LIB_DIR=%WXWIN%\lib\vc14x_dll" 
set "WXWIN=C:\Users\fcgle\source\wxWidgets-3.2.2"
REM  For Opencpn 5.8 and wxWidgets-3.2.2
cd build
cmake -T v143 -A Win32 -DOCPN_TARGET=MSVC ..
cmake --build . --target package --config relwithdebinfo >output.txt
bash ./cloudsmith-upload.sh

```

leamas, 2024-01-07:
Pushed d77044422 which from my testing ensures that the API functions indeed are available. In particular:
```
$ objdump  -TC ./opencpn | grep N2000Payload
0000000000c8bbb5 g    DF .text	00000000000000c3  Base        GetN2000Payload(NMEA2000Id, ObservedEvt)
```

nohal, 2024-01-08:
I suggest this issue to be closed as the actual problem causing plugins not to load is fixed.

We need a new issue for the Engine Dashboard (or other) plugin being in the list of plugins twice after loading the master catalog. There are multiple versions of the metadata (1.4.1.0, 1.4.2.0, 1.5.0.0 and 1.6.0.0) for it in https://github.com/OpenCPN/plugins/tree/master/metadata which of course is wrong, but still the result should not be what it is now.

The plan for actually fixing the architecture identifiers handling in the core app is outlined in #2027 and no action on plugin side is needed in foreseeable future until well past the 5.10 release.

The rest is related to the plugin itself and the recent update to its build system as far as I can tell.

BerndCirotzki, 2024-01-08:
👍 Works. Thank's to all.
I think you can close this issure if you also think so.

closed 2024-01-08 by nohal