OpenCPN/OpenCPN #3581: MbTiles that cross the 180 don't display correctly.

rgleason, 2024-01-08:
**Describe the bug**
MbTiles that cross the 180 don't display correctly.

**To Reproduce**
Steps to reproduce the behavior:

1. I'm making mbTiles of Fiji, & the 180 cuts through the right end of Fiji.  I can create mbTiles that go up to the 180 on each side, but when I load them into OpenCPN, I get very strange behavior:
2. If the 180 is on the left side of my screen, then ONLY the mbTile on the right is displayed, & if the 180 is on the right side, then only the mbTile on the left is displayed.
3. If I grab the chart & start panning OpenCPN, when the 180 crosses the center of my [screen](https://www.cruisersforum.com/forums/tags/screen.html), the mbTile that has been visible on one side disappears & the mbTile that's been invisible (on the other side) suddenly appears!
4. If I try to construct an mbTile that crosses the 180, OpenCPN doesn't display it at all (although I don't know the quality of the tool I use to build that mbTile - it's the Merge_MbTiles function of Paul Higgins' Sat2Chart program).

Jon Hacking writes:
  
> Dave (bdbcat) has asked me to file another bug, based on [this post](https://www.cruisersforum.com/forums/f134/help-faq-30943-131.html#post3856674) (#1961 of the Help/FAQ forum).  The latest versions of Chrome, Edge, & Firefox all complain of numerous deprecated errors on the Flyspray site, so I can’t find anywhere to actually file (open) a bug.  I sent you a CF PM about this, but apparently you didn’t see it.
> 
> MbTiles that cross the 180 don't display correctly.
> 
> I'm making mbTiles of Fiji, & the 180 cuts through the right end of Fiji.  I can create mbTiles that go up to the 180 on each side, but when I load them into OpenCPN, I get very strange behavior:
> 
> If the 180 is on the left side of my screen, then ONLY the mbTile on the right is displayed, & if the 180 is on the right side, then only the mbTile on the left is displayed.
> 
> If I grab the chart & start panning OpenCPN, when the 180 crosses the center of my [screen](https://www.cruisersforum.com/forums/tags/screen.html), the mbTile that has been visible on one side disappears & the mbTile that's been invisible (on the other side) suddenly appears!
> 
> If I try to construct an mbTile that crosses the 180, OpenCPN doesn't display it at all (although I don't know the quality of the tool I use to build that mbTile - it's the Merge_MbTiles function of Paul Higgins' Sat2Chart program).
> 
> Dave (bdbcat) says that [this is probably a bug](https://www.cruisersforum.com/forums/f134/help-faq-30943-131.html#post3856691) associated with displaying [charts](https://www.cruisersforum.com/forums/tags/charts.html) across the 180 (always difficult).
> 
> I can provide testing samples if needed, but making them with SAS Planet, or any tool that can create mbTiles, should be easy.
> 
> Cheers  --  Jon    s/v Ocelot

rgleason, 2024-01-08:
Similar bug in Weather_routing " [Bug 04 Routing at Date Line West from 180 is frozen #79](https://github.com/rgleason/weather_routing_pi/issues/79) "

bdbcat, 2024-01-08:
Rick...
Please arrange to get a test MBtiles file that exhibits the problem.  I expect the georef data in the tiles may differ according to the tile source.
Thanks

rgleason, 2024-01-21:
Jon says to please close this now...  https://www.cruisersforum.com/forums/f134/help-faq-30943.html#post3859340

rgleason, 2024-01-23:
Jon I can't do this cross platform backup bug reporting it is a waste of time. Please get signed up with github.

Jon-Hacking, 2024-01-24:
OK, I can file my own bugs now.  But I can no longer reproduce this bug.

closed 2024-01-24 by rgleason