OpenCPN/OpenCPN #3582: D-board GNSS Status. Windows UI is disturbed.

Hakansv, 2024-01-09:
After the recent color and font changes to gps.cpp the Windows outlook is "not so nice". 
The bitmap background is oversized and SAT-Number font is hard to read. See pict 1. (Rpi Debian don't show the same disturbed background and I suppose the same on Mac)

I don't know the historical background to use wxBitmap here but if I change to "pure" dc->drawText and transparent background the picture is , for me, more clear, see pict 2.
@nohal  @All Thoughts?
If the UI is OK with you I can PR a patch to make a MacOs test. (I have only Win and Rpi Debian.)

Pict. 1
[image]

Pict 2
[image]

nohal, 2024-01-09:
It is OK with me. Note that all I have to do with the mentioned change is I commited it, not my code.

Hakansv, 2024-01-09:
I know it's not your code. Thanks for your view. Appreciated!

commit 77124ec5 (2024-01-09): Merge pull request #3583 from Hakansv/db_satview

D-board GNSS Status Simplify globe sat-view draw  Closes #3582
closed 2024-01-09 by nohal