OpenCPN/OpenCPN #3586: Extend plugin API to allow contextual menus on waypoints, routes, tracks and AIS targets,

antipole2, 2024-01-12:
Presently, the plugin APIs allow a plugin to create, act on and remove additional menu items to the canvas contextual menu.
This works well and is really useful. For example, in JavaScript you can use this [to create a boundaried area](https://github.com/antipole2/JavaScripts-shared/blob/main/Boundaries/Boundaries.adoc).

Please can we have the ability to do these for other contexts such as waypoint, route, track and AIS target?  This would open up great possibilities for custom actions on these objected.  As an example, it would be possible to have a custom action on an AIS target such as making a DSC call, collision avoidance calculations, logging details etc.

Graham48, 2024-01-17:
I would second that as a very useful addition.
It may also be good to be able to add to the menu at the top if it's a temporary context...

antipole2, 2024-04-06:
In my recently published [PointUtility script](https://github.com/antipole2/PointUtility) the user needs to select a waypoint.  This is impossible at the moment.  The script has to search through all waypoints working out which is the nearest.  This is just one example of why this enhancement is needed.

bdbcat, 2024-04-06:
Tony...
Suggest some API function prototypes, and we can talk about it.

antipole2, 2024-04-07:
Dave...  the APIs I use are

- `AddCanvasContextMenuItem(wxMenuItem *pitem, opencpn_plugin *pplugin);`
- `RemoveCanvasContextMenuItem( int item);`
-  `OnContextMenuItemCallback(int menuID)`

The cleanest way forward could be to provide their equivalents for each of waypoints, routes and tracks, i.e.

- `AddWaypointContextMenuItem(wxMenuItem *pitem, opencpn_plugin *pplugin);`
- `RemoveWaypointContextMenuItem( int item);`
-  `OnWaypointMenuItemCallback(int menuID)`

I do not use the set viz and set grey options. I just remove the menu item when not applicable. 

The above would be great.

This is how the canvas context menu looks when the PointUtility script is running 
[image]
`Copy mark` would be moved to the waypoint context menu where it belongs.

Some of the ideas I am toying with might need a lot of menu items.
I have been reflecting on how it would be great if we could create sub-menus.  This may be too much at this time, but while you are in there, perhaps you could think about this.

Thank you for looking at this RFE.

antipole2, 2024-07-02:
Still missing from Beta 2 - really needed!

bdbcat, 2024-07-03:
Sorry, not happening for O5.10.
Deferred to 5.12

antipole2, 2024-07-22:
Still mile-stoned for v5.10

rgleason, 2025-01-28:
Having this in the API would enable Javascript_pi to handle many common needs for functionality and operations on navobj.xml,  thus offloading those tasks from OpenCPN's main focus.

antipole2, 2025-06-18:
Implemented inOCPN v11

closed 2025-06-18 by antipole2