OpenCPN/OpenCPN #3587: WebUI support for those without monitors

deanfourie1, 2024-01-12:
**Is your feature request related to a problem? Please describe.**
Yes, I have no monitor and would like to run in docker.

**Describe the solution you'd like**
A fully compatible and interactive WebUI rather then just monitor based GUI. Browse to the IP address and the complete UI is display in the browser.

**Describe alternatives you've considered**
There are none, other then attaching a monitor.

**Additional context**
Not really, I think that sums it up.

nohal, 2024-01-12:
As for non-existent alternatives, people have been using OpenCPN headless for years with VNC. If you insist on using a web browser for it instead of a normal VNC client, there is https://github.com/novnc/noVNC
Otherwise you need to look for other software, like https://github.com/wellenvogel/avnav, OpenCPN is a desktop application by design.

deanfourie1, 2024-01-12:
> As for non-existent alternatives, people have been using OpenCPN headless for years with VNC. If you insist on using a web browser for it instead of a normal VNC client, there is [...]

I understand that. And thanks for the replies and suggestions.

Just thought it would be a great addition to an already great piece of software to take it that extra step.

Adds for a lot of versatility and flexibility.

Just my 50 cents ;)

nohal, 2024-01-12:
This is a completely different use case, what you ask for is reimplementing pretty much the whole application to do something different than what it was designed for, not an addition. The web use case and the desktop use case as approached by OpenCPN have very little overlap from the technical point of view, changing that is simply not feasible.

transmitterdan, 2024-01-13:
In case someone reads this in the future, I have found remote desktop (RDP) to be a good alternative to VNC.  There is good support for xrdp at least on the RPi.  I use it all the time as none of the 4 PIs I have are connected to a monitor.  I split them between VNC and xrdp. I like xrdp the best but not by a lot.