OpenCPN/OpenCPN #3588: peer transfer: windows: Recieving side crash with routes window open

leamas, 2024-01-12:
**Describe the bug**
When transferring peer data like a route to a windows machine, this crashes if the *Route & Tracks* window is open
 
**To Reproduce**
Steps to reproduce the behavior:
1. Start windows machine
2. Open *Options* |  *Route & Mark Manager*,  leave window open
3. Start some other machine
4. Make sure there is a route available
5. Right-click on route and select *Send to Peer*
6. After scanning, select the windows machine from 1.
7. Push Send button. 
8. Watch sending side time out and receiving side crash.

**Expected behavior**
No crash. That is, the same behaviour as when the Route & Mark Manager isn't open

**Desktop (please complete the following information if applicable):**
 - Windows 10.

**Additional context**
This is not present in 5.8.4.

bdbcat, 2024-01-15:
Did a little debugging.
The crash on the Windows side is a result of resource exhaustion, of some sort.
Brought on by a deadlock condition at 

```
void RestServer::UpdateReturnStatus(RestServerResult result) {
  {
    std::lock_guard<std::mutex> lock{ret_mutex};
    return_status = result;
  }
  return_status_cv.notify_one();
}

```
Here, the mutex "ret_mutex"  is somehow already locked, so the lock_guard spins at 100% cpu.  Eventually crashes.
Tried to track who else might be holding ret_mutex, but no luck.  I am suspicious of this code, found in 2(?) places:

    std::unique_lock<std::mutex> lock{parent->ret_mutex};
    bool r = parent->return_status_cv.wait_for(lock, 10s, [&] {
      return parent->GetReturnStatus() != RestServerResult::Void;
    });
 
Does not that unique_lock "steal" the mutex, somehow?  My understanding of std::mutex stuff is fuzzy.

bdbcat, 2024-01-15:
Or, does somehow

return_status_cv.notify_one();

cause a recursive call to UpdateReturnStatus() ?

leamas, 2024-01-15:
No idea what it could be about. The fact that it is windows only and related to an open window makes it more "interesting".. However, I have some ideas on what it isn't. 

> The crash on the Windows side is a result of resource exhaustion, of some sort.
Brought on by a deadlock condition at [...] Here, the mutex "ret_mutex" is somehow already locked, so the lock_guard spins at 100% cpu. 

If there is a deadlock there cannot be any 100% CPU. Livelock?

> Does not that unique_lock "steal" the mutex, somehow? My understanding of std::mutex stuff is fuzzy.

No. One could think of  `lock` as:
  - Constructor: `parent->ret_mutex.lock()`
  - Destructor: : `parent->ret_mutex.unlock()`

So when the control flow passes the innermost closing brace it will do  `parent->ret_mutex.unlock()`. There is no way it can keep the lock after that.

Some kind of recursive loop  could might perhaps cause something like this, agreed.  Back later.

leamas, 2024-01-15:
Too many balls in the air  for me right now. Just a brain dump:

Anything actually releasing resources needs to be part of a livelock loop. However, the only thing which actually changes `return_status`is indeed `UpdateReturnStatus()`.  It would be interesting with a breakpoint here and inspecting the stack when there.

bdbcat, 2024-01-15:
I have add breakpoint where suggested.  Gets hit a lot, stack seems reasonable.  Until it fails to get lock...
I'm thinking I need to add some printf here, to track things.
I'll also try some hack to detect recursion here as well.

leamas, 2024-01-15:
>  Gets hit a lot, stack seems reasonable.

The obvious question is where it's called from. But you have already checked that...

PR #3592 (2024-01-16): rest_server: Avoid wxWin data in worker thread (#3588)
rest_server.cpp invokes methods affecting wxWindows data structures from the server thread. Fix by moving to main thread.

This is a long standing issue which seems to be present also in 5.8.4 but for some reason does not trigger anything there.

Kudos to @hakansv which found the nasty #3588 bug while working with #3584. And of course also to @bdbcat who found the basic problem.

Closes: #3588
commit cbae8109 (2024-01-16): rest_server: Avoid wxWin data in worker thread (#3588)

rest_server.cpp invokes methods affecting wxWindows data structures from
the server thread. Fix by moving to main thread.

This is a long standing issue which seems to be present also in 5.8.4
but for some reason does not trigger anything there.

Kudos to @hakansv which found the nasty #3588 bug while working
with #3584. And of course also to @bdbcat who found the basic problem.

Closes: #3588
closed 2024-01-16 by bdbcat