OpenCPN/OpenCPN #3589: WXSVG memory leaks

Kitiara, 2024-01-14:
For your information the WXSVG project is crawling with memory leaks such as the one shown below.

https://github.com/OpenCPN/OpenCPN/blob/d7992745cb928a33690f9866ed834f0eb96615eb/libs/wxsvg/include/wxSVG/SVGAnimatedLength.h#L24

Also The Rule of Three (or Five) must be performed in order to prevent memory management issues.

nohal, 2024-01-14:
Does this ever materialize in your experience? We do not use animated SVG and we do not use actually use wxSVG at all anymore on any platform where we are not stuck with wxWidgets 3.0

Kitiara, 2024-01-14:
Yes while loading **uidata/MUI_Flat/CompassRose.svg**

bdbcat, 2024-01-14:
And how is this leak manifested?  What is the user impact, if any.  How does one know there is a leak?

nohal, 2024-01-14:
Interesting, don't remember seeing it in my address sanitizer enabled builds

Kitiara, 2024-01-14:
> And how is this leak manifested? What is the user impact, if any. How does one know there is a leak?

At the very start of the app and probably everytime you load a plugin.

```
inline wxSVGAnimatedLength& operator=(const wxSVGAnimatedLength& value)
{
    m_baseVal = value.m_baseVal;
    m_animVal = value.m_animVal != NULL ? new wxSVGLength(*value.m_animVal) : NULL;
    return *this;
}
```
In this implementation, there was no call to ResetAnimVal, which means the existing content of m_animVal was not cleared, and the previous memory block was not deallocated.

Kitiara, 2024-01-14:
> Interesting, don't remember seeing it in my address sanitizer enabled builds

Oh I see, I have got the ocpnUSE_SVG preprocessor enabled.

Kitiara, 2024-01-14:
> And how is this leak manifested? What is the user impact, if any. How does one know there is a leak?

At the very start of the app and probably everytime you load a plugin.

The `wxSVGAnimatedLength` class is missing an assignment operator for handling assignments with a `wxSVGLength` value.

nohal, 2024-01-14:
Not really, `ocpnUse_SVG` enables SVG support in general and you do want it.  What handles SVG files depends on the version of wxWidgets, if it is 3.1.6 or higher, SVG support is provided by `wxBitmapBundle` (If new enough WX is detected, `ocpnUSE_wxBitmapBundle` is enabled and a wxSVG is not used as you can see in https://github.com/OpenCPN/OpenCPN/blob/master/gui/src/svg_utils.cpp#L32-L36). This is the case everywhere but on the legacy Linux platforms (Up to +/- Debian 11 equivalents), which are going to be removed completely at their EOL.

nohal, 2024-01-14:
But still I do not remember actually seeing this leak in the sanitized builds against WX 3.0, where wxSVG actually was used. Or maybe just missed them in the noise of unimportant and on our side unfixable leaks in `fontconfig` and other system libs.

Kitiara, 2024-01-14:
ocpnUSE_wxBitmapBundle did the trick, thanks.

nohal, 2024-01-14:
Just out of curiosity, do you have any idea why it was not actually applied automatically by `cmake` in your environment?

Kitiara, 2024-01-14:
No idea, it was undefined somehow.

nohal, 2024-01-14:
OK, so let's try it the other way around. What is your platform exactly so that I can add it to my farm?

Kitiara, 2024-01-14:
Windows 11 Pro - 23H2

nohal, 2024-01-14:
And there of course always is the general suggestion: Before submitting an issue, `rm -rf`` your build directory and reproduce the problem you are going to report.

nohal, 2024-01-25:
Closing. ASAN does not detect any leak actually happening. And given wxSVG is being phased out investing in it makes little sense.

closed 2024-01-25 by nohal