OpenCPN/OpenCPN #3593: Toolbars forced out of main window

leamas, 2024-01-16:
**Describe the bug**
When moving the OpenCPN out of screen the toolbars floats and stays on screen

**To Reproduce**
1. Start OpenCPN
2. Move main window so it's mostly outside the screen
3. See  toolbars  in a very strange position

**Expected behavior**
Toolbars should not follow main window out of screen 

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - Linux (Fedora, Ubuntu Jammy)

leamas, 2024-01-16:
I see this both on my Fedora 38 desktop running X11 and on Jammy running Wayland.

Hakansv, 2024-01-16:
FWIW:
On Debian X11, Bullseye o Bookworm, I see the same "halfway": The toolbar is misplaced until release of the mouse button. Then it's relocated again. So it's like a delay.
Not seen on Win.

nohal, 2024-02-10:
I'm afraid that as long as the toolbars are windows we will keep hitting stuff like this (and other like the positioning on Wayland in RPi Bookworm, toolbar appearing as another application in the window list on some WMs, weird stuff in some tiling WMs etc.) and the only real solution is to actually render them as the topmost layer of the canvas and implement the window-like functionalities ourselves...

bdbcat, 2024-02-10:
The per-canvas toolbar (lower right) is quite simple in structure.  Should be easy to modify into a child of chart canvas.
The main toolbar is more fiddly.  I have not noticed any DM problems with this toolbar, but I may be mistaken.

leamas, 2024-02-10:
Sidenote:

> like positioning on Wayland in RPi Bookworm

This not really about Wayland or Bookworm, both works fine when using gnome + raspbian or vanilla Debian. The problem is the wayfire compositor which is the default on Raspbian 12.

That said,  my own overall abysmal understanding is the same:  the root cause for this is the  TLW    implementation. Probably also applies to #3579 and #3146?

Hakansv, 2024-02-10:
I've Rpi 5 using Openplotter v4 image now on X11.
If needed for a future patch test I can switch to Wayland.

leamas, 2024-02-11:
Seems that many of us i looking on this bug. If anyone starts working on it, please assign it fisr to avoid duplicated work.

transmitterdan, 2024-02-13:
I am not working on it but the main toolbar is easily confused by various DMs. For example, you can run the Linux version of O on Windows.  Windows will hand it off to WSL which in turn has a crude WM such that a Linux program can be run on a Windows desktop.  Granted, this is an unusual case, but it will show that the toolbar is totally non-functional.  Can't click it, move it or anything and it quite often disappears and reappears as one fiddles with the canvas window.  Interestingly, the per-canvas toolbars work fine albeit slightly in the wrong place.  The slide-in-out per-canvas toolbar dialog even works.
[image]

bdbcat, 2024-02-13:
@transmitterdan 
OK, point taken.  Both toolbars need re-homing to chart canvas parent.
I'm not on it yet, but expect to get there pretty soon.  Unless anyone else wants to tackle this...

bdbcat, 2024-03-13:
Per-canvas toolbar behavior possibly corrected by https://github.com/OpenCPN/OpenCPN/commit/75f813a17000eac5d2f1f65f8c349c59a520c941
Tested here on MacOS 12.1, W10, and linux (KDE)/
More DM testing needed.

leamas, 2024-03-13:
Tested on Fedora 39/X11 and Jammy/Wayland. Both looks fine w r t the per-canvas toolbar.

Hakansv, 2024-03-13:
I've run it on RPi's using bookworm, X11, and bullseye.
If related I still see the same as mentioned above. The main toolbar doesn't follow a O window mouse drag until mouse-up on left button. Then it's placed on its new position. Canvas tools are though following smoothly also while mouse-down.

bdbcat, 2024-03-13:
Thanks for the tests,  Good news.
Main toolbar is next, whenever.

bdbcat, 2024-03-30:
Main toolbar refactor now implemented by 18b5c1220f5792a2837e0debbd7d57e178aad0dd
Good Luck

Hakansv, 2024-03-30:
Dave..
On Win10 175% scaled
Nice! Now I can reset "User interface scale factor" to "0" from former "-2" to get a adequate tools size.
The main tool is now instantly opened after hid. No "flickering"

Notes:
1. The main toolbar is drawn also on the second canvas. If intentional we may discuss? I may think it's copiously.  Se pict 1
2. WMM tool bar icon does not fit. Pict 2

Two canvas mode:
[image]

WMM tool:
[image]

Hakansv, 2024-03-30:
Toolbar size on Win10
On a Win scaled to 175% the size is fine as mentioned.
On a another Win10 100% scaled I've to set the scale factor to "-2" to get a adequate size. So it may be the tools are slightly oversized? Let's await others possible comments on this.

transmitterdan, 2024-03-30:
@bdbcat Nice job.  Much better now.  I test using WSL default window manager and O is now usable in that environment where before it was unusable.  Now I see an interesting thing with the toolbar tooltips:
[image]

The roll-over tooltips for the chart window such as AIS info work as expected. The tooltip pops up next to the mouse cursor. But for the toolbar the tooltips pop up at seemingly random locations. Not sure what to make of that. It's as if the tooltip is appearing somewhere on the chart canvas possibly at the currently drawn location instead of next to the mouse curser over the tool-bar icon.

I realize WSL is not a sanctioned user environment, but I find it helps expose weird interactions between GUI elements.

nohal, 2024-03-30:
Interesting, under Gnome the toolbars are shown on both the canvases (which they should not), but tooltips always on the left one, even when interacting with the right canvas toolbar...

transmitterdan, 2024-03-30:
On Windows the "grab" or "title" bar blinks when the tooltip appears.  I think that's a clue. Possibly something is manually fiddling with window focus and not getting it right on all platforms.  Normally tooltips handle themselves ok.

nohal, 2024-03-30:
The tooltips on the toolbar are special as they are not actually the normal wxWidgets tooltips.

bdbcat, 2024-03-31:
Tooltip action also corrected by a3d2fdc59e1f88220438efed1e552245c30e0102
Not tested on WSL.

Hakansv, 2024-03-31:
Win10:
The new toolbar scale is better but on my (scaled)Win it needs adaption. It works with the patch below. If Retina also need It can't test. 
Tooltips are still fine on my system.

WMM live icon size is now fine.

```
@@ -1324,12 +1324,14 @@ void MyFrame::UpdateAllToolbars(ColorScheme cs) {
 }
 
 void MyFrame::SetAllToolbarScale() {
   g_toolbar_scalefactor = g_Platform->GetToolbarScaleFactor(g_GUIScaleFactor);
+  g_toolbar_scalefactor *= OCPN_GetWinDIPScaleFactor();
 }
 
 void MyFrame::SetGPSCompassScale() {
   g_compass_scalefactor = g_Platform->GetCompassScaleFactor(g_GUIScaleFactor);
+  g_compass_scalefactor *= OCPN_GetWinDIPScaleFactor();
 }
```

transmitterdan, 2024-03-31:
> Not tested on WSL.
Situation largely unchanged on WSL.  I don't consider WSL's default display manager as a valid platform.  But it seems to say something about how the toolbar identifies the screen coordinates to locate the tooltip. Apparently, this confusion about the location of the cursor relative to the toolbar icons causes a subtler problem as well on WSL. If the tooltip appears upon the toolbar itself, then clicking the toolbar icon does nothing.  However, when the tooltip appears well away from the toolbar then clicking the icon works.  Previously, the toolbar icons never worked on WSL so this is better.  But still I think there is some subtle issue that is exposed by testing on WSL.

Dual toolbar issue is resolved on Windows and WSL.

bdbcat, 2024-03-31:
When time allows, I'll spin up a WSL environment and test.  Low priority.

bdbcat, 2024-04-01:
Ooops .  Wrong thread...

leamas, 2024-04-01:
All looks good to me. Closing, please add a note if issue should be reopened.

bdbcat, 2024-06-16:
I have reverted partially the "fix" to this issue, so now needs retesting.
Thanks

leamas, 2024-07-24:
Tested on both native Fedora/gnome/wayland/x86_64 and Debian/XFXE/X11/aarch64/ without problems. 

Furthermore, this has been part of beta1/beta2 without any complaints.

Closing

closed 2024-07-24 by leamas