OpenCPN/OpenCPN #3596: Send to peer. Win to Android. Pin code is not remembered

Hakansv, 2024-01-17:
**Describe the bug**
Send to peer from Win to Android does not remember the pin code. It has to be entered all times.

Steps to reproduce the behavior:
1. Select a route and "send to peer"
2. Select a Android server. - OK
3. Enter the pin code presented at the Android unit. - OK. The route is transferred.
4. On Win change the route and send again to the same Android unit.
5. Now, again I'm asked to enter a pin code.  - Reproducible many times. Also after stop and start both client and server.
Checking the config files, [Settings/REST....] and the Android system is filed on WIN and Win on Android.

**Expected behavior**
The pin code shall only be needed one time as between other system. 

**Desktop (please complete the following information if applicable):**
 - OS:Win and Android 
 - Version Win10 O v 5.9.0. today's master.  
 - Android 8.0 and 13.0  OCPN v 8.5.4 [106]

**Additional context**
Sorry I didn't tested to Android in before cycle.

leamas, 2024-01-18:
Long test round with @Hakansv  which should be granted tons of kudos.

Hakansv, 2024-01-18:
Well, well the publisher solved it and should have the credit.
Thansk

PR #3598 (2024-01-18): Fix repeated Android peer transfers  (#3596)
The problem reported in #3596 turns out to apply to all 5.8.4 servers. Make sure the client does not override old keys unless the server is known to be 5.9+.

Closes: #3596
commit 6fc4c536 (2024-01-18): Fix repeated Android peer transfers  (#3596)

The problem reported in #3596 turns out to apply to all 5.8.4
servers. Make sure the client does not override old keys unless
the server is known to be 5.9+.

Closes: #3596
closed 2024-01-18 by bdbcat