OpenCPN/OpenCPN #3600: S52 presentation library memory leak

nohal, 2024-01-20:
This is the last remaining memory leak detected by ASAN in the core application at 0d620e67d69f8904042058a169df437f20ef7625

```
Direct leak of 30368 byte(s) in 2359 object(s) allocated from:
    #0 0x7faff12d9e28 in operator new(unsigned long) (/lib64/libasan.so.8+0xd9e28) (BuildId: 7fcb7759bc17ef47f9682414b6d99732d6a6ab0c)
    #1 0x92f359 in std::__new_allocator<char*>::allocate(unsigned long, void const*) /usr/include/c++/13/bits/new_allocator.h:151
    #2 0x1892254 in std::allocator_traits<std::allocator<char*> >::allocate(std::allocator<char*>&, unsigned long) /usr/include/c++/13/bits/alloc_traits.h:482
    #3 0x1892254 in std::_Vector_base<char*, std::allocator<char*> >::_M_allocate(unsigned long) /usr/include/c++/13/bits/stl_vector.h:378
    #4 0x189191f in char** std::vector<char*, std::allocator<char*> >::_M_allocate_and_copy<__gnu_cxx::__normal_iterator<char* const*, std::vector<char*, std::allocator<char*> > > >(unsigned long, __gnu_cxx::__normal_iterator<char* const*, std::vector<char*, std::allocator<char*> > >, __gnu_cxx::__normal_iterator<char* const*, std::vector<char*, std::allocator<char*> > >) /usr/include/c++/13/bits/stl_vector.h:1616
    #5 0x18909c2 in std::vector<char*, std::allocator<char*> >::operator=(std::vector<char*, std::allocator<char*> > const&) /usr/include/c++/13/bits/vector.tcc:238
    #6 0x188811b in ChartSymbols::BuildLookup(Lookup&) /home/nohal/source/OpenCPN/libs/s52plib/src/chartsymbols.cpp:491
    #7 0x1883939 in ChartSymbols::ProcessLookups(pugi::xml_node&) /home/nohal/source/OpenCPN/libs/s52plib/src/chartsymbols.cpp:254
    #8 0x188c442 in ChartSymbols::LoadConfigFile(s52plib*, wxString const&) /home/nohal/source/OpenCPN/libs/s52plib/src/chartsymbols.cpp:711
    #9 0x17e657c in s52plib::S52_load_Plib(wxString const&, bool) /home/nohal/source/OpenCPN/libs/s52plib/src/s52plib.cpp:1158
    #10 0x17de4ce in s52plib::s52plib(wxString const&, bool) /home/nohal/source/OpenCPN/libs/s52plib/src/s52plib.cpp:295
    #11 0x10c00ab in LoadS57() /home/nohal/source/OpenCPN/gui/src/ocpn_frame.cpp:8343
    #12 0x107fa01 in MyFrame::BuildMenuBar() /home/nohal/source/OpenCPN/gui/src/ocpn_frame.cpp:3392
    #13 0x107f6fb in MyFrame::ApplyGlobalSettings(bool) /home/nohal/source/OpenCPN/gui/src/ocpn_frame.cpp:3358
    #14 0x102ebe1 in MyApp::OnInit() /home/nohal/source/OpenCPN/gui/src/ocpn_app.cpp:1727
    #15 0x103c4e7 in wxAppConsoleBase::CallOnInit() /usr/include/wx-3.2/wx/app.h:93
    #16 0x7faff0313341 in wxEntry(int&, wchar_t**) (/lib64/libwx_baseu-3.2.so.0+0x113341) (BuildId: 989c209d44037d18092f742b47617361331adcb2)
    #17 0x101da8c in main /home/nohal/source/OpenCPN/gui/src/ocpn_app.cpp:855
    #18 0x7fafee446149 in __libc_start_call_main (/lib64/libc.so.6+0x28149) (BuildId: 788cdd41a15985bf8e0a48d213a46e07d58822df)
    #19 0x7fafee44620a in __libc_start_main_impl (/lib64/libc.so.6+0x2820a) (BuildId: 788cdd41a15985bf8e0a48d213a46e07d58822df)
    #20 0x792cd4 in _start (/home/nohal/source/OpenCPN/build/opencpn+0x792cd4) (BuildId: f98005a49fb80d7c7223b6a4899c795f84c6da6c)
```

bdbcat, 2024-01-26:
Quick fix solution, just to prove my understanding of the root cause.
This LUPRec class could stand some modernization, for sure.  Too much old-fashioned storage allocation by ptr.
On my list.

nohal, 2024-01-26:
Changed appearance, but ASAN is still convinced it is there, all it reports now is

```
Direct leak of 30368 byte(s) in 2359 object(s) allocated from:
    #0 0x7f5b818d9e28 in operator new(unsigned long) (/lib64/libasan.so.8+0xd9e28) (BuildId: 7fcb7759bc17ef47f9682414b6d99732d6a6ab0c)
    #1 0x1e19b26 in ChartSymbols::BuildLookup(Lookup&) /usr/include/c++/13/bits/new_allocator.h:151
```

bdbcat, 2024-01-27:
Another, more extensive re-factor, dropping some evil pointer vars from LUPRec struct.

nohal, 2024-01-27:
ASAN reports big leaks and a malloc/delete mismatch in https://github.com/OpenCPN/OpenCPN/blob/master/libs/s52plib/src/s52plib.cpp#L518

I can get rid of all of it and make ASAN happy by 
```diff
diff --git a/libs/s52plib/src/chartsymbols.cpp b/libs/s52plib/src/chartsymbols.cpp
index 755163b31..91b9cd4d4 100644
--- a/libs/s52plib/src/chartsymbols.cpp
+++ b/libs/s52plib/src/chartsymbols.cpp
@@ -474,7 +474,7 @@ void ChartSymbols::ProcessSymbols(pugi::xml_node &node) {
 
 
 void ChartSymbols::BuildLookup(Lookup &lookup) {
-  LUPrec *LUP = (LUPrec *)calloc(1, sizeof(LUPrec));
+  LUPrec *LUP = new LUPrec();
   //plib->pAlloc->Add(LUP);
 
   LUP->RCID = lookup.RCID;
diff --git a/libs/s52plib/src/s52plib.cpp b/libs/s52plib/src/s52plib.cpp
index 49463cd61..e7f64ae2f 100755
--- a/libs/s52plib/src/s52plib.cpp
+++ b/libs/s52plib/src/s52plib.cpp
@@ -9245,7 +9245,7 @@ void s52plib::GetAndAddCSRules(ObjRazRules *rzRules, Rules *rules) {
 
   if (NULL == LUP)  // Not found
   {
-    NewLUP = (LUPrec *)calloc(1, sizeof(LUPrec));
+    NewLUP = new LUPrec();
     //pAlloc->Add(NewLUP);
 
     NewLUP->DISC = rzRules->LUP->DISC;  // as a default
```

FWIW not even without this patch I can reproduce the crash @Hakansv  reports in Zulip

bdbcat, 2024-01-27:
Well, yes.
This is some truly awful code from the bad old early days of OCPN.
Fixed.  Thanks.

closed 2024-01-27 by nohal