OpenCPN/OpenCPN #3601: ver 5.8.4-2 will not start on opensuse tumbleweed

cmcgrath5035, 2024-01-20:
**Describe the bug**
From CLI: Flatpak run org.opencpn.OpenCPN

yields following message:

Gtk-Message: 14:16:27.405: Failed to load module "canberra-gtk-module"

(opencpn:2): Gdk-WARNING **: 14:16:27.458: The program 'opencpn' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadAccess (attempt to access private resource denied)'.
  (Details: serial 175 error_code 10 request_code 130 (MIT-SHM) minor_code 1)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

Is this an openCPN issue or Flatpak issue?

Appears to be a missing dependency in the Flatpak runtime

**Expected behavior**
openCPN earlier versions ran OK

**Desktop (please complete the following information if applicable):**
 - OS:opensuse Tumbleweed
 - Flatpak Version 1.15.6-1.2

nohal, 2024-01-20:
The message about `canberra-gtk-module` is harmless info

Can you try `flatpak run --share=ipc org.opencpn.OpenCPN`?
What desktop environment are you using?

cmcgrath5035, 2024-01-20:
flatpak run --share=ipc org.opencpn.OpenCPN 
opens openCPN, appears good, I need to configure charts

Plasma5/QT5 desktop

You are quick :)

nohal, 2024-01-20:
@leamas, I have well exceeded my flatpak knowledge here, do we need/want something like

```diff
diff --git a/flatpak/org.opencpn.OpenCPN.yaml b/flatpak/org.opencpn.OpenCPN.yaml
index 6e8972c6f..9f87e4975 100644
--- a/flatpak/org.opencpn.OpenCPN.yaml
+++ b/flatpak/org.opencpn.OpenCPN.yaml
@@ -33,7 +33,7 @@ finish-args:
     - --socket=x11
     - --socket=pulseaudio
     - --filesystem=home
-    - --share=network
+    - --share=ipc,network
     - --device=all
     - --allow=canbus
     - --talk-name=org.opencpn.OpenCPN.*
```

and if we do why the heck does the flatpak work for everybody else, me included, without it?

cmcgrath5035, 2024-01-20:
I find this in Flatpak documentation
--share=ipc - share IPC namespace with the host (necessary for X11)
I am running X11, perhaps you run Wayland?

nohal, 2024-01-21:
I am running both Wayland and X11 on my various machines, so it does not seem to be the problem, at least not the only problem. But I don't have a combination of X11 and KDE or a SuSE machine in my "collection" at the moment.

leamas, 2024-01-21:
sorry, a bit late to the party.

>  I have well exceeded my flatpak knowledge here, do we need/want something like [...]

@nohal : I have no idea, it might be an artifact from from recent IPC testing. Current code uses dbus on Flatpak, so we might need to update this.  Doubt it is related to this crash, though.

BTW: This manifest is just the one used for developer and nightly builds. The "real" stuff building the official flatpak releases lives at https://github.com/flathub/org.opencpn.OpenCPN

@cmcgrath5035 : Could you possibly try to create a stacktrace and submit it using instructions [here](https://opencpn-manuals.github.io/main/opencpn-dev/stacktrace.html#_flatpak)?

leamas, 2024-01-21:
@cmcgrath5035: The plot thickens. Are you using the official  Flatpak package available from https://flathub.org/apps/org.opencpn.OpenCPN? These packages are compiled using `share=network` , hence the question.

leamas, 2024-01-21:
>  Doubt it is related to this crash, though.

Well, of course it is. Let's wait for replies from @cmcgrath5035  before we jump into conclusions, though.

leamas, 2024-01-21:
> flatpak run --share=ipc org.opencpn.OpenCPN

Doc's says ipc is  not strictly  required for running X11,  but without it shared memory is disabled which gives Real Bad Performance. We should probably add it also to the official release. Need to contemplate a little, though.

cmcgrath5035, 2024-01-21:
Sorry, I am on US-EST and got distracted by football game last night.

I switched my TW system to Wayland, the startup problem remained, running with --share=ipc still OK.
I loaded openCPN via Flatpak commands on CLI, yields this
[image]

I am off to figure out creating a Stack Trace, it has been a while

BTW, my TW box is an AMD machine. I also have an AMD laptop running Win10, stock Flatpak openCPN runs fine there

cmcgrath5035, 2024-01-21:
Does this help?
Ran strace -o tracefile flatpak run  org.opencpn.OpenCPN

tracef
[tracefile.txt](https://github.com/OpenCPN/OpenCPN/files/14001091/tracefile.txt)

cmcgrath5035, 2024-01-21:
PS I had to rename tracefile to tracefile.txt for upload to this interface

leamas, 2024-01-21:
Sorry, but the strace is probably not that useful here.  A stacktrace wold, however.

The instructions for that ([above](https://opencpn-manuals.github.io/main/opencpn-dev/stacktrace.html#_flatpak)) should be pretty straight-forward. If not, just get in touch so I can clarify (and fix the instructions)

cmcgrath5035, 2024-01-21:
My bad, is missed the "here"
Took me a bit, first had to create a --user remote then downloaded the wrong version.
recovered from that but got an unusual result.
The first gdb  run crashed but produced  "no stack" when bt executed.
Hmmm
so I did (gdb) run again and openCPN opened normally !

Might the issue here be that I have the downloaded flatpak repo in a 'System' flathub remote, not a user?
I typically use the Discover GUI, only moved to CLI when the openCPN app silently crashed after startup.
Discover seems to do everything in system space, not --user space.
file that follows is the gdb dialog for the two runs.
[debugfile.txt](https://github.com/OpenCPN/OpenCPN/files/14001584/debugfile.txt)

leamas, 2024-01-21:
hm... you have no crash, and hence no stacktrace,  when running under gdb.  Frustrating.

System vs User repos should not be a concern as long as Flatpak finds everything needed, and it obviously does.

Contemplating updating the regular repo with `share=ipc`. Please give me some time.

cmcgrath5035, 2024-01-21:
Take as much time as you need, I know how to make it run in the interim.

My complements to both openCPN and Flatpak documentation teams.
Following the the "how-too" documents very easy and informative.

I am a little surprised that using the Discover GUI reveals nothing about --system vs --user,
perhaps I should say nothing obvious.

leamas, 2024-01-22:
@cmcgrath5035 

A test build is available. Could you possibly test it using `flatpak install --user https://dl.flathub.org/build-repo/77737/org.opencpn.OpenCPN.flatpakref` and report back?

cmcgrath5035, 2024-01-22:
Hmmm, I responded to the email, here is what I sent:
---------------------------
I ran this install line, but then could not figure out how to configure 'flatpak run' to choose the newer test version
This is what is loaded

carl@PVE-LinuxSRV8-TW:~> flatpak list
Name                       Application ID                           Version              Branch          Origin             Installation
Freedesktop Platform       org.freedesktop.Platform                 22.08.20             22.08           flathub            system
Mesa                       org.freedesktop.Platform.GL.default      23.3.0               22.08           flathub            system
Mesa                       org.freedesktop.Platform.GL.default      23.3.0               22.08           flathub-user       user
Mesa (Extra)               org.freedesktop.Platform.GL.default      23.3.0               22.08-extra     flathub            system
Mesa (Extra)               org.freedesktop.Platform.GL.default      23.3.0               22.08-extra     flathub-user       user
Mesa                       org.freedesktop.Platform.GL.default      23.3.2               23.08           flathub-user       user
Mesa (Extra)               org.freedesktop.Platform.GL.default      23.3.2               23.08-extra     flathub-user       user
openh264                   org.freedesktop.Platform.openh264        2.1.0                2.2.0           flathub            system
openh264                   org.freedesktop.Platform.openh264        2.1.0                2.2.0           flathub-user       user
Freedesktop SDK            org.freedesktop.Sdk                      22.08.20             22.08           flathub-user       user
Breeze GTK theme           org.gtk.Gtk3theme.Breeze                 5.27.8               3.22            flathub            system
Breeze GTK theme           org.gtk.Gtk3theme.Breeze                 5.27.8               3.22            flathub-user       user
OpenCPN                    org.opencpn.OpenCPN                      5.8.4-2+1637c28      stable          flathub            system
OpenCPN                    org.opencpn.OpenCPN                      5.8.4-3+1637c28      test            opencpn-origin     user
OpenCPN debug symbols      org.opencpn.OpenCPN.Debug                                     stable          flathub-user       user

So I opened the Discover GUI, it displayed two entries for openCPN, I chose the second one and got this
[image]

Which is the test build.
Open and seems to run fine

Seems good to go, Thanks

closed 2024-01-22 by bdbcat